TheImagingSource/tiscamera #412: internal data stream error in tcamsrc

nhi-vanye, 2021-06-30:
Running the binary bits from 0.13.1
```apt list | grep tisc

tiscamera/now 0.13.1 amd64 [installed,local]
tiscamera-dutils/now 0.13.2. amd64 [installed,local]
```

Having issues reading anything out of the camera on Linux (worked with no issues on Windows)

```
GST_DEBUG=tcamsrc:9 gst-launch-1.0 tcamsrc serial=35910714 ! videoconvert ! ximagesink
0:00:00.011185731 12508 0x55b3257c06c0 INFO                 tcamsrc gsttcamsrc.cpp:396:apply_element_property: Set camera serial to 35910714
2021.06.30T13:11:29:127        <DEBUG> Logger:108:
The following library versions are used:
	Tcam:	0.13.1_master/528f4d94_rev_3022
	Aravis:	not_configured_version_not_configured
Setting pipeline to PAUSED ...
0:00:00.015654922 12508 0x55b3257c06c0 DEBUG                tcamsrc gsttcamsrc.cpp:766:gst_tcam_src_change_state: State change: NULL -> READY
0:00:00.015666693 12508 0x55b3257c06c0 INFO                 tcamsrc gsttcamsrc.cpp:608:open_source_element: Searching for '35910714'
0:00:00.015674179 12508 0x55b3257c06c0 DEBUG                tcamsrc gsttcamsrc.cpp:613:open_source_element: Comparing '35910714' to '35910714'
2021.06.30T13:11:29:131        <DEBUG> AFU050Device.cpp:532: adding int 429785090 1 4
2021.06.30T13:11:29:132        <ERROR> AFU050Device.cpp:675: get_control returned with: 1
2021.06.30T13:11:29:132        <ERROR> AFU050Device.cpp:675: get_control returned with: 1
2021.06.30T13:11:29:132        <DEBUG> AFU050Device.cpp:532: adding int 429785092 3 4
2021.06.30T13:11:29:133        <ERROR> AFU050Device.cpp:675: get_control returned with: 1
2021.06.30T13:11:29:133        <ERROR> AFU050Device.cpp:675: get_control returned with: 1
2021.06.30T13:11:29:133        <ERROR> AFU050Device.cpp:675: get_control returned with: 1
2021.06.30T13:11:29:133        <ERROR> AFU050Device.cpp:675: get_control returned with: 1
2021.06.30T13:11:29:133        <DEBUG> AFU050Device.cpp:532: adding int 429793283 4 6
2021.06.30T13:11:29:134        <DEBUG> AFU050Device.cpp:532: adding int 429793284 4 7
2021.06.30T13:11:29:134        <DEBUG> AFU050Device.cpp:532: adding int 429793285 4 8
2021.06.30T13:11:29:135        <ERROR> AFU050Device.cpp:675: get_control returned with: 1
2021.06.30T13:11:29:135        <ERROR> AFU050Device.cpp:675: get_control returned with: 1
2021.06.30T13:11:29:135        <DEBUG> AFU050Device.cpp:532: adding int 429785121 3 7
2021.06.30T13:11:29:135        <DEBUG> AFU050Device.cpp:532: adding int 429785120 3 3
2021.06.30T13:11:29:136        <DEBUG> AFU050Device.cpp:532: adding int 429785920 3 8
2021.06.30T13:11:29:136        <DEBUG> PipelineManager.cpp:127: Received 1 formats.
2021.06.30T13:11:29:136        <DEBUG> PropertyHandler.cpp:273: Created 12 properties
0:00:00.021404406 12508 0x55b3257c06c0 DEBUG                tcamsrc gsttcamsrc.cpp:678:open_source_element: Ghost pad target set
0:00:00.021438850 12508 0x55b3257c06c0 INFO                 tcamsrc gsttcamsrc.cpp:748:open_source_element: Opened device has serial: '35910714' type: 'libusb'
0:00:00.021443655 12508 0x55b3257c06c0 INFO                 tcamsrc gsttcamsrc.cpp:775:gst_tcam_src_change_state: Opened source element
Pipeline is live and does not need PREROLL ...
/GstPipeline:pipeline0/GstTcamSrc:tcamsrc0/GstTcamMainSrc:tcamsrc-mainsrc: camera-buffers = 10
/GstPipeline:pipeline0/GstTcamSrc:tcamsrc0/GstTcamMainSrc:tcamsrc-mainsrc: num-buffers = -1
/GstPipeline:pipeline0/GstTcamSrc:tcamsrc0/GstTcamMainSrc:tcamsrc-mainsrc: drop-incomplete-buffer = true
/GstPipeline:pipeline0/GstTcamSrc:tcamsrc0/GstTcamMainSrc:tcamsrc-mainsrc: do-timestamp = true
Setting pipeline to PLAYING ...
2021.06.30T13:11:29:137        <INFO> ImageSource.cpp:68: Source changed to state STOPPED
New clock: GstSystemClock
2021.06.30T13:11:29:137        <DEBUG> AFU050Device.cpp:660: set_video_format transfer ended with 34
2021.06.30T13:11:29:137        <DEBUG> PipelineManager.cpp:222: Found device fourcc 'MJPG' - 47504a4d
2021.06.30T13:11:29:915        <DEBUG> AFU050Device.cpp:660: set_video_format transfer ended with 34
2021.06.30T13:11:29:915        <INFO> ImageBuffer.cpp:38: allocating data buffer
2021.06.30T13:11:29:915        <INFO> ImageBuffer.cpp:38: allocating data buffer
2021.06.30T13:11:29:915        <INFO> ImageBuffer.cpp:38: allocating data buffer
2021.06.30T13:11:29:915        <INFO> ImageBuffer.cpp:38: allocating data buffer
2021.06.30T13:11:29:915        <INFO> ImageBuffer.cpp:38: allocating data buffer
2021.06.30T13:11:29:915        <INFO> ImageBuffer.cpp:38: allocating data buffer
2021.06.30T13:11:29:915        <INFO> ImageBuffer.cpp:38: allocating data buffer
2021.06.30T13:11:29:915        <INFO> ImageBuffer.cpp:38: allocating data buffer
2021.06.30T13:11:29:915        <INFO> ImageBuffer.cpp:38: allocating data buffer
2021.06.30T13:11:29:915        <INFO> ImageBuffer.cpp:38: allocating data buffer
2021.06.30T13:11:29:915        <INFO> PipelineManager.cpp:476: Pipeline creation successful.
2021.06.30T13:11:29:915        <INFO> PipelineManager.cpp:486: source ->  sink
2021.06.30T13:11:29:915        <INFO> ImageSink.cpp:51: Pipeline started playing
2021.06.30T13:11:29:915        <INFO> AFU050Device.cpp:261: Received 10 buffer from external allocator.
2021.06.30T13:11:29:915        <INFO> AFU050Device.cpp:468: Starting stream...
2021.06.30T13:11:29:916        <DEBUG> ImageSource.cpp:57: PLAYING....
2021.06.30T13:11:29:916        <INFO> PipelineManager.cpp:83: All pipeline elements set to PLAYING.
/GstPipeline:pipeline0/GstTcamSrc:tcamsrc0/GstTcamMainSrc:tcamsrc-mainsrc.GstPad:src: caps = image/jpeg, width=(int)2592, height=(int)1944, framerate=(fraction)15/1
ERROR: from element /GstPipeline:pipeline0/GstTcamSrc:tcamsrc0/GstTcamMainSrc:tcamsrc-mainsrc: Internal data stream error.
Additional debug info:
gstbasesrc.c(3055): gst_base_src_loop (): /GstPipeline:pipeline0/GstTcamSrc:tcamsrc0/GstTcamMainSrc:tcamsrc-mainsrc:
streaming stopped, reason not-negotiated (-4)
Execution ended after 0:00:00.966257992
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
2021.06.30T13:11:30:103        <INFO> ImageSource.cpp:68: Source changed to state STOPPED
2021.06.30T13:11:30:103        <INFO> ImageSink.cpp:55: Pipeline stopped playing
Setting pipeline to NULL ...
2021.06.30T13:11:30:137        <INFO> CaptureDeviceImpl.cpp:159: Closed device DFK AFU050-L34.
Freeing pipeline ...

```

Tried various combinations of different caps, the end goal is to capture a single image (not video) from the camera.

To that end - is it possible to bypass GStreamer and simply access the libusb camera directly ? I'm on an embedded machine (that already has V4L) and would like to simplify deployment/runtime, it looks like none of the library symbols are exported into installed headers. Best place to start peeling the onion ?

TIS-Stefan, 2021-06-30:
Hello 

Please use tcam**bin** instead of tcamsrc. The AFU050 provides MJPEG and the videoconvert module does not handle that. The tcambin module has an internal MJPEG decoder.

Stefan

nhi-vanye, 2021-06-30:
I tried the instructions in #163 - tcam-capture UI started, but the "format", "Resolution" and "FPS" drop downs were unfilled and clikcking "capture" resulted in an error

```
2021.06.30T13:42:34:254        <DEBUG> AFU050PropertyHandler.cpp:74: Setting int Contrast to: 32
2021.06.30T13:42:34:255        <DEBUG> AFU050PropertyHandler.cpp:74: Setting int Sharpness to: 0
Traceback (most recent call last):
  File "/home/roffer/src/tiscamera/tools/tcam-capture/tcam-capture", line 426, in start_recording_video
    self.view.start_recording_video(self.settings.get_video_type())
  File "/home/roffer/src/tiscamera/tools/tcam-capture/tcam_capture/TcamView.py", line 220, in start_recording_video
    self.videosaver.start_recording_video(video_type)
  File "/home/roffer/src/tiscamera/tools/tcam-capture/tcam_capture/MediaSaver.py", line 208, in start_recording_video
    self._create_pipeline()
  File "/home/roffer/src/tiscamera/tools/tcam-capture/tcam_capture/MediaSaver.py", line 138, in _create_pipeline
    self.caps.to_string(),
AttributeError: 'NoneType' object has no attribute 'to_string'
2021.06.30T13:42:36:231        <DEBUG> AFU050PropertyHandler.cpp:74: Setting int Exposure to: 300
2021.06.30T13:42:36:235        <DEBUG> AFU050PropertyHandler.cpp:81: Setting bool Exposure Auto to: 1
2021.06.30T13:42:36:240        <DEBUG> AFU050PropertyHandler.cpp:74: Setting int Gain to: 68
```

TIS-Stefan, 2021-06-30:
Do you get an image with gst-launch-1.0, as I wrote above?
However, it looks for me, that there is a issue with selecting video format and frame rate. I hope, you can work with the default values.

nhi-vanye, 2021-06-30:
No, here's the update

```
(py3) roffer@ubuntu18:~/SRC/tiscamera/examples/c$ tcam-ctrl -l
Model: DFK AFU050-L34 Serial: 35910714 Type: libusb
(py3) roffer@ubuntu18:~/SRC/tiscamera/examples/c$ tcam-ctrl -c 35910714
Available gstreamer-1.0 caps:
image/jpeg,width=2592,height=1944,framerate={15/1};
image/jpeg,width=1920,height=1080,framerate={30/1};
image/jpeg,width=1280,height=960,framerate={60/1}
```

```
GST_DEBUG=tcambin:9 GST_DEBUG_NO_COLOR=1 gst-launch-1.0 tcambin serial=35910714 ! video/x-raw,format=BGRx,width=1920,height=1080,framerate=30/1 ! videoconvert ! ximagesink
0:00:00.014455209 12772 0x558a7a6d4520 DEBUG                tcambin gsttcambin.cpp:1520:gst_tcambin_init:<GstTcamBin@0x558a7a73a050> init
Setting pipeline to PAUSED ...
0:00:00.048721369 12772 0x558a7a6d4520 INFO                 tcambin gsttcambin.cpp:1130:gst_tcam_bin_change_state: NULL_TO_READY
0:00:00.048799969 12772 0x558a7a6d4520 DEBUG                tcambin gsttcambin.cpp:425:gst_tcambin_create_source: Creating source...
0:00:00.049377557 12772 0x558a7a6d4520 INFO                 tcambin gsttcambin.cpp:438:gst_tcambin_create_source: Setting source serial to 35910714
2021.06.30T13:56:19:339        <DEBUG> Logger:108:
The following library versions are used:
	Tcam:	0.13.1_master/528f4d94_rev_3022
	Aravis:	not_configured_version_not_configured
2021.06.30T13:56:19:342        <DEBUG> AFU050Device.cpp:532: adding int 429785090 1 4
2021.06.30T13:56:19:343        <ERROR> AFU050Device.cpp:675: get_control returned with: 1
2021.06.30T13:56:19:343        <ERROR> AFU050Device.cpp:675: get_control returned with: 1
2021.06.30T13:56:19:343        <DEBUG> AFU050Device.cpp:532: adding int 429785092 3 4
2021.06.30T13:56:19:345        <ERROR> AFU050Device.cpp:675: get_control returned with: 1
2021.06.30T13:56:19:345        <ERROR> AFU050Device.cpp:675: get_control returned with: 1
2021.06.30T13:56:19:345        <ERROR> AFU050Device.cpp:675: get_control returned with: 1
2021.06.30T13:56:19:345        <ERROR> AFU050Device.cpp:675: get_control returned with: 1
2021.06.30T13:56:19:345        <DEBUG> AFU050Device.cpp:532: adding int 429793283 4 6
2021.06.30T13:56:19:346        <DEBUG> AFU050Device.cpp:532: adding int 429793284 4 7
2021.06.30T13:56:19:346        <DEBUG> AFU050Device.cpp:532: adding int 429793285 4 8
2021.06.30T13:56:19:347        <ERROR> AFU050Device.cpp:675: get_control returned with: 1
2021.06.30T13:56:19:347        <ERROR> AFU050Device.cpp:675: get_control returned with: 1
2021.06.30T13:56:19:347        <DEBUG> AFU050Device.cpp:532: adding int 429785121 3 7
2021.06.30T13:56:19:347        <DEBUG> AFU050Device.cpp:532: adding int 429785120 3 3
2021.06.30T13:56:19:348        <DEBUG> AFU050Device.cpp:532: adding int 429785920 3 8
2021.06.30T13:56:19:348        <DEBUG> PipelineManager.cpp:127: Received 1 formats.
2021.06.30T13:56:19:348        <DEBUG> PropertyHandler.cpp:273: Created 12 properties
0:00:00.058627593 12772 0x558a7a6d4520 INFO                 tcambin gsttcambin.cpp:464:gst_tcambin_create_source:<tcambin0> Opened device has serial: '35910714' type: 'libusb'
0:00:00.058672178 12772 0x558a7a6d4520 INFO                 tcambin gsttcambin.cpp:467:gst_tcambin_create_source:<tcambin0> caps of src: image/jpeg, width=(int)2592, height=(int)1944, framerate=(fraction){ 15/1 }; image/jpeg, width=(int)1920, height=(int)1080, framerate=(fraction){ 30/1 }; image/jpeg, width=(int)1280, height=(int)960, framerate=(fraction){ 60/1 }
0:00:00.058744743 12772 0x558a7a6d4520 INFO                 tcambin gsttcambin.cpp:587:gst_tcambin_create_elements: creating elements
0:00:00.059120637 12772 0x558a7a6d4520 DEBUG                tcambin gsttcambin.cpp:566:create_and_add_element: Adding tcamdutils(0x558a7a7880f0) to pipeline
0:00:00.059165249 12772 0x558a7a6d4520 DEBUG                tcambin gsttcambin.cpp:566:create_and_add_element: Adding videoconvert(0x558a7a78ff80) to pipeline
0:00:00.059234276 12772 0x558a7a6d4520 INFO                 tcambin gsttcambin.cpp:1159:gst_tcam_bin_change_state: READY_TO_PAUSED
0:00:00.059259045 12772 0x558a7a6d4520 INFO                 tcambin gsttcambin.cpp:1184:gst_tcam_bin_change_state:<tcambin0> caps of sink: video/x-raw, format=(string)BGRx, width=(int)1920, height=(int)1080, framerate=(fraction)30/1
0:00:00.059288228 12772 0x558a7a6d4520 INFO                 tcambin gsttcambin.cpp:1188:gst_tcam_bin_change_state:<tcambin0> caps of src: image/jpeg, width=(int)2592, height=(int)1944, framerate=(fraction){ 15/1 }; image/jpeg, width=(int)1920, height=(int)1080, framerate=(fraction){ 30/1 }; image/jpeg, width=(int)1280, height=(int)960, framerate=(fraction){ 60/1 }
2021.06.30T13:56:19:349        <ERROR> tcamgstbase.cpp:1230: Could not negotiate caps
0:00:00.059472249 12772 0x558a7a6d4520 ERROR                tcambin gsttcambin.cpp:1241:gst_tcam_bin_change_state:<tcambin0> Unable to work with given caps: video/x-raw, format=(string)BGRx, width=(int)1920, height=(int)1080, framerate=(fraction)30/1
0:00:00.059517949 12772 0x558a7a6d4520 INFO                 tcambin gsttcambin.cpp:1294:gst_tcam_bin_change_state: Changing state: READY->READY
ERROR: Pipeline doesn't want to pause.
Setting pipeline to NULL ...
0:00:00.059808714 12772 0x558a7a6d4520 INFO                 tcambin gsttcambin.cpp:1294:gst_tcam_bin_change_state: Changing state: READY->READY
0:00:00.104682074 12772 0x558a7a6d4520 INFO                 tcambin gsttcambin.cpp:1294:gst_tcam_bin_change_state: Changing state: READY->NULL
2021.06.30T13:56:19:395        <INFO> ImageSource.cpp:68: Source changed to state STOPPED
2021.06.30T13:56:19:395        <INFO> CaptureDeviceImpl.cpp:159: Closed device DFK AFU050-L34.
Freeing pipeline ...
0:00:00.105098954 12772 0x558a7a6d4520 DEBUG                tcambin gsttcambin.cpp:1571:gst_tcambin_dispose: dispose
```

nhi-vanye, 2021-06-30:
Skipping the `video/x-raw` step and I get 

```
GST_DEBUG=tcambin:9 GST_DEBUG_NO_COLOR=1 gst-launch-1.0 tcambin serial=35910714 ! videoconvert ! ximagesink
0:00:00.010607743 12798 0x55edd501f520 DEBUG                tcambin gsttcambin.cpp:1520:gst_tcambin_init:<GstTcamBin@0x55edd5084050> init
Setting pipeline to PAUSED ...
0:00:00.021695449 12798 0x55edd501f520 INFO                 tcambin gsttcambin.cpp:1130:gst_tcam_bin_change_state: NULL_TO_READY
0:00:00.021702181 12798 0x55edd501f520 DEBUG                tcambin gsttcambin.cpp:425:gst_tcambin_create_source: Creating source...
0:00:00.022263453 12798 0x55edd501f520 INFO                 tcambin gsttcambin.cpp:438:gst_tcambin_create_source: Setting source serial to 35910714
2021.06.30T14:01:04:790        <DEBUG> Logger:108:
The following library versions are used:
	Tcam:	0.13.1_master/528f4d94_rev_3022
	Aravis:	not_configured_version_not_configured
2021.06.30T14:01:04:796        <DEBUG> AFU050Device.cpp:532: adding int 429785090 1 4
2021.06.30T14:01:04:796        <ERROR> AFU050Device.cpp:675: get_control returned with: 1
2021.06.30T14:01:04:796        <ERROR> AFU050Device.cpp:675: get_control returned with: 1
2021.06.30T14:01:04:796        <DEBUG> AFU050Device.cpp:532: adding int 429785092 3 4
2021.06.30T14:01:04:802        <ERROR> AFU050Device.cpp:675: get_control returned with: 1
2021.06.30T14:01:04:802        <ERROR> AFU050Device.cpp:675: get_control returned with: 1
2021.06.30T14:01:04:802        <ERROR> AFU050Device.cpp:675: get_control returned with: 1
2021.06.30T14:01:04:802        <ERROR> AFU050Device.cpp:675: get_control returned with: 1
2021.06.30T14:01:04:802        <DEBUG> AFU050Device.cpp:532: adding int 429793283 4 6
2021.06.30T14:01:04:803        <DEBUG> AFU050Device.cpp:532: adding int 429793284 4 7
2021.06.30T14:01:04:803        <DEBUG> AFU050Device.cpp:532: adding int 429793285 4 8
2021.06.30T14:01:04:804        <ERROR> AFU050Device.cpp:675: get_control returned with: 1
2021.06.30T14:01:04:804        <ERROR> AFU050Device.cpp:675: get_control returned with: 1
2021.06.30T14:01:04:804        <DEBUG> AFU050Device.cpp:532: adding int 429785121 3 7
2021.06.30T14:01:04:804        <DEBUG> AFU050Device.cpp:532: adding int 429785120 3 3
2021.06.30T14:01:04:805        <DEBUG> AFU050Device.cpp:532: adding int 429785920 3 8
2021.06.30T14:01:04:805        <DEBUG> PipelineManager.cpp:127: Received 1 formats.
2021.06.30T14:01:04:805        <DEBUG> PropertyHandler.cpp:273: Created 12 properties
0:00:00.038126824 12798 0x55edd501f520 INFO                 tcambin gsttcambin.cpp:464:gst_tcambin_create_source:<tcambin0> Opened device has serial: '35910714' type: 'libusb'
0:00:00.038178332 12798 0x55edd501f520 INFO                 tcambin gsttcambin.cpp:467:gst_tcambin_create_source:<tcambin0> caps of src: image/jpeg, width=(int)2592, height=(int)1944, framerate=(fraction){ 15/1 }; image/jpeg, width=(int)1920, height=(int)1080, framerate=(fraction){ 30/1 }; image/jpeg, width=(int)1280, height=(int)960, framerate=(fraction){ 60/1 }
0:00:00.038626972 12798 0x55edd501f520 INFO                 tcambin gsttcambin.cpp:587:gst_tcambin_create_elements: creating elements
0:00:00.038951538 12798 0x55edd501f520 DEBUG                tcambin gsttcambin.cpp:566:create_and_add_element: Adding tcamdutils(0x55edd50d40f0) to pipeline
0:00:00.038996811 12798 0x55edd501f520 DEBUG                tcambin gsttcambin.cpp:566:create_and_add_element: Adding videoconvert(0x55edd50d3640) to pipeline
0:00:00.039069706 12798 0x55edd501f520 INFO                 tcambin gsttcambin.cpp:1159:gst_tcam_bin_change_state: READY_TO_PAUSED
0:00:00.039163102 12798 0x55edd501f520 INFO                 tcambin gsttcambin.cpp:1184:gst_tcam_bin_change_state:<tcambin0> caps of sink: video/x-raw, format=(string)BGRx, width=(int)[ 1, 2147483647 ], height=(int)[ 1, 2147483647 ], framerate=(fraction)[ 0/1, 2147483647/1 ], pixel-aspect-ratio=(fraction)1/1; video/x-raw, width=(int)[ 1, 2147483647 ], height=(int)[ 1, 2147483647 ], framerate=(fraction)[ 0/1, 2147483647/1 ], pixel-aspect-ratio=(fraction)1/1, format=(string){ I420, YV12, YUY2, UYVY, AYUV, RGBx, BGRx, xRGB, xBGR, RGBA, BGRA, ARGB, ABGR, RGB, BGR, Y41B, Y42B, YVYU, Y444, v210, v216, NV12, NV21, GRAY8, GRAY16_BE, GRAY16_LE, v308, RGB16, BGR16, RGB15, BGR15, UYVP, A420, RGB8P, YUV9, YVU9, IYU1, ARGB64, AYUV64, r210, I420_10BE, I420_10LE, I422_10BE, I422_10LE, Y444_10BE, Y444_10LE, GBR, GBR_10BE, GBR_10LE, NV16, NV24, NV12_64Z32, A420_10BE, A420_10LE, A422_10BE, A422_10LE, A444_10BE, A444_10LE, NV61, P010_10BE, P010_10LE, IYU2, VYUY, GBRA, GBRA_10BE, GBRA_10LE, GBR_12BE, GBR_12LE, GBRA_12BE, GBRA_12LE, I420_12BE, I420_12LE, I422_12BE, I422_12LE, Y444_12BE, Y444_12LE, GRAY10_LE32, NV12_10LE32, NV16_10LE32 }
0:00:00.039202338 12798 0x55edd501f520 INFO                 tcambin gsttcambin.cpp:1188:gst_tcam_bin_change_state:<tcambin0> caps of src: image/jpeg, width=(int)2592, height=(int)1944, framerate=(fraction){ 15/1 }; image/jpeg, width=(int)1920, height=(int)1080, framerate=(fraction){ 30/1 }; image/jpeg, width=(int)1280, height=(int)960, framerate=(fraction){ 60/1 }
2021.06.30T14:01:04:807        <ERROR> tcamgstbase.cpp:1230: Could not negotiate caps
0:00:00.039379555 12798 0x55edd501f520 ERROR                tcambin gsttcambin.cpp:1241:gst_tcam_bin_change_state:<tcambin0> Unable to work with given caps: video/x-raw, format=(string)BGRx, width=(int)[ 1, 2147483647 ], height=(int)[ 1, 2147483647 ], framerate=(fraction)[ 0/1, 2147483647/1 ], pixe [... cut ...]
0:00:00.039678741 12798 0x55edd501f520 INFO                 tcambin gsttcambin.cpp:1294:gst_tcam_bin_change_state: Changing state: READY->READY
0:00:00.076481169 12798 0x55edd501f520 INFO                 tcambin gsttcambin.cpp:1294:gst_tcam_bin_change_state: Changing state: READY->NULL
2021.06.30T14:01:04:844        <INFO> ImageSource.cpp:68: Source changed to state STOPPED
2021.06.30T14:01:04:844        <INFO> CaptureDeviceImpl.cpp:159: Closed device DFK AFU050-L34.
Freeing pipeline ...
0:00:00.077286279 12798 0x55edd501f520 DEBUG                tcambin gsttcambin.cpp:1571:gst_tcambin_dispose: dispose
```

nhi-vanye, 2021-06-30:
Yeah Go ME!!! :-)

I have done both - tried the official bits, then built from scratch the 0.13.1 tag (in the same ubuntu 18.04 VM) - This is running on a vSphere host with the USB device mapped into the VM.

Is it possible to remove GStreamer out of the path and read direct from the libusb camera ?

TIS-Stefan, 2021-06-30:
You may look into the source code https://github.com/TheImagingSource/tiscamera/tree/master/src/libusb and grab the parts you need.

Stefan

nhi-vanye, 2021-07-01:
@TIS-Stefan - could you repeat your working execution but with GST_DEBUG set?

Decided I need to get to the root cause of this... Trying to work out why its can't handle the caps negotiation.

TIS-Stefan, 2021-07-02:
Hello

Can you please tell me, whether you get an image with:
gst-launch-1.0 tcambin serial=nnnnn ! video/x-raw, format=BGRx, width=1920,height=1080, framerate=30/1 ! videoconvert ! ximagesink
This totally unclear to me right now.

Stefan

nhi-vanye, 2021-07-02:
```
roffer@ubuntu18:~$ tcam-ctrl -l
Model: DFK AFU050-L34 Serial: 35910714 Type: libusb

roffer@ubuntu18:~$ GST_DEBUG=tcambin:9 gst-launch-1.0 tcambin serial=35910714 ! video/x-raw,format=BGRx,width=1920,height=1080,framerate=30/1 ! videoconvert ! ximagesink
0:00:00.054907893  7756 0x55e20eba9720 DEBUG                tcambin gsttcambin.cpp:1520:gst_tcambin_init:<GstTcamBin@0x55e20ec1a050> init
Setting pipeline to PAUSED ...
0:00:00.434364989  7756 0x55e20eba9720 INFO                 tcambin gsttcambin.cpp:1130:gst_tcam_bin_change_state: NULL_TO_READY
0:00:00.434389834  7756 0x55e20eba9720 DEBUG                tcambin gsttcambin.cpp:425:gst_tcambin_create_source: Creating source...
0:00:00.435275837  7756 0x55e20eba9720 INFO                 tcambin gsttcambin.cpp:438:gst_tcambin_create_source: Setting source serial to 35910714
0:00:00.462451051  7756 0x55e20eba9720 INFO                 tcambin gsttcambin.cpp:464:gst_tcambin_create_source:<tcambin0> Opened device has serial: '35910714' type: 'libusb'
0:00:00.462533793  7756 0x55e20eba9720 INFO                 tcambin gsttcambin.cpp:467:gst_tcambin_create_source:<tcambin0> caps of src: image/jpeg, width=(int)2592, height=(int)1944, framerate=(fraction){ 15/1 }; image/jpeg, width=(int)1920, height=(int)1080, framerate=(fraction){ 30/1 }; image/jpeg, width=(int)1280, height=(int)960, framerate=(fraction){ 60/1 }
0:00:00.462633032  7756 0x55e20eba9720 INFO                 tcambin gsttcambin.cpp:587:gst_tcambin_create_elements: creating elements
0:00:00.490519953  7756 0x55e20eba9720 DEBUG                tcambin gsttcambin.cpp:566:create_and_add_element: Adding tcamdutils(0x55e20ec680f0) to pipeline
0:00:00.490617828  7756 0x55e20eba9720 DEBUG                tcambin gsttcambin.cpp:566:create_and_add_element: Adding videoconvert(0x55e20ec6ff80) to pipeline
0:00:00.490752692  7756 0x55e20eba9720 INFO                 tcambin gsttcambin.cpp:1159:gst_tcam_bin_change_state: READY_TO_PAUSED
0:00:00.490782628  7756 0x55e20eba9720 INFO                 tcambin gsttcambin.cpp:1184:gst_tcam_bin_change_state:<tcambin0> caps of sink: video/x-raw, format=(string)BGRx, width=(int)1920, height=(int)1080, framerate=(fraction)30/1
0:00:00.490827922  7756 0x55e20eba9720 INFO                 tcambin gsttcambin.cpp:1188:gst_tcam_bin_change_state:<tcambin0> caps of src: image/jpeg, width=(int)2592, height=(int)1944, framerate=(fraction){ 15/1 }; image/jpeg, width=(int)1920, height=(int)1080, framerate=(fraction){ 30/1 }; image/jpeg, width=(int)1280, height=(int)960, framerate=(fraction){ 60/1 }
0:00:00.491047841  7756 0x55e20eba9720 ERROR                tcambin gsttcambin.cpp:1241:gst_tcam_bin_change_state:<tcambin0> Unable to work with given caps: video/x-raw, format=(string)BGRx, width=(int)1920, height=(int)1080, framerate=(fraction)30/1
0:00:00.491124539  7756 0x55e20eba9720 INFO                 tcambin gsttcambin.cpp:1294:gst_tcam_bin_change_state: Changing state: READY->READY
ERROR: Pipeline doesn't want to pause.
Setting pipeline to NULL ...
0:00:00.491469697  7756 0x55e20eba9720 INFO                 tcambin gsttcambin.cpp:1294:gst_tcam_bin_change_state: Changing state: READY->READY
0:00:00.534726680  7756 0x55e20eba9720 INFO                 tcambin gsttcambin.cpp:1294:gst_tcam_bin_change_state: Changing state: READY->NULL
Freeing pipeline ...
0:00:00.535084524  7756 0x55e20eba9720 DEBUG                tcambin gsttcambin.cpp:1571:gst_tcambin_dispose: dispose
```

TIS-Stefan, 2021-07-05:
Hello 
I do not dare to ask again:
Do you get an image with pipeline
gst-launch-1.0 tcambin serial=nnnnn ! video/x-raw, format=BGRx, width=1920,height=1080, framerate=30/1 ! videoconvert ! ximagesink

Please answer with "yes" or "no".

Stefan

TIS-Stefan, 2021-07-05:
(BTW: The 0.14 release does not work with the AFU050 currently. Therefore do not update)

nhi-vanye, 2021-07-05:
> Hello
> I do not dare to ask again:
> Do you get an image with pipeline
> gst-launch-1.0 tcambin serial=nnnnn ! video/x-raw, format=BGRx, width=1920,height=1080, framerate=30/1 ! videoconvert ! ximagesink
> 
> Please answer with "yes" or "no".
> 
> Stefan

No

TIS-Stefan, 2021-07-05:
Hello
Do you have the opportunity to run the camera on a native system, not on a VM?
Is the USB speed in the VM correctly set? VMWare sets USB speed to 1.1 by default, if I remember correctly.

nhi-vanye, 2021-07-05:
I don't have immediate access to a ubuntu 18 hw system (hence the VM) - I did add a USB 3.1 controller to the VM (this is a vSphere 7 host). I can attach it to a physical ubuntu 18 system that but it will take me a day or so to get that rebuilt.

I have tried the camera on a Windows 10 system with no issues - so the camera is (most probably) good.

Eventually I need to back port this to Ubuntu 14 and have a hardware system ready for that - I just wanted to get something working using a known tested environment before attempting the back port.

Unclear how to verify the bus speed the device is seeing

```
Bus 001 Device 003: ID 199e:8209 The Imaging Source Europe GmbH
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 ?
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x199e The Imaging Source Europe GmbH
  idProduct          0x8209
  bcdDevice            8.13
  iManufacturer           2 The Imaging Source Europe GmbH
  iProduct                1 DFK AFU050-L34
  iSerial                 3 35910714
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength          273
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         2
      bFunctionClass        255 Vendor Specific Class
      bFunctionSubClass       3
      bFunctionProtocol       0
      iFunction               0
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      1
      bInterfaceProtocol      0
      iInterface              0
      ** UNRECOGNIZED:  0d 24 01 00 01 4f 00 80 8d 5b 00 01 01
      ** UNRECOGNIZED:  12 24 02 01 01 02 00 00 00 00 00 00 00 00 03 08 00 04
      ** UNRECOGNIZED:  0c 24 05 03 01 00 00 03 3a 02 00 00
      ** UNRECOGNIZED:  1c 24 06 04 95 d6 3e 92 f2 56 e1 11 83 f7 37 70 48 24 01 9b 08 01 03 03 e0 f9 3f 00
      ** UNRECOGNIZED:  09 24 03 02 01 01 00 04 00
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               9
    Interface Descriptor:
[... 95 lines cut ...]
        bMatrixCoefficients                 0 (Unspecified)
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               1
Device Qualifier (for other device speed):
  bLength                10
  bDescriptorType         6
  bcdUSB               2.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 ?
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  bNumConfigurations      1
Device Status:     0x0000
  (Bus Powered)
```

But my gut says its on the VM's USB 2.0 hub...

TIS-Stefan, 2021-07-05:
Hello

lsusb -t 

show the speed of connections.

nhi-vanye, 2021-07-05:
```
roffer@ubuntu18:~$ lsusb -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 10000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 480M
    |__ Port 1: Dev 2, If 0, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 2: Dev 6, If 0, Class=Vendor Specific Class, Driver=, 480M
    |__ Port 2: Dev 6, If 1, Class=Video, Driver=, 480M
    |__ Port 3: Dev 4, If 0, Class=Hub, Driver=hub/7p, 12M
    |__ Port 4: Dev 5, If 0, Class=Hub, Driver=hub/7p, 480M
```

You've given me a couple of approaches to try out - I'll get back to you in a bit...

TIS-Stefan, 2021-07-05:
The speeds look all perfectly to me.

nhi-vanye, 2021-07-05:
@TIS-Stefan can you give me the output with GS_DEBUG set for when you run the pipeline, I feel bad about wasting your time on this but I don't have a known good system to compare it with, I'd like to see the result of the caps negotiation that I'm failing on...

I suspect I am going to need to dive into the code sooner or later as I need this on Ubuntu 14...

TIS-Stefan, 2021-07-05:
Here we go:

export GST_DEBUG=5
export GST_DEBUG_FILE=.afu050.log

gst-launch-1.0 tcambin serial=50499910 ! video/x-raw, format=BGRx, width=1920,height=1080, framerate=30/1 ! videoconvert ! ximagesink

Gives:
[afu050.zip](https://github.com/TheImagingSource/tiscamera/files/6764029/afu050.zip)

Stefan

TIS-Stefan, 2021-07-05:
BTW: You are not wasting my time. It is my job to help solving issues like this.

TIS-Stefan, 2021-07-06:
Hello

I found an old project for the DFK AFU050, which is completely independent of tiscamera modules:

[afu050.zip](https://github.com/TheImagingSource/tiscamera/files/6768511/afu050.zip)

Please follow the reame.txt in the zip file.
I adapted the makefile to GStreamer 1.0 and gtk3.0. But, if I remember correctly, the sample can be built with GStreamer 0.1 and GTK 2 too . You only need to adapt the make file. Then you can even use the old Ubuntu 14, if needed.

Stefan

nhi-vanye, 2021-07-06:
@TIS-Stefan - it looks like the latest zip file (expect old project) has the same contents as the previous zip file (log file).

Thanks by the way, this will hopefully help me towards the short term project needs while I work on the longer term solution...

TIS-Stefan, 2021-07-06:
I am sorry.... seems I am getting old.

[libafu050.zip](https://github.com/TheImagingSource/tiscamera/files/6769015/libafu050.zip)

Here we go...

nhi-vanye, 2021-07-07:
I've replicated the error on a clean physical Ubuntu 18.04 LTS system (replaced disk on my vSphere host and installed Ubuntu onto new disk).

To get the list of required packages I did a git checkout then `./scripts/dependency-manager install --runtime` before installing  `tiscamera_0.13.1.3020_amd64_ubuntu_1804.deb` and `tiscamera-dutils_0.13.2.377_amd64.deb`

So its still seems that tcambin can't handle the video/x-raw format

No that I've proved its not environmental (clean install on second (physical hardware) system), I'll be taking a look at the internals...

```
0:00:00.031399314 21044 0x5570cde21d20 ERROR                tcambin gsttcambin.cpp:1241:gst_tcam_bin_change_state:<tcambin0> Unable to work with given caps: video/x-raw, format=(string)BGRx, width=(int)1920, height=(int)1080, framerate=(fraction)30/1
```

```
offer@nuc-1:~$ GST_DEBUG=tcambin:5 gst-launch-1.0 tcambin serial=35910714 ! video/x-raw,format=BGRx,width=1920,height=1080,framerate=30/1 ! videoconvert ! ximagesink
0:00:00.012602194 21044 0x5570cde21d20 DEBUG                tcambin gsttcambin.cpp:1520:gst_tcambin_init:<GstTcamBin@0x5570cde94050> init
Setting pipeline to PAUSED ...
0:00:00.023454821 21044 0x5570cde21d20 INFO                 tcambin gsttcambin.cpp:1130:gst_tcam_bin_change_state: NULL_TO_READY
0:00:00.023465174 21044 0x5570cde21d20 DEBUG                tcambin gsttcambin.cpp:425:gst_tcambin_create_source: Creating source...
0:00:00.024041696 21044 0x5570cde21d20 INFO                 tcambin gsttcambin.cpp:438:gst_tcambin_create_source: Setting source serial to 35910714
0:00:00.030382420 21044 0x5570cde21d20 INFO                 tcambin gsttcambin.cpp:464:gst_tcambin_create_source:<tcambin0> Opened device has serial: '35910714' type: 'libusb'
0:00:00.030429115 21044 0x5570cde21d20 INFO                 tcambin gsttcambin.cpp:467:gst_tcambin_create_source:<tcambin0> caps of src: image/jpeg, width=(int)2592, height=(int)1944, framerate=(fraction){ 15/1 }; image/jpeg, width=(int)1920, height=(int)1080, framerate=(fraction){ 30/1 }; image/jpeg, width=(int)1280, height=(int)960, framerate=(fraction){ 60/1 }
0:00:00.030502567 21044 0x5570cde21d20 INFO                 tcambin gsttcambin.cpp:587:gst_tcambin_create_elements: creating elements
0:00:00.030991440 21044 0x5570cde21d20 DEBUG                tcambin gsttcambin.cpp:566:create_and_add_element: Adding tcamdutils(0x5570cdede0f0) to pipeline
0:00:00.031038326 21044 0x5570cde21d20 DEBUG                tcambin gsttcambin.cpp:566:create_and_add_element: Adding videoconvert(0x5570cdee5f80) to pipeline
0:00:00.031129266 21044 0x5570cde21d20 INFO                 tcambin gsttcambin.cpp:1159:gst_tcam_bin_change_state: READY_TO_PAUSED
0:00:00.031147754 21044 0x5570cde21d20 INFO                 tcambin gsttcambin.cpp:1184:gst_tcam_bin_change_state:<tcambin0> caps of sink: video/x-raw, format=(string)BGRx, width=(int)1920, height=(int)1080, framerate=(fraction)30/1
0:00:00.031180608 21044 0x5570cde21d20 INFO                 tcambin gsttcambin.cpp:1188:gst_tcam_bin_change_state:<tcambin0> caps of src: image/jpeg, width=(int)2592, height=(int)1944, framerate=(fraction){ 15/1 }; image/jpeg, width=(int)1920, height=(int)1080, framerate=(fraction){ 30/1 }; image/jpeg, width=(int)1280, height=(int)960, framerate=(fraction){ 60/1 }
0:00:00.031399314 21044 0x5570cde21d20 ERROR                tcambin gsttcambin.cpp:1241:gst_tcam_bin_change_state:<tcambin0> Unable to work with given caps: video/x-raw, format=(string)BGRx, width=(int)1920, height=(int)1080, framerate=(fraction)30/1
0:00:00.031452799 21044 0x5570cde21d20 INFO                 tcambin gsttcambin.cpp:1294:gst_tcam_bin_change_state: Changing state: READY->READY
ERROR: Pipeline doesn't want to pause.
Setting pipeline to NULL ...
0:00:00.031784592 21044 0x5570cde21d20 INFO                 tcambin gsttcambin.cpp:1294:gst_tcam_bin_change_state: Changing state: READY->READY
0:00:00.094885471 21044 0x5570cde21d20 INFO                 tcambin gsttcambin.cpp:1294:gst_tcam_bin_change_state: Changing state: READY->NULL
Freeing pipeline ...
0:00:00.095243884 21044 0x5570cde21d20 DEBUG                tcambin gsttcambin.cpp:1571:gst_tcambin_dispose: dispose
```

TIS-Stefan, 2021-07-07:
There is no error message visible.
Try
export GST_DEBUG:5
(see https://gstreamer.freedesktop.org/documentation/tutorials/basic/debugging-tools.html?gi-language=c)
then run the pipeline.

Did the simple project I sent to you not work?

Stefan

nhi-vanye, 2021-07-07:
[github-412.txt.gz](https://github.com/TheImagingSource/tiscamera/files/6777661/github-412.txt.gz)

I had issues with the simple example around its usage - "no element appsrc", while it would have been a quick fix to validate the camera's interface/capabilities - this is going into a commercial product so I'd rather have something that is supportable by all parties...

TIS-Stefan, 2021-07-07:
If Gstreamer was not installed, then there is no appsrc.....
GStreamer is very common. 
As I wrote, the lib for AFU050 is not related to GStreamer. GStreamer is used to display the received image buffers only. You could have rendered them on your own or in an OpenCV window etc.

You may try again now. 

Stefan

TIS-Stefan, 2021-07-07:
For Gstreamer:
Maybe here https://gstreamer.freedesktop.org/documentation/tutorials/index.html?gi-language=c is a good start.

Stefan

nhi-vanye, 2021-07-07:
Doh! Sorry - juggling multiple systems.

The simple program is now running - yeah!!

TIS-Stefan, 2021-07-07:
Hi

first time, I got a positive answer! I suggest, you stay with the simple program

Stefan

nhi-vanye, 2021-07-08:
Hi Stefan,

Got enough working - can you email me (address on my profile page) - The code comes with no copyright/license so I can't use it/derive from it into a commercial project. 

Thanks.

Richard.

TIS-Stefan, 2021-07-08:
Hi Richard 

It is Open Source and has the same license as our GStreamer modules.

http://www.apache.org/licenses/LICENSE-2.0

Stefan

closed 2021-07-28 by TIS-Stefan