TheImagingSource/tiscamera #413: dfm 25gx236-ml

faccolo, 2021-07-01:
Good morning,
sorry for the questions I'm new to the linux world!

I bought the following Cam:
DFM 25GX236 -ML to use in lan mode with raspberry pi 4.

I followed the steps outlined here:

https://github.com/TheImagingSource/tiscamera/

unfortunately I can not have any image and  video.

if i use this pipeline:
gst-launch-1.0 tcambin ! videoconvert ! ximagesink
only REC my desktop...
if i use 
tcam-capture
....in trigger mode waiting for first image...

[image]

TIS-Stefan, 2021-07-01:
Hello

maybe the Pi is too weak for the high frame rate. Please try lower frame rates and lower resolutions. 

Stefan

faccolo, 2021-07-01:
Thanks for the reply.

i lowered frame rate to 5fps, but still wait for the trigger.
[image]

TIS-Stefan, 2021-07-01:
What are you doing with all these wires?
Did you try 640x480@30fps?
A question:

> if i use this pipeline:
> gst-launch-1.0 tcambin ! videoconvert ! ximagesink
> only REC my desktop...
> 
What does this mean?

Stefan

faccolo, 2021-07-01:
> What are you doing with all these wires?
> Did you try 640x480@30fps?
> A question:

[image]
use  external power supply @12 V

when i use this pipeline 
gst-launch-1.0 tcambin ! videoconvert ! ximagesink

t seems that the terminal freezes
[image]

TIS-Stefan, 2021-07-01:
What happens, if you use the pipeline
gst-launch-1.0 tcambin ! video/x-raw,format=BGRx,width=640,height=480,framerate=30/1 ! videoconvert ! ximagesink
?

Does the green LED lights on the RJ 45 connector of the camera?

I am sorry for guessing around. 

Stefan

faccolo, 2021-07-01:
genius!

I see something ... 

how did you do?

TIS-Stefan, 2021-07-01:
Well, I did nothing, you did. tcam-capture has sometimes an issue with changing video formats and frame rates. In particular it hangs, if no frames are delivered.  That will be fixed in tiscamera 0.15 hopefully.

Now you can enhance the width, height and frame rate until you get no more images.

Stefan

closed 2021-07-01 by TIS-Stefan