TheImagingSource/tiscamera #414: [Feature Request] Release Packages for Ubuntu 20.04

Megamouse, 2021-07-05:
Hi, I'm struggling to build the source with Ubuntu 20.04.
Is it possible to provide packages for the latest release analog to the 18.04 releases?

TIS-Stefan, 2021-07-05:
Hello Megamouse,

Ubuntu 20.04 builds fine on my machine. Which error messages do you receive?

Stefan

Megamouse, 2021-07-05:
I tried on two seperate PCs.

On my current PC I'm getting this during setup:

```
The following packages have unmet dependencies:
 libgirepository1.0-dev : Depends: libgirepository-1.0-1 (= 1.64.0-2) but 1.64.1-1~ubuntu20.04.1 is to be installed
                          Depends: gobject-introspection (= 1.64.0-2) but it is not going to be installed
                          Depends: gir1.2-glib-2.0 (= 1.64.0-2) but 1.64.1-1~ubuntu20.04.1 is to be installed
                          Depends: gir1.2-freedesktop (= 1.64.0-2) but 1.64.1-1~ubuntu20.04.1 is to be installed
 libgstreamer-plugins-base1.0-dev : Depends: libgl1-mesa-dev but it is not going to be installed
                                    Depends: libegl1-mesa-dev but it is not going to be installed
                                    Depends: libgles2-mesa-dev but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
```

This leads to:

```
$ cmake -DBUILD_ARAVIS=OFF -DBUILD_TOOLS=OFF -DBUILD_DOCUMENTATION=OFF ..
CMake Error at /usr/share/cmake-3.16/Modules/FindPkgConfig.cmake:463 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake-3.16/Modules/FindPkgConfig.cmake:643 (_pkg_check_modules_internal)
  cmake/modules/FindGObjectIntrospection.cmake:39 (pkg_check_modules)
  src/gobject/CMakeLists.txt:20 (find_package)
-- Configuring incomplete, errors occurred!
```

Whereas on my other PC also with Ubuntu 20.04 I got another error entirely during compilation:
`tcam_prop_get_type not defined in Tcam-0.1.h`

TIS-Stefan, 2021-07-05:
Hello

"Unable to correct problems, you have held broken packages"
That is a problem on your computer and has no relation to tiscamera, except you can not download the required packages.
You can search in the internet for "Unable to correct problems, you have held broken packages". I did so and found
https://unix.stackexchange.com/questions/389156/how-to-fix-held-broken-packages
it is not related to your issue exactly, but I think, there are hints how to fix your Linux issue of package sources. I hope, this helps fixing your issue.

Stefan

Megamouse, 2021-07-05:
Is it possible though to provide the tiscam packages ?
It's a bit inconvenient if you buy a camera and the only pre-build-packages available are for Ubuntu 18.04

TIS-Stefan, 2021-07-05:
We have currently no 20.04 packages. Ubuntu 18.04 is our reference system.
However, the prebuild packages need the missing packages too, so that would not fix your Linux system issue.
You will encounter more issues with that, e.g. on updates, if you do not fix that issue.

Stefan

Megamouse, 2021-07-06:
It turns out that my entire environment was broken due to one of the dependency install scripts that are recommended in the README.MD. I reinstalled Ubuntu root because it took too much time trying to fix it and manually installed the dependencies listed by `dependency-manager list --compilation`.

closed 2021-07-06 by Megamouse

TIS-Stefan, 2021-07-06:
Hello

Your comment alerted me. I set up a completely new Ubuntu 20.04 in a virtual machine. Then I did
```
https://github.com/TheImagingSource/tiscamera.git
cd tiscamera/
./scripts/dependency-manager install
mkdir build
cd build
cmake -DBUILD_ARAVIS=ON ..
make -j
```
It worked perfectly.  Thus, I do not think, that the dependency-manager broke your system. I rather think, it was broken before, otherwise apt / apt-get wont show an error.  However, I am guessing.

Stefan

Megamouse, 2021-07-06:
Did you try the script instead?

Megamouse, 2021-07-06:
I think I ran:
./scripts/dependency-manager install
And then:
sudo ./scripts/install-dependencies.sh --compilation --runtime

TIS-Stefan, 2021-07-07:
Hello

you did it correctly. I did exactly, what is what I wrote above.  Therefore, I guess your sources list or what ever causes the apt error, was broken before.

Stefan