TheImagingSource/tiscamera #416: Pull the latest acquired image from a camera

Aytoris, 2021-07-20:
Hi,
I am working with a DMK 33UP2000 camera and I'm using a hardware trigger (60 Hz).

I would like to get the latest triggered image to process it.

To get images, I use Gstreamer with Python : the "tcambin" object is the input of my pipeline and an "appsink" is its output. 
Here is my pipeline:
`"tcambin name=source ! capsfilter name=caps ! appsink name=sink"`

The appsink is configured like this:
`        appsink.set_property("max-buffers", 5)`
`        appsink.set_property("drop", 1)`
`        appsink.set_property("emit-signals", 1)`

I tried to get the latest sample by using
`new_sample = appsink.emit("pull-sample")`
but I actually get an old sample, which might come from a buffer, because when I pull a few more samples, I eventually get the right one.

I tried 
`        appsink.set_property("max-buffers", 1)`
but saw no differences.

Is there a better way to get the last sample?
Thank you

TIS-Stefan, 2021-07-20:
I have no real good idea.
You may enhance your pipeline to
`"tcambin name=source ! capsfilter name=caps ! appsink name=sink **sync=false**"`
Also you must make sure, that the camera runs with a frame rate a little bit higher than 60 fps. Please consider, the exposure time sums into delivery time too. You should be totally fine with 100 fps.

Stefan

Aytoris, 2021-07-22:
Hi Stefan, thanks for the tip.
It seems like the "tcam-capture" program succeeds in displayed the lastest image when in trigger mode and manually triggering acquisitions. I tried looking into its code but it's too complex for me right now.

TIS-Stefan, 2021-07-22:
tcam-capture uses a callback function, which is called by the appsink for each image. In the callback is the appsink.emit("pull-sample") executed.
Therefore, the last call of this callback gives the last sample of the last trigger.
A callback function can be written as
```
def on_new_buffer(self, appsink):
        self.newsample = True
        if self.samplelocked is False:
            try:
                self.sample = appsink.get_property('last-sample')
                if self.ImageCallback is not None:
                    self.__convert_sample_to_numpy()
                    self.ImageCallback(self, *self.ImageCallbackData);

            except GLib.Error as error:
                print("Error on_new_buffer pipeline: {0}".format(error))
                raise
        return False
```
The appsink must be informed, that there should be a callback called:
` self.appsink.connect('new-sample', self.on_new_buffer)`

You may have a look into the sample at https://github.com/TheImagingSource/Linux-tiscamera-Programming-Samples/tree/master/Save%20an%20image%20on%20trigger%20in%20Python. I could imagine, that this sample does, what you want to do. If I am right, please apologize, that I did not pointed to this sample earlier.

Stefan

Aytoris, 2021-07-22:
I have tried this code but using `get_property("last-sample")` is actually "slower" than `emit("pull-sample)`: it gives the correct picture 2 pictures earlier than "last-sample". I don't fully get it haha

Aytoris, 2021-07-29:
Okay I figured that the problem did not come from tiscamera. Thank you for your reply!

closed 2021-07-29 by Aytoris

TIS-Stefan, 2021-07-29:
Thank you very much for your feedback. I hope it works for you correct now.