TheImagingSource/tiscamera #418: UTF-8/Unicode character error

mateustormin, 2021-08-06:
I am trying to use tiscamera with a GigE Vision Camera but once I plug the camera on the network interface I get the following error:
[image]

I believe its because of the manufacturer name on the camera
[image]

tcam-gigetools provides me the correct values (I guess because it doesnt try to decode the vendor name:
[image]

am I right? is there anything I can do to work this around?

TIS-Stefan, 2021-08-06:
Hello 

It seems, you do not use a The Imaging Source camera, this I am not able to help

Stefan

mateustormin, 2021-08-06:
Yes sir, you are right, i am not using a The Imaging Source camera, but I am just the developer, R&D brought me this generic camera and I need to make it a product.

Although I did not chose the camera, engineering could use TIS cameras if the application would work, but since you cannot help I guess i'm gonna have to move on with vendor's SDK.

Anyway thanks for your time and stay safe!

closed 2021-08-06 by mateustormin