TheImagingSource/tiscamera #420: No callback on software trigger in C++

mbcel, 2021-08-23:
I am using the tcamcamera.cpp class from the examples to get an image on software trigger. For this I am initializing the camera as follows:
```
std::shared_ptr<gsttcam::Property> exposureProperty = tcamCamera.get_property("Exposure Auto");
    exposureProperty->set(tcamCamera, "Off");
    std::shared_ptr<gsttcam::Property> whitebalanceProperty = tcamCamera.get_property("Whitebalance Auto");
    whitebalanceProperty->set(tcamCamera, "Off");
    std::shared_ptr<gsttcam::Property> gainAutoProperty = tcamCamera.get_property("Gain Auto");
    gainAutoProperty->set(tcamCamera, "Off");
    std::shared_ptr<gsttcam::Property> triggerModeProperty = tcamCamera.get_property("Trigger Mode");
    triggerModeProperty->set(tcamCamera, "On");
```
then I trigger with:
```
std::shared_ptr<gsttcam::Property> triggerProperty = tcamCamera.get_property("Software Trigger");
        triggerProperty->set(tcamCamera, true);
```

and no callback is received. With normal live feed the callback receives images normally. I am using the aravis driver.

TIS-Edgar, 2021-08-24:
Hi,

The code segments you posted appear to be correct.
What camera model are you using? 
Knowing that I can try an reproduce this issue.

Does software trigger work when using tcam-capture?
I am asking because that way we will know that there are no other issues preventing your code from working.

tehkillerbee, 2021-09-14:
Aren't you missing the "Trigger Source" Property? I think it should be set to Software for software triggering.

//g_value_set_string(&str, "Line1");
g_value_set_string(&str, "Software");
cameraInst->set_property("Trigger Source", str);

TIS-Edgar, 2024-04-12:
Closed due to inactivity.