TheImagingSource/tiscamera #425: External trigger makes the camera stuck

P-i-N, 2021-09-22:
I have encountered a weird problem, when using DMK 33UP1300 with external trigger:

When the camera is externally triggered during exposition, it gets "stuck" and never comes back - it has to be physically unplugged and plugged again, which is very annoying.

In other words, our situation is this:
We are using external trigger signal at 200 Hz, so the camera takes an image every 5ms. This works fine as long as the camera exposure time is kept under 5ms. If I set the exposure time to just 6ms or more (knowing it does not make any sense to have exposure longer than triggering period), I get timeout errors and the camera NEVER recovers.

After this happens, almost nothing can be done to make the camera work again:
- enabling/disabling external trigger does not help
- restarting our app/reinitializing/recreating GStreamer pipelines does not help
- `tcam-capture` refuses to show any image (but controls are still visible)
- rebooting the OS does not help (?!?!)

The only thing that works is unplugging and plugging camera USB cable. Is there anything I can do in software to reset the camera and make it work again?

```bash
$ tcam-ctrl --version
Versions: 
        Tcam:   0.14.0_master/031c7762_rev_3054
        Aravis: 0.7_version_031c7762
```

Device used:
```bash
$ tcam-ctrl --list
Model: DMK 33UP1300 Serial: 05120894 Type: v4l2
```

Camera state after it is stuck:
```bash
$ tcam-ctrl -p 05120894
Gamma                                   (int) min=1 max=500 step=1 default=100 value=100 category=Image group=Gamma
Gain                                    (int) min=144 max=1200 step=1 default=144 value=144 category=Exposure group=Gain
Exposure Time (us)                      (int) min=50 max=10000000 step=1 default=333 value=935 category=Exposure group=Exposure
Exposure Auto                           (bool) default=true value=false category=Exposure group=Exposure
Exposure Auto Reference                 (int) min=0 max=255 step=1 default=128 value=128 category=Exposure group=Exposure
Gain Auto                               (bool) default=true value=false category=Exposure group=Gain
Trigger Mode                            (bool) default=false value=false category=Special group=Trigger Mode
Software Trigger                        (button) category=Special group=Trigger Mode
Trigger Delay (us)                      (int) min=0 max=1000000 step=1 default=0 value=159 category=Special group=Trigger Mode
Strobe Enable                           (bool) default=false value=false category=Special group=Strobe Enable
Strobe Polarity                         (bool) default=false value=false category=Special group=Strobe Enable
Strobe Exposure                         (bool) default=false value=false category=Special group=Strobe Enable
GPOut                                   (int) min=0 max=1 step=1 default=0 value=0 category=Special group=GPIO
GPIn                                    (int) min=0 max=1 step=1 default=0 value=1 category=Special group=GPIO
Offset X                                (int) min=0 max=1024 step=16 default=0 value=0 category=Partial Scan group=Offset Auto Center
Offset Y                                (int) min=0 max=1008 step=4 default=0 value=0 category=Partial Scan group=Offset Auto Center
Offset Auto Center                      (bool) default=true value=true category=Partial Scan group=Offset Auto Center
Trigger Polarity                        (enum)  default=Rising Edge category=Special group=Trigger Mode
                                                value=Rising Edge
                                                         Falling Edge
                                                         Rising Edge
Trigger Burst Count                     (int) min=1 max=1000 step=1 default=1 value=1 category=Special group=Trigger Mode
Trigger Debounce Time (us)              (int) min=0 max=1000000 step=1 default=0 value=0 category=Special group=Trigger Mode
Trigger Mask Time (us)                  (int) min=0 max=1000000 step=1 default=0 value=0 category=Special group=Trigger Mode
Trigger Noise Surpression Time (us)     (int) min=0 max=1000000 step=1 default=0 value=0 category=Special group=Trigger Mode
Auto Functions ROI Control              (bool) default=true value=true category=Special group=Auto Functions ROI Control
Auto Functions ROI Left                 (int) min=0 max=1264 step=1 default=0 value=320 category=Special group=Auto Functions ROI Control
Auto Functions ROI Top                  (int) min=0 max=1008 step=1 default=0 value=256 category=Special group=Auto Functions ROI Control
Auto Functions ROI Width                (int) min=16 max=1280 step=1 default=0 value=640 category=Special group=Auto Functions ROI Control
Auto Functions ROI Height               (int) min=16 max=1024 step=1 default=0 value=512 category=Special group=Auto Functions ROI Control
Exposure Auto Upper Limit Auto          (bool) default=true value=true category=Exposure group=Exposure
Exposure Auto Upper Limit               (int) min=50 max=1000000 step=1 default=333 value=4761 category=Exposure group=Exposure
Override Scanning Mode                  (int) min=1 max=16 step=1 default=1 value=0 category=Partial Scan group=Override Scanning Mode
Auto Functions ROI Preset               (enum)  default=Center 50% category=Special group=Auto Functions ROI Control
                                                value=Center 50%
                                                         Bottom Half
                                                         Center 25%
                                                         Center 50%
                                                         Custom Rectangle
[... 82 lines cut ...]
2021.09.21T17:13:16:782        <DEBUG> V4l2Device.cpp:1593: Queueing buffer(0x7f85da8010) with length 1310720
2021.09.21T17:13:16:782        <DEBUG> V4l2Device.cpp:1602: Successfully queued v4l2_buffer
2021.09.21T17:13:16:782        <DEBUG> V4l2Device.cpp:1593: Queueing buffer(0x7f85c67010) with length 1310720
2021.09.21T17:13:16:783        <DEBUG> V4l2Device.cpp:1602: Successfully queued v4l2_buffer
2021.09.21T17:13:16:783        <DEBUG> V4l2Device.cpp:1593: Queueing buffer(0x7f85b26010) with length 1310720
2021.09.21T17:13:16:784        <DEBUG> V4l2Device.cpp:1602: Successfully queued v4l2_buffer
2021.09.21T17:13:16:784        <DEBUG> V4l2Device.cpp:1593: Queueing buffer(0x7f859e5010) with length 1310720
2021.09.21T17:13:16:784        <DEBUG> V4l2Device.cpp:1602: Successfully queued v4l2_buffer
2021.09.21T17:13:16:784        <DEBUG> V4l2Device.cpp:1593: Queueing buffer(0x7f858a4010) with length 1310720
2021.09.21T17:13:16:785        <DEBUG> V4l2Device.cpp:1602: Successfully queued v4l2_buffer
2021.09.21T17:13:16:786        <DEBUG> V4l2Device.cpp:507: Setting stream timeout to 2
2021.09.21T17:13:16:786        <INFO> V4l2Device.cpp:529: Starting stream in work thread.
2021.09.21T17:13:16:786        <DEBUG> ImageSource.cpp:57: PLAYING....
2021.09.21T17:13:16:786        <INFO> PipelineManager.cpp:83: All pipeline elements set to PLAYING.
2021.09.21T17:13:20:789        <ERROR> V4l2Device.cpp:1459: Timeout while waiting for new image buffer.
2021.09.21T17:13:24:793        <ERROR> V4l2Device.cpp:1459: Timeout while waiting for new image buffer.
2021.09.21T17:13:28:796        <ERROR> V4l2Device.cpp:1459: Timeout while waiting for new image buffer.
2021.09.21T17:13:32:801        <ERROR> V4l2Device.cpp:1459: Timeout while waiting for new image buffer.
2021.09.21T17:13:36:805        <ERROR> V4l2Device.cpp:1459: Timeout while waiting for new image buffer.
2021.09.21T17:13:36:805        <WARNING> V4l2Device.cpp:1480: Did not receive image for long time.
2021.09.21T17:13:40:809        <ERROR> V4l2Device.cpp:1459: Timeout while waiting for new image buffer.
2021.09.21T17:13:44:813        <ERROR> V4l2Device.cpp:1459: Timeout while waiting for new image buffer.
2021.09.21T17:13:48:815        <ERROR> V4l2Device.cpp:1459: Timeout while waiting for new image buffer.
... timeouts keep going forever
```

TIS-Stefan, 2021-09-22:
Hello!
 
What you describe can happen, therefore, I suggest two things: Use a frame rate above 200fps, because the image exposure time and the delivery time sum up in trigger mode. 

Secondly you may set the "Trigger Mask Time" property to 5ms, so the camera wont accept any triggers, that come not in 5ms interval.

I hope, this helps.

P-i-N, 2021-09-22:
Thank you, Stefan, for super fast reply! :) I will try changing the trigger mask time and see, if it helps.

P-i-N, 2021-09-22:
OK, I have tried adjusting the trigger mask time, but the problem persists.

At first, I thought the issue was caused by my GStreamer init/setup, but after few tests, I've noticed it's present also when using V4L2 backend. It can be reproduced just using `tcam-capture` as well.

From my observation, it looks like the camera really hates when another trigger signal comes during image exposure. It puts the camera into some limp mode and it's impossible to use afterwards. `tcam-capture` then shows "In Trigger Mode. Waiting for first image..." forever.

Since the only known working solution for me is unplugging/plugging USB power, I ended up doing exactly that. So, if anyone encounters this on nVidia Jetson Nano, this might be helpful: [https://forums.developer.nvidia.com/t/usb-power-control/79037/7](https://forums.developer.nvidia.com/t/usb-power-control/79037/7)

[realtek_hub_power_cycle.c.txt](https://github.com/TheImagingSource/tiscamera/files/7211363/realtek_hub_power_cycle.c.txt)

TIS-Stefan, 2021-09-22:
Which value do use for trigger mask time? If I remember correctly, its unit is µs. so you may set it to 5000 
Did you also enhance the frame rate? exposure time + 1/framerate must fit into your trigger frequency.

P-i-N, 2021-09-22:
When I set the trigger mask time to 5000, I was missing every other frame - which seems to be correct behaviour, since every second trigger signal was ignored. Then, setting exposure time to 10ms (10000us) made the camera stuck again.

I then decreased the trigger mask time to 4500, which gave me steady 200 fps image stream, but the problem persisted.

Exposure times are usualy between 1000us and 3000us. FPS is set to "210/1".

Also, if I start randomly switching FPS in `tcam-capture` from "210/1" to "1/1" and back, it makes the camera stuck as well even when the exposure is less than 5ms and I have to unplug/plug. (I know that's probably wrong parameter usage combined with external triggerering, but I would assume the camera can handle this and recover)

P-i-N, 2021-09-22:
The trigger signal is generated by specialized HW device and can't be changed. But I will try to make some fake trigger source using Arduino and see, if I get the same result even with lower frequencies.

Another observation, just for info:
Randomly disconnecting and reconnecting just the trigger wires at the HW source, the camera hangs after few tries. Exposure time was 1000us (autoexposure disabled), external triggering enabled, trigger mask time was 4500us (but happened with 0us as well), FPS "210/1", only using `tcam-capture`.

TIS-Stefan, 2021-09-23:
> Randomly disconnecting and reconnecting just the trigger wires at the HW source, the camera hangs after few tries. Exposure time was 1000us (autoexposure disabled), external triggering enabled, trigger mask time was 4500us (but happened with 0us as well), FPS "210/1", only using tcam-capture.

Is it possible, that the trigger signal is not clear and that you get during connect and disconnect trigger signals in due to bouncing?
I must admit, until now I saw not so much issues with triggering, as you report. Did you check your trigger signal with an oscilloscope?

P-i-N, 2021-09-23:
OK, I am back with more results! :)

I changed the 200 Hz trigger signal to 10 Hz trigger signal, meaning the trigger period was 100ms. Everything worked fine as long as the exposure time was less than 100ms. Once I set the exposure to 100ms or more, the camera got stuck again, like a clockwork!

Also, here is how the 200 Hz trigger signal looks on oscilloscope:

[image]

When the trigger cable is connected (we are using RJ12 connector on the source side):

[image]

... and when disconnected:

[image]

Bouncing when connecting the cable might be an issue, but I would still expect the camera to recover from it after the signal stabilizes. And more importantly - bouncing does not really explain why the camera gets stuck, when exposure time is longer than trigger period.

TIS-Stefan, 2021-09-23:
Hello

First of all, I must apologize for wasting your time. I checked on my own today with firmware 781, which is the one in your camera. I could reproduce your issue immediately. The firmware, which is currently in release test does not expose this issue. If the camera is overwhelmed with triggers, then these triggers are simply ignored and nothing happens, except the number of frames you will receive may goes down. How far it goes down seems to depend on the trigger frequency : framerate ratio. If I set 210 fps I can trigger with 1000Hz and get 200 fps. However, triggering with 200Hz gives the wanted result.

Please create a contact at https://www.theimagingsource.com/support/contact/, so I can send my test firmware to you. The firmware is still in release test and therefore not released for the production line by now.

Stefan

P-i-N, 2021-09-23:
No worries and thank you very very much, Stefan! :) I am glad it got resolved after all!

I will contact you through the contact form.

hidon, 2022-05-25:
Hello, problem is solved?

TIS-Stefan, 2022-05-25:
No, the problem is not solved.

Zion-Go, 2023-06-21:
> Stefan
Hello. 

Can I ask DFK 33UX250 has the same firmware and problems? 

Thank you.

TIS-Stefan, 2023-06-21:
Hello
The firmware and hardware are different. However, after nearly two years I am completely lost, so I can not answer the second question. 

Stefan

TIS-Stefan, 2023-06-21:
You cameras seem to have the current 1086 firmware. Do you encounter this issue?

Zion-Go, 2023-06-21:
> You cameras seem to have the current 1086 firmware. Do you encounter this issue?

Hello! Stefan. 

My issues are different. I'm using 2 cameras as I said in the request, 45020167, 45020169.

I currently have the issue like this. I'm using ROS2 to publish hardware synchronized stereo imgs. When I launched 2 cameras, sometimes it can receive imgs from both cameras. Sometimes it can only receive imgs from one cameras.

And I also need to unplug the USB cables to reinitiate the cameras to start the pipeline. Or we can't continuously start the cameras again. Then the above conditions could repeat.. Currently just cannot debug the issues. 

Thank you!

TIS-Stefan, 2023-06-21:
Hello
So it is a completely different issue. I already asked an engineer to have a look. 

Did you take care of the frame rates? If both cameras are connected to the same USB 3 controller of your desktop computer, then you must reduce the frame rates, other the USB bandwidth is exhausted. If 75fps is the maximum frame rate, then you should set the frame rate to 30/1 for both cameras. Then there is enough bandwidth. Accordingly you must trigger a little bit less then 30Hz

Stefan

Zion-Go, 2023-06-21:
> Hello
> So it is a completely different issue. I already asked an engineer to have a look.
>
> Did you take care of the frame rates? If both cameras are connected to the same USB 3 controller of your [...]

Sorry for the misleading. 
For the frame rates, I set them low as 15/1 for both actually. But I'm using Jetson agx xavier type C port with a USB 3. 0 hub to connect cameras. Can type C port be a problem for bandwidth?

TIS-Stefan, 2023-06-21:
The C Port should not be an issue. 
What is your trigger frequency?

TIS-Stefan, 2023-06-21:
Can you please run 
lsusb -t 
and show me the output.

Zion-Go, 2023-06-21:
> The C Port should not be an issue. 
> What is your trigger frequency? 

I saw your comments about exposure time and 1/framerate should fit into the trigger frequency.
My trigger frequency is 0.08+1/2=0.58..

TIS-Stefan, 2023-06-21:
Hello 

The USB setting is perfectly ok, as expected.
My understanding is also, that the trigger frequency is fitting well too. 
So the issue is, you do not get images from both cameras, if you trigger them. In your setup, the cameras will send an image for sure. But it may happens, that there are packet losts and then tiscamera will drop incomplete frames. 
I must say, I am lost. In order to narrow down the issue, I would start with lower frame rates and trigger frequencies, just to see the reliability of image data transfer. Then I would rise the frame rates and trigger frequency in order to see, whether the image data transfer becomes unreliable.

I assume, you use standard USB 3 cables of 3m length maximum. 
You could check with lspci which USB Controller chips are in use. Only ASMedia could make trouble in some cases. 

I am sorry for not having a better answer. 

Stefan

Zion-Go, 2023-06-21:
> Hello
>
> The USB setting is perfectly ok, as expected.
> My understanding is also, that the trigger frequency is fitting well too.
> So the issue is, you do not get images from both cameras, if you [...]

Hello! 

The routine and advice that you would like to conduct are quite helpful already. I will keep on. 

Thank you for your time. 

Best regards, 
Zion

TIS-Edgar, 2024-04-12:
Closed due to inactivity.

closed 2024-04-12 by TIS-Edgar