TheImagingSource/tiscamera #426: gstreamer videoscale support

ben00052, 2021-09-23:
I am currently running into a problem with DFK27AUJ003 cameras while using gstreamer 1.10.5 on a fedora 25/KDE system.  Yes I know I am kind of out of date.  I believe I am still on tiscamera 0.8.0, 0.14.0 does not detect the camera when running tcam-ctrl -l, most likely due to firmware.

When running:
gst-launch-1.0 v4l2src ! video/x-bayer,width=3872,height=2764,framerate=3/1 ! bayer2rgb ! videoscale ! videoconvert ! ximagesink
everything works fine and I get an image I can resize without anything getting cropped off.

When running:
gst-launch-1.0 tcamsrc ! video/x-bayer,width=3872,height=2764,framerate=3/1 ! bayer2rgb ! videoscale ! videoconvert ! ximagesink
I get no image and an error:
ERROR: from element /GstPipeline:pipeline0/GstVideoScale:videoscale0: GStreamer error: negotiation problem.
Additional debug info:
gstvideoscale.c(730): gst_video_scale_fixate_caps (): /GstPipeline:pipeline0/GstVideoScale:videoscale0:
Error calculating the output scaled size - integer overflow

It works fine without the videoscale argument however.  

I would like to get scaling of some sort working side by side with the auto white balance, but getting the full image on a smaller screen is more important than the color adjustment.  If there is a way to enable the auto white balance outside of the gstreamer constructor, that would work great, or an alternate way to get a full uncropped image into a smaller resolution.

TIS-Stefan, 2021-09-24:
Hello

Unfortunately it is very hard for me to support the 0.8.0 version. But I would like to suggest to switch videoconvert and videoscale. Then videoscale should receive the format it would like to have. At least, it is worth a try.

Stefan

ben00052, 2021-09-24:
No luck swapping around convert and sample, but I appreciate the suggestion.  Next up, seeing if I can work around using the bayer2rgb.

TIS-Stefan, 2021-09-24:
Unfortunatelly, you can not. 

Is it possible to build tiscamera from the repository on your system? Then you can use the current version and the error should be gone. 
Otherwise, if v4l2src is sufficient, you can stay with that

Stefan

ben00052, 2021-09-24:
I did build 14, but it would not recognize the camera when I ran tcam-ctrl -l.  Back in https://github.com/TheImagingSource/tiscamera/issues/352 it was my understanding it had to do with the firmware on the camera itself, although re-reading it I am second guessing my memory now.

TIS-Stefan, 2021-09-24:
Well, the camera is detected by hardware id, as far as I know and that is firmware independent. Please let me know the serial number of the camera, so I will check its firmware version. Maybe there is a relation to firmware version.

ben00052, 2021-09-24:
26120624 and 17714160
Also of note, it is not giving me "no device found" either, like 0.8.0 did when nothing was plugged in. 
--help works fine, and -c 26120624 tells me "device with given serial number does not exist"

TIS-Stefan, 2021-09-24:
The camera 26120624 is from this year, it should be detected anyway.
I suppose, both cameras are shown in dmesg output, right?
Which kernel do you use?  Maybe its version is the reason. 
So you may have to stay with 0.8.0 and try to find a way around bayer2rgb.

ben00052, 2021-09-24:
I had not checked dmesg before, but yes, it shows up , with SN, in that output.

ben00052, 2021-09-24:
Looks like the kernel is 4.13.16-100.fc25.x86_64.  
Do you know when the error was fixed?  I might be able to get an earlier release than 14 but newer than 8 running on the system

TIS-Stefan, 2021-09-24:
Hello 
4.13 should be fine, if I remember correctly. Thus I am puzzled. 
You may do
export TCAM_LOG=DEBUG
tcam-ctrl -l
and see the output. Maybe it give some information.

Stefan

ben00052, 2021-09-27:
No additional debugging information with export TCAM_LOG=DEBUG.  I it will be a while before I can try upgrading the OS, so for the moment it looks like v4l2src is it.

ben00052, 2021-10-06:
Ok, I am trying to get 14 built on a Fedora 34 system, but I am running into a problem building it i did not see back on 25.
`
Consolidate compiler generated dependencies of target gsttcammainsrc
[ 48%] Building CXX object src/gstreamer-1.0/CMakeFiles/gsttcammainsrc.dir/gstmetatcamstatistics.cpp.o
In file included from /usr/include/glib-2.0/glib/gthread.h:32,
                 from /usr/include/glib-2.0/glib/gasyncqueue.h:32,
                 from /usr/include/glib-2.0/glib.h:32,
                 from /usr/include/gstreamer-1.0/gst/gst.h:27,
                 from /home/user/tiscamera-v-tiscamera-0.14.0/src/gstreamer-1.0/gstmetatcamstatistics.h:22,
                 from /home/user/tiscamera-v-tiscamera-0.14.0/src/gstreamer-1.0/gstmetatcamstatistics.cpp:17:
/home/user/tiscamera-v-tiscamera-0.14.0/src/gstreamer-1.0/gstmetatcamstatistics.cpp: In function â€˜GType tcam_statistics_meta_api_get_type()â€™:
/usr/include/glib-2.0/glib/gatomic.h:117:19: error: argument 2 of â€˜__atomic_loadâ€™ must not be a pointer to a â€˜volatileâ€™ type
  117 |     __atomic_load (gapg_temp_atomic, &gapg_temp_newval, __ATOMIC_SEQ_CST); \
      |     ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/glib-2.0/glib/gthread.h:260:7: note: in expansion of macro â€˜g_atomic_pointer_getâ€™
  260 |     (!g_atomic_pointer_get (location) &&                             \
      |       ^~~~~~~~~~~~~~~~~~~~
/home/user/tiscamera-v-tiscamera-0.14.0/src/gstreamer-1.0/gstmetatcamstatistics.cpp:25:9: note: in expansion of macro â€˜g_once_init_enterâ€™
   25 |     if (g_once_init_enter(&type))
      |         ^~~~~~~~~~~~~~~~~
make[2]: *** [src/gstreamer-1.0/CMakeFiles/gsttcammainsrc.dir/build.make:90: src/gstreamer-1.0/CMakeFiles/gsttcammainsrc.dir/gstmetatcamstatistics.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:665: src/gstreamer-1.0/CMakeFiles/gsttcammainsrc.dir/all] Error 2
make: *** [Makefile:156: all] Error 2
`

ben00052, 2021-10-06:
It looks like glib2-devel is 2.68.4, for what it is worth.

TIS-Stefan, 2021-10-07:
Hello

According to  https://github.com/vmware/open-vm-tools/issues/509 it is an issue between g++ 11 and glib2>=2.66. Which compiler do you use?
```
cd tiscamera/build
cmake -LAH | grep CXX <= want output
CMAKE_CXX_COMPILER:FILEPATH <= this will point to compiler
```
execute
<compiler from previous command with absolute path> --version <= want output

Unfortunately we can not test on own, because we simple do not have any Fedora installations. 

Stefan

ben00052, 2021-10-08:
That looks like it should be the last thing I need on your end for this project.  Thank you, hopefully I can go a few more years before I need to bother you again.