TheImagingSource/tiscamera #435: ModuleNotFoundError: No module named 'tcam_capture.TcamDevice'

seymseym, 2021-11-02:
I can not open the tcam application.
Camera is working with this command:
`gst-launch-1.0 tcambin ! videoconvert ! ximagesink`

But nothing happens when I click on the tcam-capture icon on the applications.
And when I try to use tcam-capture command on cmd;

I get this error:
```
root@upboard:/home/upboard/tiscamera/build# tcam-capture
Traceback (most recent call last):
  File "/usr/bin/tcam-capture", line 28, in <module>
    from tcam_capture.TcamDevice import TcamDevice, TcamDeviceIndex
ModuleNotFoundError: No module named 'tcam_capture.TcamDevice'
```

TIS-Stefan, 2021-11-02:
Did you run "sudo make install"?

seymseym, 2021-11-02:
Yes, I did every step one by one carefully

TIS-Stefan, 2021-11-02:
Hello

Maybe the Python paths are changed in your system. Please check the output of
`python3 -c "import sys; print(sys.path)"`

It should be something lile
`['', '/usr/lib/python38.zip', '/usr/lib/python3.8', '/usr/lib/python3.8/lib-dynload', '/home/user/.local/lib/python3.8/site-packages', '/usr/local/lib/python3.8/dist-packages', '/usr/lib/python3/dist-packages']`

TIS-Stefan, 2021-11-02:
... and also please show the output of 
`lsb_release -a`

It should similar to 
```
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.3 LTS
Release:	20.04
Codename:	focal

```

seymseym, 2021-11-03:
The python path:

```
root@upboard:/home/upboard# python3 -c "import sys; print(sys.path)"
['', '/usr/lib/python36.zip', '/usr/lib/python3.6', '/usr/lib/python3.6/lib-dynload', '/usr/local/lib/python3.6/dist-packages', '/usr/lib/python3/dist-packages']
```
How to arrange the true path? 

```
root@upboard:/home/upboard# lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.6 LTS
Release:	18.04
Codename:	bionic
```

TIS-Stefan, 2021-11-03:
In the build directory should be a file "install_manifest.txt". This is also needed.

ElTinmar, 2021-11-10:
Had the same issue, turned out all the tcam-capture .py files were in the anaconda site-packages folder (I compiled within the base anaconda environment). I moved them all to '/home/user/.local/lib/python3.8/site-packages' and it worked.

vinit13792, 2022-02-08:
I am facing the same issue.

my python paths are:
`ubuntu-pc@ubuntupc-ThinkStation-P720:~/tiscamera/build$ python3 -c "import sys; print(sys.path)"
['', '/usr/lib/python38.zip', '/usr/lib/python3.8', '/usr/lib/python3.8/lib-dynload', '/home/ubuntu-pc/.local/lib/python3.8/site-packages', '/usr/local/lib/python3.8/dist-packages', '/usr/lib/python3/dist-packages']`

the output of lsb_release -a:
`ubuntu-pc@ubuntupc-ThinkStation-P720:~/tiscamera/build$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.3 LTS
Release:	20.04
Codename:	focal
`
What works for me?
The camera window opens with video stream with the following gstreamer code:
`gst-launch-1.0 tcambin ! video/x-raw,format=BGRx ! videoconvert ! ximagesink`

What doesn't work for me?
`ubuntu-pc@ubuntupc-ThinkStation-P720:~/tiscamera/build$ tcam-capture
Traceback (most recent call last):
  File "/usr/bin/tcam-capture", line 28, in <module>
    from tcam_capture.TcamDevice import TcamDevice, TcamDeviceIndex
ModuleNotFoundError: No module named 'tcam_capture'`

TIS-Stefan, 2022-02-08:
Hello

I ran the commands you listed and I get exactly the same results regarding paths and versions. Only difference: tcam-capture runs.
Please look into the file tiscamera/build/install_manifest.txt. It contains
```
/usr/bin/tcam-capture
/usr/lib/python3/dist-packages/tcam_capture/ImageProvider.py
/usr/lib/python3/dist-packages/tcam_capture/MediaSaver.py
/usr/lib/python3/dist-packages/tcam_capture/ViewItem.py
/usr/lib/python3/dist-packages/tcam_capture/TcamSlider.py
/usr/lib/python3/dist-packages/tcam_capture/FileNameGenerator.py
/usr/lib/python3/dist-packages/tcam_capture/TcamCaptureData.py
/usr/lib/python3/dist-packages/tcam_capture/DeviceDialog.py
/usr/lib/python3/dist-packages/tcam_capture/PropertyDialog.py
/usr/lib/python3/dist-packages/tcam_capture/TcamDevice.py
/usr/lib/python3/dist-packages/tcam_capture/TcamSpinBox.py
/usr/lib/python3/dist-packages/tcam_capture/ResizeableRectItem.py
/usr/lib/python3/dist-packages/tcam_capture/TcamView.py
/usr/lib/python3/dist-packages/tcam_capture/TcamSignal.py
/usr/lib/python3/dist-packages/tcam_capture/CapsDesc.py
/usr/lib/python3/dist-packages/tcam_capture/TcamScreen.py
/usr/lib/python3/dist-packages/tcam_capture/Encoder.py
/usr/lib/python3/dist-packages/tcam_capture/PropertyWidget.py
/usr/lib/python3/dist-packages/tcam_capture/Settings.py
/usr/lib/python3/dist-packages/tcam_capture/OptionsDialog.py
/usr/lib/python3/dist-packages/tcam_capture/Cache.py
/usr/lib/python3/dist-packages/tcam_capture/FPSCounter.py
/usr/share/applications/tcam-capture.desktop
/usr/lib/python3/dist-packages/tcam_capture/image_dir.py
```
Please check, whether the files exist in the give path.

Do you use Anaconda or pythonbox(?)? It does not look like that.

Stefan

vinit13792, 2022-02-08:
Hi Mr. Stefan,

I tried the following and it worked:
`ubuntu-pc@ubuntupc-ThinkStation-P720:~/tiscamera/build$ . ./env.sh`

It just needed me to append directories to the PATH and library search path for the dynamic linker and GStreamer, thus enabling usage of tiscamera resources and additional commands like tcam-ctrl and tcam-capture

vinit13792, 2022-02-09:
> Can you please show the tiscamera/build/install_manifest.txt? Thank you.

`/usr/share/theimagingsource/tiscamera//examples/python/00-list-devices.py`
`/usr/share/theimagingsource/tiscamera//examples/python/11-device-state.py`
`/usr/share/theimagingsource/tiscamera//examples/python/09-device-lost.py`
`/usr/share/theimagingsource/tiscamera//examples/python/07-appsink.py`
`/usr/share/theimagingsource/tiscamera//examples/python/01-list-properties.py`
`/usr/share/theimagingsource/tiscamera//examples/python/06-softwaretrigger.py`
`/usr/share/theimagingsource/tiscamera//examples/python/03-live-stream.py`
`/usr/share/theimagingsource/tiscamera//examples/python/02-set-properties.py`
`/usr/share/theimagingsource/tiscamera//examples/python/08-save-stream.py`
`/usr/share/theimagingsource/tiscamera//examples/python/05-set-format.py`
`/usr/share/theimagingsource/tiscamera//examples/python/04-list-formats.py`
`/usr/share/theimagingsource/tiscamera//examples/README.md`
`/usr/share/theimagingsource/tiscamera//examples/c/04-list-formats.c`
`/usr/share/theimagingsource/tiscamera//examples/c/07-appsink.c`
`/usr/share/theimagingsource/tiscamera//examples/c/Makefile`
`/usr/share/theimagingsource/tiscamera//examples/c/02-set-properties.c`
`/usr/share/theimagingsource/tiscamera//examples/c/11-device-state.c`
`/usr/share/theimagingsource/tiscamera//examples/c/08-save-stream.c`
`/usr/share/theimagingsource/tiscamera//examples/c/05-set-format.c`
`/usr/share/theimagingsource/tiscamera//examples/c/09-device-lost.c`
`/usr/share/theimagingsource/tiscamera//examples/c/00-list-devices.c`
`/usr/share/theimagingsource/tiscamera//examples/c/06-softwaretrigger.c`
`/usr/share/theimagingsource/tiscamera//examples/c/01-list-properties.c`
`/usr/share/theimagingsource/tiscamera//examples/c/03-live-stream.c`
`/usr/share/theimagingsource/tiscamera//examples/c/10-metadata.c`
`/usr/lib/libtcam.so.0.14.0`
`/usr/lib/libtcam.so.0`
`/usr/lib/libtcam.so`
`/usr/lib/libtcamprop.so.0.1`
`/usr/lib/libtcamprop.so`
`/usr/share/gir-1.0/Tcam-0.1.gir`
`/usr/lib/x86_64-linux-gnu/girepository-1.0/Tcam-0.1.typelib`
`/usr/include/tcamprop.h`
`/usr/include/Tcam-0.1.h`
`/usr/lib/libtcamgstbase.so`
`/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgsttcammainsrc.so`
`/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgsttcamsrc.so`
`/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgsttcamwhitebalance.so`
`/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgsttcamautoexposure.so`
`/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgsttcamautofocus.so`
`/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgsttcambin.so`
`/usr/include/gstreamer-1.0/gstmetatcamstatistics.h`
`/usr/lib/libtcam-algorithms.so.0.14.0`
`/usr/lib/libtcam-algorithms.so.0`
`/usr/lib/libtcam-algorithms.so`
`/usr/lib/libtcam-dfk73.so.0.14.0`
`/usr/lib/libtcam-dfk73.so.0`
`/usr/lib/libtcam-dfk73.so`
`/usr/lib/libtcam-uvc-extension.so`
`/usr/lib/libtcam-network.so.0.14.0`
`/usr/lib/libtcam-network.so.0`
`/usr/lib/libtcam-network.so`
`/usr/lib/pkgconfig/tcam.pc`
`/usr/share/theimagingsource/tiscamera/images/23uc256.png`
`/usr/share/theimagingsource/tiscamera/images/tiscamera_logo.png`
`/usr/share/theimagingsource/tiscamera/images/tis_logo_small.jpg`
`/usr/share/theimagingsource/tiscamera/images/tiscamera_favicon_32x32.ico`
`/usr/share/theimagingsource/tiscamera/images/favicon_16x16.png`
`/usr/share/theimagingsource/tiscamera/images/tis_logo.jpg`
`/usr/share/theimagingsource/tiscamera/images/tiscamera_logo_small.png`
`/lib/systemd/system/gige-daemon.service`
`/usr/share/theimagingsource/tiscamera/uvc-extension/usb2.json`
`/usr/share/theimagingsource/tiscamera/uvc-extension/usb23.json`
`/usr/share/theimagingsource/tiscamera/uvc-extension/usb33.json`
`/usr/share/theimagingsource/tiscamera/uvc-extension/usb37.json`
`/etc/udev/rules.d/80-theimagingsource-cameras.rules`
`/usr/bin/tcam-ctrl`
`/usr/bin/tcam-capture`
`/usr/lib/python3.8/site-packages/tcam_capture/TcamDevice.py`
`/usr/lib/python3.8/site-packages/tcam_capture/PropertyWidget.py`
`/usr/lib/python3.8/site-packages/tcam_capture/Encoder.py`
`/usr/lib/python3.8/site-packages/tcam_capture/DeviceDialog.py`
`/usr/lib/python3.8/site-packages/tcam_capture/TcamSlider.py`
`/usr/lib/python3.8/site-packages/tcam_capture/ImageProvider.py`
`/usr/lib/python3.8/site-packages/tcam_capture/TcamCaptureData.py`
`/usr/lib/python3.8/site-packages/tcam_capture/CapsDesc.py`
`/usr/lib/python3.8/site-packages/tcam_capture/Settings.py`
`/usr/lib/python3.8/site-packages/tcam_capture/TcamSignal.py`
`/usr/lib/python3.8/site-packages/tcam_capture/FileNameGenerator.py`
`/usr/lib/python3.8/site-packages/tcam_capture/Cache.py`
`/usr/lib/python3.8/site-packages/tcam_capture/ViewItem.py`
`/usr/lib/python3.8/site-packages/tcam_capture/FPSCounter.py`
`/usr/lib/python3.8/site-packages/tcam_capture/TcamView.py`
`/usr/lib/python3.8/site-packages/tcam_capture/MediaSaver.py`
`/usr/lib/python3.8/site-packages/tcam_capture/ResizeableRectItem.py`
`/usr/lib/python3.8/site-packages/tcam_capture/TcamSpinBox.py`
`/usr/lib/python3.8/site-packages/tcam_capture/TcamScreen.py`
`/usr/lib/python3.8/site-packages/tcam_capture/PropertyDialog.py`
`/usr/lib/python3.8/site-packages/tcam_capture/OptionsDialog.py`
`/usr/share/applications/tcam-capture.desktop`
`/usr/lib/python3.8/site-packages/tcam_capture/image_dir.py`
`/usr/bin/dfk73udev`
`/usr/bin/tcam-uvc-extension-loader`
`/usr/bin/camera-ip-conf`
`/usr/bin/gige-daemon`
`/usr/lib/libtcam_gigewrapper.so`
`/usr/lib/python3.8/site-packages/tcamgigetool/tcam_gigetool.py`
`/usr/lib/python3.8/site-packages/tcamgigetool/__main__.py`
`/usr/lib/python3.8/site-packages/tcamgigetool/__init__.py`
`/usr/lib/python3.8/site-packages/tcamgigetool/controller.py`
`/usr/lib/python3.8/site-packages/tcamgigetool/version.py`
`/usr/bin/tcam-gigetool`

TIS-Stefan, 2022-02-09:
Hello

Thank you for the file.
You posted following Python3 paths:
```
['',
'/usr/lib/python38.zip',
'/usr/lib/python3.8',
'/usr/lib/python3.8/lib-dynload',
'/home/ubuntu-pc/.local/lib/python3.8/site-packages',
'/usr/local/lib/python3.8/dist-packages',
'/usr/lib/python3/dist-packages']
```
For some unkown reason, tcam-capture was installed into 
```
/usr/lib/python3.8/site-packages/
```
Please do not ask, why this happens, I simply do not know. 

You can workaround with
```
export PYTHONPATH="/path/to/add:$PYTHONPATH"
```
Alternatively you can

```
cmake -DTCAM_INSTALL_PYTHON3_MODULES=/usr/lib/python3.8/ ..
```

Stefan

vinit13792, 2022-02-15:
> Hello
>
> Thank you for the file. You posted following Python3 paths:
>
> ```
> ['',
> '/usr/lib/python38.zip',
> '/usr/lib/python3.8',
> '/usr/lib/python3.8/lib-dynload', [...]

Thank you Mr. Stefan! It's working now.

closed 2022-02-15 by TIS-Stefan