TheImagingSource/tiscamera #436: GStreamer: pipeline have not been created

seymseym, 2021-11-03:
The pipelines I create on the OpenCV are not working.

const char *pipeline = " tcambin serial=40120113 ! video/x-raw, format=BGRx, width=640,height=480, framerate=60/1 ! videoconvert ! appsink";

I am sure about the serial number of the camera and `gst-launch-1.0 tcambin ! videoconvert ! ximagesink` works fine on the command line.

I get `opencv-master/modules/videoio/src/cap_gstreamer.cpp (616) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
`
Error

TIS-Stefan, 2021-11-03:
Hello 

the test with gst-launch-1.0 is a very good approach. You could have used

`gst-launch-1.0 tcambin serial=40120113 ! video/x-raw, format=BGRx, width=640,height=480, framerate=60/1 ! videoconvert ! ximagesink
`
In order to make sure, whether the serial number and video format is correct.  It works on my computer with a DFK 27U30135.
Also, "tcam-ctrl -l" shows the cameras with serial numbers and "tcam-ctrl -c <serial>" shows the available resolutions and frame rates.

Please show your code that creates the cv::videocapture object. I suppose it quite simple. 

Stefan
P.S.  You can remove the videoconvert in the pipeline passed to cv::videocapture()

seymseym, 2021-11-03:
Here are the pipeline and the videocapture object:
```
    const char *pipeline = "gst-launch-1.0 tcambin serial=40120113 ! video/x-raw, format=BGRx, width=640,height=480, framerate=60/1 ! videoconvert ! ximagesink";

    cv::VideoCapture *cap = new cv::VideoCapture(*pipeline, cv::CAP_GSTREAMER);
```
I still get the same errors. Maybe it's because the code below?

void VideoStreamer::openVideoCamera()
{
   cap->open(0);
   double fps = cap->get(cv::CAP_PROP_FPS);
   tUpdate.start(1000/fps);
}

**Note:** By using  v4l2-ctl --list-devices, I am sure that video0 exists, so open(0) should not be wrong then, right?

```
GStreamer warning: Embedded video playback halted; module v4l2src0 reported: Cannot identify device '/dev/video103'.
[ WARN:0] global /home/upboard/opencv-master/modules/videoio/src/cap_gstreamer.cpp (1053) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global /home/upboard/opencv-master/modules/videoio/src/cap_gstreamer.cpp (616) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
```
Even though video0 is the right one for the camera, I get a silly error like video103 not exists.. 
And pipeline error is still there.

TIS-Stefan, 2021-11-03:
Hello

const char *pipeline = "gst-launch-1.0 

Remove that "gst-launch-1.0" ! Start with "tcambin"

Stefan

seymseym, 2021-11-04:
Hello,

Tried that, didn't work

TIS-Stefan, 2021-11-04:
I looked in to the OpenCV samples....

```
void test()
{
	const char *pipeline = "tcambin serial=10710002 ! video/x-raw, format=GRAY8, framerate=30/1, width=640, height=480  ! videoconvert ! appsink"; 
         cv::VideoCapture *cap = new cv::VideoCapture(pipeline,cv::CAP_GSTREAMER);
	if (!cap->isOpened()) 
	{
		printf("=ERR= can't create video capture\n");
		return;
	}
        cv::Mat frame;
        *cap >> frame; // get a new frame from camera
```
The line

`cv::VideoCapture *cap = new cv::VideoCapture(pipeline,cv::CAP_GSTREAMER);`

opens the camera already, so you are right, your code 

`cap->open(0);   
`
will not work, because the camera is opened.
 
Stefan

seymseym, 2021-11-04:
I also tried commenting that line but I get pipeline errors when I do that

```
[ WARN:0] global /home/upboard/opencv-master/modules/videoio/src/cap_gstreamer.cpp (2076) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module v4l2src0 reported: Cannot identify device '/dev/video116'.
[ WARN:0] global /home/upboard/opencv-master/modules/videoio/src/cap_gstreamer.cpp (1053) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global /home/upboard/opencv-master/modules/videoio/src/cap_gstreamer.cpp (616) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
QObject::startTimer: Timers cannot have negative intervals
```

seymseym, 2021-11-04:
I somehow solved this problem, but I have one more question regarding this topic.

This pipeline does not work when using it on **Python**. Do I need to make some changes on it, or is the problem elsewhere?

```
cap = cv2.VideoCapture("tcambin serial=40120113 ! video/x-raw, format=BGRx, width=640,height=480, framerate=60/1 ! videoconvert ! appsink") 
cap.set(3, 640)
cap.set(4, 480)

   # Start reading the webcam feed frame by frame.
    ret, frame = cap.read()
    if not ret:
        break
```

TIS-Stefan, 2021-11-04:
Hello 

Maybe that OpenCV compile for Python is not compiled with use GStreamer? Also I miss the second parameter to VideoCapture() which is "cv::CAP_GSTREAMER" in C++. Thus maybe OpenCV does not know, what to do with that string.

Stefan

latsa, 2022-07-25:
> I somehow solved this problem
Would you care to tell us how?

TIS-Edgar, 2024-04-12:
Closed due to inactivity.

closed 2024-04-12 by TIS-Edgar