TheImagingSource/tiscamera #437: Adjusting the camera properties using Gstreamer Pipeline

seymseym, 2021-11-04:
**I have 2 questions regarding this topic:**

**1)**
 
Is it possible to set the whitebalancing on the gstreamer pipeline?
Like this example that changes the saturation:

` gst-launch-1.0 videotestsrc ! videobalance saturation=0.0 ! videoconvert ! ximagesink`

Camera outputs look different in indoor and outdoor applications. So I want to change the white balance properties exactly like the above example, is that possible? 

**2**
Camera captures are in somekind of a 'fisheye' mode. Can I change this using Gstreamer? Or Is there a way to remove this property internally from the camera? I won't ever use the fisheye look and it seems to be the default choice, is there a way to remove this completely?

**Camera Model:** DFM 27UR0135-ML - USB 3.0 color board camera

TIS-Stefan, 2021-11-04:
Hello

Question 1:
The whitebalance is done in software and there only, after the pipeline has been started. Thus the automatic of Whitebalance is enabled and your image does not look green.

Adapting with gst-launch-1.0 is not possible with tiscamera 0.14. Except you use a pipeline like

`tcamsrc serial=51310104 ! video/x-bayer,format=gbrg,width=744,height=480,framerate=30/1 ! tcamwhitebalance auto=false red=64 green=64 blue=64 ! tcamautoexposure ! bayer2rgb ! videoconvert ! ximagesink`

This is a pipeline for a DFK 22BUC03.
First of all, you start with 
tcam-ctrl -l
so you know the serial number of your camera. Then you query its video formats with 
tcam-ctrl -c <serial number>
Now you know the video format parameters, in particular the bayer start pattern of the sensor.
With tcam-capture you can query the optimal red, green and blue values for your light situations. 64 is the neutral value.
The line
gst-instpect-1.0 tcamwhitebalance lists the available properties. The same can be done with exposure. You can specify fixed gain and exposure, or limit the automatic to special ranges for gain and exposure.

Question 2)
It is not the camera, but your optics, if you see a fisheye. This can not be fixed by Gstreamer. If this must be fixed. you need to measure the intrinsics of your lens and create a correction matrix, which is applied to the images of the camera. This can be done with OpenCV : https://docs.opencv.org/3.4.15/dc/dbb/tutorial_py_calibration.html. 
(Maybe it is an interesting idea to write a Gstreamer module, that can apply this matrix to a stream....)
An 8mm lens should have no or very low fisheye effect. The lower the focal length, the bigger the fisheye effect. This can be reduced by use of low distortion lenses, but they are more expensive.

Stefan

seymseym, 2021-11-05:
Hello,
Thank you for the explanation.

_"You can specify fixed gain and exposure, or limit the automatic to special ranges for gain and exposure."_

How am I going to set the gain and exposure **from the pipeline**? Is there a way to easily set them like this saturation example? **gst-launch-1.0 videotestsrc ! videobalance saturation=0.0 ! videoconvert ! ximagesink**

TIS-Stefan, 2021-11-05:
Gain and exposure are set in tcamautoexposure. You may run gst-inspect-1.0 tcamautoexposure in order to see its properties.
Saturation is not available with the pipeline we used before and the tiscamera-dutils do not implement a property interface like tcamwhitebalance and tcamautoexposure. I am sorry.

seymseym, 2021-11-05:
Sorry, I could'nt figure how to set the **gain and exposure** properties in the **pipeline**. Can you give a working example? 

For ex. I wanted to set the **gain-min = 2** The line below is not working:

`const char *pipeline = "tcambin serial=40120113 ! video/x-bayer, format=gbrg, framerate=60/1 ! tcamwhitebalance auto=false red=64 green=64 blue=64 ! tcamautoexposure gain-min = 2 ! bayer2rgb ! appsink";`

```

**Information about tcamautoexposure:**
Pad Templates:
  SINK template: 'sink'
    Availability: Always
    Capabilities:
      video/x-raw
                 format: { (string)GRAY8, (string)GRAY16_LE }
                  width: [ 1, 2147483647 ]
                 height: [ 1, 2147483647 ]
              framerate: [ 0/1, 2147483647/1 ]
      video/x-bayer
                 format: { (string)rggb, (string)bggr, (string)gbrg, (string)grbg }
                  width: [ 1, 2147483647 ]
                 height: [ 1, 2147483647 ]
              framerate: [ 0/1, 2147483647/1 ]
  
  SRC template: 'src'
    Availability: Always
    Capabilities:
      video/x-raw
                 format: { (string)GRAY8, (string)GRAY16_LE }
                  width: [ 1, 2147483647 ]
                 height: [ 1, 2147483647 ]
              framerate: [ 0/1, 2147483647/1 ]
      video/x-bayer
                 format: { (string)rggb, (string)bggr, (string)gbrg, (string)grbg }
                  width: [ 1, 2147483647 ]
                 height: [ 1, 2147483647 ]
              framerate: [ 0/1, 2147483647/1 ]

Element has no clocking capabilities.
Element has no URI handling capabilities.

Pads:
  SINK: 'sink'
    Pad Template: 'sink'
  SRC: 'src'
    Pad Template: 'src'

Element Properties:
  name                : The name of the object
                        flags: readable, writable
                        String. Default: "tcamautoexposure0"
  parent              : The parent of the object
                        flags: readable, writable
                        Object of type "GstObject"
  qos                 : Handle Quality-of-Service events
                        flags: readable, writable
                        Boolean. Default: false
  auto-exposure       : Automatically adjust exposure
                        flags: readable, writable
                        Boolean. Default: true
  auto-gain           : Automatically adjust gain
                        flags: readable, writable
                        Boolean. Default: true
  auto-iris           : Automatically adjust the iris, if camera allows adjustments
                        flags: readable, writable
                        Boolean. Default: true
  camera              : Gstreamer element that shall be manipulated
                        flags: readable, writable
                        Object of type "GstElement"
  brightness-reference: Ideal average brightness of buffer
                        flags: readable, writable
                        Integer. Range: 0 - 255 Default: 128 
  exposure-min        : Minimum value exposure can take
                        flags: readable, writable
                        Integer. Range: 0 - 2147483647 Default: 0 
  exposure-max        : Maximum value exposure can take
                        flags: readable, writable
                        Integer. Range: 0 - 2147483647 Default: 2147483647 
  gain-min            : Minimum value gain can take
                        flags: readable, writable
                        Double. Range:               0 -   1,797693e+308 Default:               0 
  gain-max            : Maximum value gain can take
                        flags: readable, writable
                        Double. Range:               0 -   1,797693e+308 Default:   1,797693e+308 
```

TIS-Stefan, 2021-11-05:
Hello

I asked the programmer and wait for an answer.,

I wonder, why you use the test program gst-launch-1.0. If you wrote an e.g. Python script, you had access to all properites, including saturation, if tiscamera-dutils are installed.

Stefan

seymseym, 2021-11-05:
Hello,

Thank you so much for your interest.

Actually, I am working on an OpenCV application on C++. The camera captures were either so bright or somehow broken and I couldn't solve this issue with the tiscamera library. I've also reached you through e-mail and get this answer:

_..because the properties, which you want to set do not exist.
Maybe you use OpenCV with GStreamer and pass a pipeline to the constructor of cv::videocapture_

Since then, I am using the pipeline for the captures and **getting much better frames.**
Therefore I want to **keep the pipeline** and solve this situation in an easy way by using it.

TIS-Stefan, 2021-11-05:
Hello

I made a few C++ samples at https://github.com/TheImagingSource/Linux-tiscamera-Programming-Samples that show, how to get a cv::Mat from a Gstreamer Pipeline. 
With these samples you have full control over all camera properties. And you can change the property values, while your software runs. 

What do you think about this?

Stefan

TIS-Edgar, 2024-04-12:
Closed due to inactivity.