TheImagingSource/tiscamera #438: usb2 GPIO control missing

ElTinmar, 2021-11-10:
Hi, I have a DFK72BUC02, everything is working fine but I seem to be unable to control the GPIO. Is it supported as of now ?
I have the following errors:

tcam-uvc-extension-loader -vf /usr/share/theimagingsource/tiscamera/uvc-extension/usb2.json 

Error while mapping 'ExtIO': errno: 2 - No such file or directory
Error while mapping 'GainR': errno: 2 - No such file or directory
Error while mapping 'GainG': errno: 2 - No such file or directory
Error while mapping 'GainB': errno: 2 - No such file or directory
Error while mapping 'Trigger': errno: 2 - No such file or directory
Error while mapping 'Trigger Global Reset Shutter': errno: 2 - No such file or directory
Error while mapping 'Software Trigger': errno: 2 - No such file or directory
Error while mapping 'Binning': errno: 2 - No such file or directory
Error while mapping 'Skipping': errno: 2 - No such file or directory
Error while mapping 'X Offset': errno: 2 - No such file or directory
Error while mapping 'Y Offset': errno: 2 - No such file or directory
Error while mapping 'Strobe Enable': errno: 2 - No such file or directory
Error while mapping 'Strobe Polarity': errno: 2 - No such file or directory

Thanks for the great work !

ElTinmar, 2021-11-10:
Also I update the firmware to dfk72uc02_af_146.euvc

./firmware-update -i -d 12110338

Device manufacturer: The Imaging Source Europe GmbH
Product name:        DFx 72BUC02
Serial number:       12110338
VendorID:ProductID:  199e:8307
Firmware version:    146
UVC mode is:         on
Camera EEPROM size:  16384

TIS-Stefan, 2021-11-10:
Hello

if the firmware 3012 is in the camera and you run v4l2-ctrl --all, you should get something like this:
```
                 brightness 0x00980900 (int)    : min=0 max=255 step=1 default=12 value=12
                           gain 0x00980913 (int)    : min=4 max=63 step=1 default=16 value=36
              exposure_absolute 0x009a0902 (int)    : min=1 max=300000 step=1 default=127 value=127
                 focus_absolute 0x009a090a (int)    : min=0 max=1000 step=1 default=0 value=0
                        privacy 0x009a0910 (bool)   : default=0 value=0
                        trigger 0x0199e208 (bool)   : default=0 value=0
               software_trigger 0x0199e209 (button) :
                          gainr 0x0199e921 (int)    : min=0 max=63 step=1 default=1 value=36
                          gaing 0x0199e922 (int)    : min=0 max=63 step=1 default=1 value=36
                        binning 0x0199e925 (int)    : min=1 max=4 step=1 default=1 value=1
                       x_offset 0x0199e927 (int)    : min=0 max=2562 step=2 default=0 value=0
                       y_offset 0x0199e928 (int)    : min=0 max=1944 step=2 default=0 value=0
   trigger_global_reset_shutter 0x0199e92a (bool)   : default=0 value=0
```
However, GPIO is no available.

Stefan

TIS-Edgar, 2024-04-12:
Closed due to inactivity.

closed 2024-04-12 by TIS-Edgar