TheImagingSource/tiscamera #439: DFK 23GM021 & 外触发 & ros包

yyqgood, 2021-11-22:
你好：
有关DFK 23GM021这款相机，我们使用外触发的方式，请问：有没有ROS包的相关支持？
期待你的回复，谢谢！

TIS-Stefan, 2021-11-22:
Hello

I will answer in English (my Chinese is not very well.)
ROS supports Gstreamer 1.0, thus you can use the camera with out plugins in ROS. Maybe ROS has a GigeVision support too, then none of our software is needed. You may check this.

Stefan

yyqgood, 2021-11-24:
1. Thank you for your reply! I will try it. 

2. I now find a problem: we are going to use this camera for autonomous driving;
I saw: If it is set to trigger mode, both auto exposure and auto white balance will be turned off? We can only manually set the relevant parameters?
If this is the case, for complex outdoor scenes, I think it is not appropriate to manually set the relevant parameters.
Don't know if I missed something? Looking forward to your reply!

yyqgood, 2021-11-24:
3. If the DFK 23GM021 is really not suitable for autonomous driving, what model of camera can you recommend (the recommended camera needs to be able to set the trigger mode)?

TIS-Stefan, 2021-11-24:
Hello

> I now find a problem: we are going to use this camera for autonomous driving;
> I saw: If it is set to trigger mode, both auto exposure and auto white balance will be turned off? We can only manually [...]
The automatics need incoming images to adapt to the light situation. Therefore, the new values for exposure and gain have an effect on the next image. If you have a very low trigger frequency, e.g. 0.2 Hz and the light situation changes between two triggers, then the image incoming is exposed wrongly. But if your trigger frequency is "high", e.g. 10 frames per second, or even 5 frames per second, then this effect wont be that strong. You can try that on your own, if setting trigger mode and enable automatics.
This is for all of our cameras, because we do not use extra sensors for light measurement in the camera. Therefore, your question 3 is answered too.

In case you are stopping, e.g, in front of a traffic light, and you get too less images per second, your software could fire a software trigger, so the automatics can adapt. Which means, you have a callback and a watchdog checking for incoming images in a regular time interval, e.g. 1 second.

Stefan

yyqgood, 2021-11-24:
I'm very happy to have received your response!

Our application conditions and scenarios are: use the external trigger mode; the vehicle is driving; real-time detection of obstacles such as vehicles and pedestrians; under these conditions and scenarios, does the camera DFK 23GM021 meet?

If not, do you have any recommended camera selection?

Looking forward to your reply! Thanks!

TIS-Stefan, 2021-11-24:
You can start with your camera. I am not the person to justify, whether the camera meets your demands. If the detection results are too bad, you should consider using a different camera, e.g. with higher resolution. 

In case your system runs on a Jetson (Nano, Xavier NX, Xavier AGX) you may consider using a DFM 36CX390. This camera is very light sensitive and uses 24bit color depth internally. Therefore, no automatics for exposure and gain are used. You may have a look at https://www.theimagingsource.com/embedded-vision/fpd-link-iii-cameras/board-cameras-with-fpd-link-iii/dfm36cx390ml/

TIS-Edgar, 2024-04-12:
Closed due to inactivity.

closed 2024-04-12 by TIS-Edgar