TheImagingSource/tiscamera #440: DYK 33GX250 no caps

kerecsenszabo, 2021-11-26:
tcam-ctrl -l
Model: DYK 33GX250 Serial: 18020217 Type: aravis

tcam-ctrl -c 18020217
Available gstreamer-1.0 caps:
EMPTY

And therefore nothing works. I have installed tiscam with aravis.

TIS-Stefan, 2021-11-26:
Hello

you got it right. Therefore, I wrote to your email you sent to me, that you need Aravis directly, not tiscamera nor Gstreamer. I also sent a C++ sample do you, which as far as I remember the four polarization images separates.

Didn't you got my answer some minutes before?

Stefan

TIS-Stefan, 2021-11-26:
​ 

Dear Mr Szabó,

I also answered to your github issue.

Yours sincerely

Stefan Geissler
--------------------------------------------------------------------
The Imaging Source Europe GmbH
Ueberseetor 18
D-28217 Bremen
Germany
Tel   +49/421/33591-0
Fax   +49/421/33591-80
Registergericht: Amtsgericht Bremen HRB 11 927
Geschäftsführer: Rolf Bollhorst 
--------------------------------------------------------------------

E-MAIL 

  

Dear Mr Szabó,

Thank you for your email.
You can use the polarization camera in Linux with Aravis library only. It is available at https://github.com/AravisProject/aravis
I attached a small C++ sample. It uses OpenCV for displaying the image. 
I hope, this helps a little bit!

Have a nice weekend!

Yours sincerely

Stefan Geissler
--------------------------------------------------------------------
The Imaging Source Europe GmbH
Ueberseetor 18
D-28217 Bremen
Germany
Tel   +49/421/33591-0
Fax   +49/421/33591-80
Registergericht: Amtsgericht Bremen HRB 11 927
Geschäftsführer: Rolf Bollhorst 
--------------------------------------------------------------------

E-MAIL 

  

 Hi,

How do I get a DYK 33GX250 camera stream working on Linux?

Thank you,
Kerecsen Szabó 

  
 Was this message helpful?   Yes   [https://theimagingsource.deskpro.com/ticket-rate/OVHE-2933-MIKF/MYJSNH5J2DW9BAW/354602?setrating=1]   It was OK   [https://theimagingsource.deskpro.com/ticket-rate/OVHE-2933-MIKF/MYJSNH5J2DW9BAW/354602?setrating=0]   No   [https://theimagingsource.deskpro.com/ticket-rate/OVHE-2933-MIKF/MYJSNH5J2DW9BAW/354602?setrating=-1] 

View and manage this ticket online: https://theimagingsource.deskpro.com/tickets/72710 

  The Imaging Source [https://www.theimagingsource.com] · https://theimagingsource.deskpro.com/ 

 (#EDOOMYJSNH5J2DW9BAW)

kerecsenszabo, 2021-11-26:
I see, thank you for the rapid response.

closed 2021-11-26 by kerecsenszabo