TheImagingSource/tiscamera #441: tiscamera on arm32, DMx 41BU02, "Device did not provide ANY valid caps."

rettigl, 2021-12-01:
Hello,
I'm trying to get tiscamera running with a DMx 41BU02 USB camera on a raspberry pi, running Raspbian buster (linux 5.4.83-v71+).
Compiling works, and the camera is found by the tool:
```
tcam-ctrl -l
Model: DMx 41BU02 Serial: 23210037 Type: v4l2
```

However, if I want to start a stream using the provided python example, I get the following errors:
```
./03-live-stream.py
0:00:02.283877183 18401   0xa0bce0 ERROR            tcammainsrc gsttcammainsrc.cpp:727:gst_tcam_mainsrc_get_all_camera_caps: Device did not provide ANY valid caps. Refusing playback.
0:00:02.291359501 18401   0xa0bce0 ERROR                tcambin gsttcambin.cpp:1284:gst_tcam_bin_change_state:<bin> Unable to work with given caps: video/x-raw, format=(string)BGRx, width=(int)[ 1, 2147483647 ], height=(int)[ 1, 2147483647 ], framerate=(fraction)[ 0/1, 2147483647/1 ], pixel-aspect-ratio=(fraction)1/1; video/x-raw, width=(int)[ 1, 2147483647 ], height=(int)[ 1, 2147483647 ], framerate=(fraction)[ 0/1, 2147483647/1 ], pixel-aspect-ratio=(fraction)1/1, format=(string){ I420, YV12, YUY2, UYVY, AYUV, RGBx, BGRx, xRGB, xBGR, RGBA, BGRA, ARGB, ABGR, RGB, BGR, Y41B, Y42B, YVYU, Y444, v210, v216, NV12, NV21, GRAY8, GRAY16_BE, GRAY16_LE, v308, RGB16, BGR16, RGB15, BGR15, UYVP, A420, RGB8P, YUV9, YVU9, IYU1, ARGB64, AYUV64, r210, I420_10BE, I420_10LE, I422_10BE, I422_10LE, Y444_10BE, Y444_10LE, GBR, GBR_10BE, GBR_10LE, NV16, NV24, NV12_64Z32, A420_10BE, A420_10LE, A422_10BE, A422_10LE, A444_10BE, A444_10LE, NV61, P010_10BE, P010_10LE, IYU2, VYUY, GBRA, GBRA_10BE, GBRA_10LE, GBR_12BE, GBR_12LE, GBRA_12BE, GBRA_12LE, I420_12BE, I420_12LE, I422_12BE, I422_12LE, Y444_12BE, Y444_12LE, GRAY10_LE32, NV12_10LE32, NV16_10LE32 }
Press Ctrl-C to stop.
0:00:02.292374317 18401   0x8ce320 ERROR                tcambin gsttcambin.cpp:1284:gst_tcam_bin_change_state:<bin> Unable to work with given caps: video/x-raw, format=(string)BGRx, width=(int)[ 1, 2147483647 ], height=(int)[ 1, 2147483647 ], framerate=(fraction)[ 0/1, 2147483647/1 ], pixel-aspect-ratio=(fraction)1/1; video/x-raw, width=(int)[ 1, 2147483647 ], height=(int)[ 1, 2147483647 ], framerate=(fraction)[ 0/1, 2147483647/1 ], pixel-aspect-ratio=(fraction)1/1, format=(string){ I420, YV12, YUY2, UYVY, AYUV, RGBx, BGRx, xRGB, xBGR, RGBA, BGRA, ARGB, ABGR, RGB, BGR, Y41B, Y42B, YVYU, Y444, v210, v216, NV12, NV21, GRAY8, GRAY16_BE, GRAY16_LE, v308, RGB16, BGR16, RGB15, BGR15, UYVP, A420, RGB8P, YUV9, YVU9, IYU1, ARGB64, AYUV64, r210, I420_10BE, I420_10LE, I422_10BE, I422_10LE, Y444_10BE, Y444_10LE, GBR, GBR_10BE, GBR_10LE, NV16, NV24, NV12_64Z32, A420_10BE, A420_10LE, A422_10BE, A422_10LE, A444_10BE, A444_10LE, NV61, P010_10BE, P010_10LE, IYU2, VYUY, GBRA, GBRA_10BE, GBRA_10LE, GBR_12BE, GBR_12LE, GBRA_12BE, GBRA_12LE, I420_12BE, I420_12LE, I422_12BE, I422_12LE, Y444_12BE, Y444_12LE, GRAY10_LE32, NV12_10LE32, NV16_10LE32 }
0:00:02.292769329 18401   0x8ce320 ERROR                tcambin gsttcambin.cpp:1284:gst_tcam_bin_change_state:<bin> Unable to work with given caps: video/x-raw, format=(string)BGRx, width=(int)[ 1, 2147483647 ], height=(int)[ 1, 2147483647 ], framerate=(fraction)[ 0/1, 2147483647/1 ], pixel-aspect-ratio=(fraction)1/1; video/x-raw, width=(int)[ 1, 2147483647 ], height=(int)[ 1, 2147483647 ], framerate=(fraction)[ 0/1, 2147483647/1 ], pixel-aspect-ratio=(fraction)1/1, format=(string){ I420, YV12, YUY2, UYVY, AYUV, RGBx, BGRx, xRGB, xBGR, RGBA, BGRA, ARGB, ABGR, RGB, BGR, Y41B, Y42B, YVYU, Y444, v210, v216, NV12, NV21, GRAY8, GRAY16_BE, GRAY16_LE, v308, RGB16, BGR16, RGB15, BGR15, UYVP, A420, RGB8P, YUV9, YVU9, IYU1, ARGB64, AYUV64, r210, I420_10BE, I420_10LE, I422_10BE, I422_10LE, Y444_10BE, Y444_10LE, GBR, GBR_10BE, GBR_10LE, NV16, NV24, NV12_64Z32, A420_10BE, A420_10LE, A422_10BE, A422_10LE, A444_10BE, A444_10LE, NV61, P010_10BE, P010_10LE, IYU2, VYUY, GBRA, GBRA_10BE, GBRA_10LE, GBR_12BE, GBR_12LE, GBRA_12BE, GBRA_12LE, I420_12BE, I420_12LE, I422_12BE, I422_12LE, Y444_12BE, Y444_12LE, GRAY10_LE32, NV12_10LE32, NV16_10LE32 }
0:00:02.292813365 18401   0x8ce320 WARN                     bin gstbin.c:2832:reset_state:<bin> Failed to switch back down to PAUSED
```

Indeed, I cannot get any information from the camera:
```
tcam-ctrl -c 23210037
Available gstreamer-1.0 caps:
EMPTY
```

Am I missing something?

TIS-Stefan, 2021-12-02:
I just tested on an PI4 with Rasbpian 5.10. It works perfectly, after the tiscamera repository was built and cloned. Maybe you can use 5.10 too? Unfortunately I am guessing into the blue only.

rettigl, 2021-12-29:
Hi,
I also tested with Raspian 5.10, and also on an Ubuntu 18 Installation now, to no avail, similar response. Is this camera not compatible? Or am I missing an option?

TIS-Stefan, 2021-12-29:
I did the same test 27days ago and my old DMK 41BU02 worked perfectly.  Therefore, there is no incompatibility. 
Please make sure, you have enough power. The camera needs the full 500mA provided by USB 2.0. (Ok, the PI4 has USB 3.0, so 900mA must be provided. If there is not enough power, there is a 4 color symbol in the upper right corner).
Also check the USB 2.0 cable. 
You may check the UDEV rules, please refer to https://theimagingsource.deskpro.com/de/kb/articles/using-usb-2-0-ccd-cameras-in-linux. But the camera is listed as v4l2 device already.
Does v4l2-ctl --all show something with sense?
Does q4vl2 handle the camera? 
If yes, then there is a Gstreamer issue, which I do not know at the moment.

I am very sorry for guessing again, because the camera works fine on my computers: PI4 and PC with Ubuntu 20.04.

Does the camera work on a Windows PC?

rettigl, 2021-12-29:
Oh, thank you, this brought me on the right track. It was a permission problem!

closed 2021-12-29 by rettigl

TIS-Stefan, 2021-12-29:
Very well! I had no idea about the permission on my list....

Ich wünsche Ihnen einen guten Rutsch und ein erfolgreiches 2022!

Stefan