TheImagingSource/tiscamera #444: value error operation forbidden on released memoryview object. Segmentation fault (core dumped)

vinit13792, 2022-01-06:
Hardware specs:
1. DFK 37BUX178
2. Raspberry Pi 4, 4GB

OS:
Ubuntu RPi4 21.04

What did I do?
I was running the code example from here: 
[https://github.com/TheImagingSource/Linux-tiscamera-Programming-Samples/blob/master/Snap%20an%20Image%20and%20convert%20to%20OpenCV%20in%20Python/Program.py]

This program takes a video stream, captures a frame, converts that frame to opencv image. This is what I understand from the description and code given on the link mentioned.

And I get the error ValueError: Operation forbidden on release memoryview object. Segmentation fault (core dumped)

Whats the solution for this? How do I move forward?

TIS-Stefan, 2022-01-06:
Hello

As written parallel in email to me (which doubles my work load), please try lower frame rates. The PI4 is a weak computer and if there is running something in parallel to image data transfer, the USB controller is blocked and can not write all image data packets into memory. This issue can be lowered, if you use lower frame rates. Start with the lowest one, offered in the selection. 

BTW: Your issue description here is excellent, I appreciate that!

Stefan

vinit13792, 2022-01-06:
> Hello
> 
> As written parallel in email to me (which doubles my work load), please try lower frame rates. The PI4 is a weak computer and if there is running something in parallel to image data transfer, the USB controller is blocked and can not write all image data packets into memory. This issue can be lowered, if you use lower frame rates. Start with the lowest one, offered in the selection.
> 
> Edit: I forget: Please try tcam-capture program provided with tiscamera and check, whether you get an image.
> 
> BTW: Your issue description here is excellent, I appreciate that!
> 
> Stefan

Hi Mr. Stefan,

The following are the options I get prompted when I run the above mentioned program.
1. Select Camera, in my case it is DFK 37BUX178
2. Select format, there are some eight of them
```
1: rggb
2: rggb16
3: {RGBx,xRGB,BGRx,xBGR,RGBA,ARGB,BGRA,ABGR}
4: {RGBx,xRGB,BGRx,xBGR,RGBA,ARGB,BGRA,ABGR}
5: {RGBx,xRGB,BGRx,xBGR,RGBA,ARGB,BGRA,ABGR}
6: {RGBx,xRGB,BGRx,xBGR,RGBA,ARGB,BGRA,ABGR}

```
I select option 3.

Then I get the following option:
`
1:  640x480
`
I select 1, since there is only one option no matter which options I select from previous list. Although, higher the number from previous list, higher the resolution is we get.

I think this is what FPS is because this resembles the FPS option in tcam-capture.
```
1:  280/1
2:  5000000/20833
3:  120/1
4:  60/1
5:  30/1
6:  15/1
7:  5/1
8:  1/1
```
I tried lowering the resolution from option 5 till 8. I get the same error.

What else can I do to remove this error? Will changing my system from Raspberry Pi to an Intel-like system solve the problem?

TIS-Stefan, 2022-01-06:
Please select option 1, rggb. You should get more resolutions then.  30 fps is a very fine selection for a 640x480 resolution. 

Please let me know your result.

BTW: 3,4,5,6 in the format selection looks somehow wrong. Its rather a list of formats in each row. I do not know, why Gstramaer shows that list.

Stefan

vinit13792, 2022-01-06:
Hi Mr. Stefan,

I tried `rggb` then  `640x480` and fps `15/1.` I still get the same error.

TIS-Stefan, 2022-01-06:
Are other, in particular higher resolutions shown?
Did you start tcam-capture for testing?
 
Stefan

vinit13792, 2022-01-06:
Yes, tcam-capture works.

TIS-Stefan, 2022-01-06:
Which Python version do you use? I use Python 3.8. The program's output should be:

```
~/projects/samples/Snap an Image and convert to OpenCV in Python$ python3.8 Program.py 
1 : Model: DFK 37BUX178 Serial: 03020073 
Select : 1
1: rggb
2: rggb16
3: {RGBx,xRGB,BGRx,xBGR,RGBA,ARGB,BGRA,ABGR}
4: {RGBx,xRGB,BGRx,xBGR,RGBA,ARGB,BGRA,ABGR}
5: {RGBx,xRGB,BGRx,xBGR,RGBA,ARGB,BGRA,ABGR}
6: {RGBx,xRGB,BGRx,xBGR,RGBA,ARGB,BGRA,ABGR}
7: {RGBx,xRGB,BGRx,xBGR,RGBA,ARGB,BGRA,ABGR}
Select : 1
1:  640x480
2:  1536x1024
3:  1920x1080
4:  2048x2048
5:  3072x2048
Select : 1
1:  280/1
2:  5000000/20833
3:  120/1
4:  60/1
5:  30/1
6:  15/1
7:  5/1
8:  1/1
Select : 5
tcambin name=source ! capsfilter name=caps ! tee name=t t. ! queue ! videoconvert ! ximagesink t. ! queue ! appsink name=sink

```
And it should how two live stream windows. I works fine on my computer.

vinit13792, 2022-01-06:
I use python 3.8

What are your hardware specs?

TIS-Stefan, 2022-01-06:
I understand the Gstreamer documentation is not that easy to understand, inparticular, if you want to go deeper into some stuff.

> What are your hardware specs?

Desktop computer, x86 based. Ubuntu 20.04. 
The Python code is unchanged. Just started it. It must work in the same way on the PI4 and Nvida Jetson. 

You do not get different resolutions, as I get?

Stefan

TIS-Stefan, 2022-01-06:
BTW: tcam-capture is written in Python.....

vinit13792, 2022-01-06:
> I understand the Gstreamer documentation is not that easy to understand, inparticular, if you want to go deeper into some stuff.
>
>
> Desktop computer, x86 based. Ubuntu 20.04. The Python code is [...]

I do get the same options for resolutions as you. But it still didn't work for me. Maybe the raspberry pi 4, 4GB model has some memory and CPU optimization constraints which restricts it from working. 

Tcam-capture on the other hand didn't throw any error. it worked. I don't know why that is. I think tcam-capture should throw error if it was a hardware issue.

TIS-Stefan, 2022-01-06:
I suggest to comment the part

```
       if Tis.Snap_image(1) is True:  # Snap an image with one second timeout
                image = Tis.Get_image()  # Get the image. It is a numpy array
                image = cv2.erode(image, kernel, iterations=5)  # Example OpenCV image processing
                cv2.imshow('Window', image)  # Display the result

```
Then a live video should be shown

TIS-Stefan, 2022-01-06:
Hello !

Anything new? Or did you got any results?

Stefan

vinit13792, 2022-01-06:
> I suggest to comment the part
>
> ```
>        if Tis.Snap_image(1) is True:  # Snap an image with one second timeout
>                 image = Tis.Get_image()  # Get the image. It is a numpy [...]

Hi Mr. Stefan, 

I'll definitely try this out soon. The company who is going to sell this equipment, I was at their site, so I cannot test it after working hours. Also it's quite far from my house. I'll definitely try this out in 4 days as I will be going there after 4 days.

TIS-Stefan, 2022-01-06:
Hello

Very well! 

Stefan

vinit13792, 2022-01-11:
Hi Mr. Stefan,

I tried the above with my laptop which consists of 8GB RAM, Core i5, Ubuntu OS. It worked really well and I was able to retrieve an image, get into opencv format which BGR, do some basic image processing on it and save the image. 

The main issue i think could have been the Raspberry PI 4's hardware limitations which caused this error as i was using a 4GB version.

Thank you for your support.

closed 2022-01-11 by vinit13792

TIS-Stefan, 2022-01-11:
Hello!

Thank you very much for letting me know your results. 

On the PI 4 you may try following:
1) Try a lower frame rate.
2.) Use software trigger to trigger an image and process it, when it is avaiallbe.

With 2) you do not have image delivery and processing in parallel.

You must be able to select the same resolutions on the PI4 as I posted 5 days ago. Not 640x480 only.

Stefan

vinit13792, 2022-01-11:
> 

Hi Mr. Stefan

I think doing it on a system which has less problems is always better than doing it on system with memory constraints. Although very soon, i may test this out on Jetson Nano. I think that is much powerful than RPi4.

Regards,
Vinit.