TheImagingSource/tiscamera #446: Unable to change FourCC code in OpenCV

jaredwaller-git, 2022-02-07:
Hi,

I am using OpenCV to do image recognition. I am using the **DMK 33UX273** and I wish to change the camera pixel output from 16-bit to 8-bit by changing the FourCC code. The camera defaults in 16-bit with FourCC code _Y16_ as per the technical manual and by checking manually using OpenCV read-outs. According to the technical manual, 8-bit monochrome has FourCC code _Y800_, however when I enter this value in OpenCV the value fails to change.

`capture = cv2.VideoCapture(0)`
`capture.set(cv2.CAP_PROP_FOURCC, cv2.VideoWriter_fourcc('Y','8','0','0')) #Fourcc code fails to change from 16bit to 8bit`
`FourCC = int(capture.get(cv2.CAP_PROP_FOURCC)).to_bytes(4, 'little') #Check FourCC code`
`print(str(FourCC))`

The value defaults to _Y16_ when OpenCV fails to change FourCC code. I have, however, found that the FourCC code _GREY_ is accepted into OpenCV as a valid FourCC code for the camera, however the picture is completely black.

If I am doing something wrong with OpenCV or if there is an easier way to change the FourCC code (or programmatically change into 8-bit mode), please let me know, thanks.

TIS-Stefan, 2022-02-08:
That is an OpenCV issue...

You may try
	//cap.set(CV_CAP_PROP_FOURCC, 0x31313459);  // Y411
	//cap.set(CV_CAP_PROP_FOURCC, 0x32595559); // YUY2
	//cap.set(CV_CAP_PROP_FOURCC, 0x32424752); //RGB24
	//cap.set(CV_CAP_PROP_FOURCC, 0x41424752); //RGB32
	//cap.set(CV_CAP_PROP_FOURCC, 0x30303859); //Y800
Maybe that works better.

Stefan

jaredwaller-git, 2022-02-08:
Hi, thanks for the quick response. I tried those FourCC codes but unfortunately they did not work. 

I used v4l2-ctl to list out the UVC/V4L2 codes for the devices, and these were my results:

`pi@raspberrypi:~/Desktop/test $ v4l2-ctl --list-formats-ext -d 0`
`ioctl: VIDIOC_ENUM_FMT`
`	Type: Video Capture`

	[0]: 'GREY' (8-bit Greyscale)
		Size: Discrete 1440x1080
			Interval: Discrete 0.004s (236.005 fps)
			Interval: Discrete 0.008s (120.000 fps)
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.200s (5.000 fps)
			Interval: Discrete 1.000s (1.000 fps)
		Size: Discrete 640x480
			Interval: Discrete 0.002s (601.000 fps)
			Interval: Discrete 0.002s (480.000 fps)
			Interval: Discrete 0.004s (240.004 fps)
			Interval: Discrete 0.008s (120.000 fps)
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.200s (5.000 fps)
			Interval: Discrete 1.000s (1.000 fps)
	[1]: '' (70323159-0000-0010-8000-00aa003)
	[2]: 'Y16 ' (16-bit Greyscale)
		Size: Discrete 1440x1080
			Interval: Discrete 0.008s (120.000 fps)
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.200s (5.000 fps)
			Interval: Discrete 1.000s (1.000 fps)
		Size: Discrete 640x480
			Interval: Discrete 0.003s (377.000 fps)
			Interval: Discrete 0.004s (240.004 fps)
			Interval: Discrete 0.008s (120.000 fps)
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.200s (5.000 fps)
			Interval: Discrete 1.000s (1.000 fps)

Not sure if this helps, but I do not think my camera has the FourCC codes I need. There may be something improperly installed on my device, but in any case I am unable to programmatically change into 8-bit mode. If you can think of anything else that might help please let me know, if not I'll need to explore different solutions to get into 8-bit pixel mode.

Thanks again

TIS-Stefan, 2022-02-09:
The camera has Y800 (GREY) and Y16. It seems, we have an OpenCV issue here. Unfortunately, I am not the official OpenCV supporter. 

After looking into OpenCV source code and some testing, I found after 1,5 hours:
You can set
```
cap.set(cv::CAP_PROP_CONVERT_RGB,0);

auto fourcc =  VideoWriter::fourcc('G','R','E','Y');
//auto fourcc =  VideoWriter::fourcc('Y','1','6',' ');

if( !cap.set(cv::CAP_PROP_FOURCC, (double)fourcc))
{
    cerr << "ERROR! Unable to to set fourcc\n";
    return -1;
}
```
The first line is the important one, because it is true, which seems to be default, you always get an RGB24 image.  Also v4l2 uses "GREY" instead "Y800", therefore OpenCV with using cv::CAP_V4L2 as API need "GREY".

Checking both FOURCCs, I receive the correct cv::Mat 8UC1 and 16UC1 formats.

Stefan

jaredwaller-git, 2022-02-09:
Thank you so much Stefan, I am now seeing the 8-bit image perfectly.

closed 2022-02-09 by jaredwaller-git