TheImagingSource/tiscamera #447: About Direct Memory Access

oka1125, 2022-02-09:
Hello,

I am using ''DFK37BUX273" camera connected to Jetson TX2 (OS: Ubuntu 18.04).
    The specs of this camera is as follows:
        interface=USB3.0
        width=1440
        height=1080
        fps=238
        And the gstreamer's command is as follows:
            gst-launch-1.0 tcambin ! video/x-raw,format=BGRx,width=1440,height=1080 ! videoconvert ! ximagesink

My c++ program (with gstreamer, tcambin (backend is v4l2) ) captures frames by 238 fps and saves these frames.
But this program sometimes drops frame once in around 1000 times, especially when the CPU load is high.

Does this tiscamera library supports DMA (Direct Memory Access) ?
    If yes, please let me know how to enable DMA.

Supplementary information:
    * v4l2 supports DMA transfer as written in following site.
            - https://www.kernel.org/doc/html/latest/userspace-api/media/v4l/dmabuf.html
    * Jetson TX2 supports DMA controller as written in following site.
            - https://docs.nvidia.com/jetson/l4t-multimedia/group__V4L2Conv.html

TIS-Stefan, 2022-02-09:
> Does this tiscamera library supports DMA (Direct Memory Access) ?

We do not use DMA, because the v4l2 driver should do so. But I did not look into the v4l2 source code, so I am not sure what happens. But: tiscamera uses user_ptr, not mmap, when using v4l2.  However, you pointed to the correct links already.

> If yes, please let me know how to enable DMA.**
I can not help here.

The USB data transfer is not guaranteed. That means, if the computer has high CPU load and the memory is blocked, then the USB Controller can not save all data packages into memory. That is a situation, we can not control and from our experiences, DMA access wont solve it too.
v4l2 recognizes this by checking the bytes used in the buffer. If the count of bytes does not exceed the the expected count, we drop the frame.  
Also install the tiscamera-dutils from https://www.theimagingsource.com/support/downloads-for-linux/, if not done already. They have a much faster debayering than Gstreamer's bayer2rgb. That reduces CPU load too. 

You may use xvimagesink instead of ximagesink, because then the video is rendered be the graphics board and you have less CPU load. 

I do not dare to write... but you may avoid high CPU load or use a slightly lower frame rate.

Stefan

oka1125, 2022-02-14:
Dear Mr. Stefan.

I'm sorry for late reply.
And thank you so much for the information.

> The USB data transfer is not guaranteed. That means, if the computer has high CPU load and the memory is blocked, then the USB Controller can not save all data packages into memory. That is a [...]

I understand that frame drop won't be solved by DMA access.
So I will close this issue.

> You may use xvimagesink instead of ximagesink, because then the video is rendered be the graphics board and you have less CPU load.

I'm developing products which use TISCAMERA.
For the products, I'm using C++ with Gstreamer Appsink to get the tcamsrc's frames.
So I'm happy if your company to support DMA Access for GPU (especially Jetson) in the future.

Thank you.

closed 2022-02-14 by TIS-Stefan