TheImagingSource/tiscamera #448: Automatic Color correction for tcam capture?

vinit13792, 2022-02-15:
Hi,

Question1: Is there a chance the The Imaging Source would be interested in developing automatic color correction algorithm in its camera software suite?

Question2: If one wants to integrate the automatic color correction algorithm in tcam-capture, how is it possible?

TIS-Stefan, 2022-02-15:
Hello

Most cameras have a color correction matrix. The use of this matrix can be enabled, if the cameras are used with USB3Vision and GigEVision by the aravis library.  If tiscamera's cmake is called wit -DTCAM_ARAVIS_USB_VISION=ON, then the camera can be accessed by USB3Vision instead of v4l2. In order to let tiscamera know, that USB3Vision should be used, you add "-aravis" to the serial number passed to the tcambin.  tcam-capture will show the USB cameras twice and selecting the Aravis version of the camera you will have a different and bigger set of camera properties, than using v4l2.
This is currently with tiscamera 0.14. 
You may try that.

Stefan

vinit13792, 2022-02-15:
> 

Hi Mr.Stefan,

I built tiscamera's cmake with -DTCAM_ARAVIS_USB_VISION=ON. But i still have only v4l2 in the device list of tcam-capture.

TIS-Stefan, 2022-02-15:
Did you also use -DBUILD_ARAVIS=ON ?
You may refer to https://www.theimagingsource.com/documentation/tiscamera/building.html for the cmake paramters.

Stefan
(its not "Mr", because "Stefan" is my surname :-) )

vinit13792, 2022-02-15:
> Did you also use -DBUILD_ARAVIS=ON ? You may refer to https://www.theimagingsource.com/documentation/tiscamera/building.html for the cmake paramters.
> 
> Stefan (its not "Mr", because "Stefan" is my surname :-) )

Hi,

Yes, I have also built with DBUILD_ARAVIS=ON.

What should I call you then?

TIS-Stefan, 2022-02-15:
What should I call you then?
"Stefan" is just sufficioent. You are welcome!

Which camera model do you use? USB3Vision is supported by USB3 cameras only. Not on the USB 2.0 cameras. 

Stefan

vinit13792, 2022-02-15:
> What should I call you then? "Stefan" is just sufficioent. You are welcome!
> 
> Which camera model do you use? USB3Vision is supported by USB3 cameras only. Not on the USB 2.0 cameras.
> 
> Stefan

Hi Stefan,

I use DFK 37BUX178 6.3MP color camera.

TIS-Stefan, 2022-02-15:
The DFK 37BUX178 should be shown with tcam-ctrl -l twice. After you change the cmake parameters as mentioned above, building and installing tiscamera. 

Stefan

vinit13792, 2022-02-15:
> The DFK 37BUX178 should be shown with tcam-ctrl -l twice. After you change the cmake parameters as mentioned above, building and installing tiscamera.
> 
> Stefan

Yes, now it shows two options but it doesnt show any options show the video stream and any properties after selecting aravis.
 
[image]

TIS-Stefan, 2022-02-15:
Hello

this is weird.
Please reboot your computer. Then enter on the terminal

gst-launch-1.0 serial=10910616-aravis ! videoconvert ! ximagesink. 

Does this show an image?

Stefan

vinit13792, 2022-02-16:
> 

Hi Stefan,

I did reboot my computer and i get the following output:

`WARNING: erroneous pipeline: syntax error`

TIS-Stefan, 2022-02-16:
Hello

Sorry, my fault

I missed the "tcambin"

gst-launch-1.0 tcambin serial=10910616-aravis ! videoconvert ! ximagesink.

Stefan

vinit13792, 2022-02-16:
> 

Hi Stefan,

After making changes, i get the following output:

`Setting pipeline to PAUSED ...`
`ERROR: Pipeline doesn't want to pause.`
`Setting pipeline to NULL ...`
`Freeing pipeline ...`

vinit13792, 2022-02-16:
> 

Also when i select aravis from the list. The video stream is very slow and doesnt change realtime like v4l2. Frames per second is 30/1, but in the video its much slower than 1.

TIS-Stefan, 2022-02-16:
We know, that aravis is working. Now it is time to try tcam-capture again and seen, what is happening. Maybe we can solve the speed issue. On my computer it works fast.

vinit13792, 2022-02-16:
> 

Can you tell what is your computer specs?

TIS-Stefan, 2022-02-16:
Yes. It is an old Core i5 with Ubuntu 20.04.
Does tcam-capture show your camera with Aravis now?

vinit13792, 2022-02-17:
> Yes. It is an old Core i5 with Ubuntu 20.04. Does tcam-capture show your camera with Aravis now?

Yes, it does show, but the video stream seems very slow. i have set it to 60FPS, resolution is 3072x2048. The change in video is registered and shown at a very slow frames

TIS-Stefan, 2022-02-17:
Hmm. Did you check the exposure time? It should not be too long. 

Stefan

vinit13792, 2022-02-17:
> Hmm. Did you check the exposure time? It should not be too long.
> 
> Stefan

Hi Stefan,

This is the current exposure setting.
[image]

vinit13792, 2022-02-17:
Hi Stefan,

i reduced the exposure time, and the app stopped working. And it showed the not responding message, so i clicked on Force Quit. After re-opening the app, it shows the following:
[image]

TIS-Stefan, 2022-02-18:
I give up. I do not get as much images as expected from the camera using Aravis. Only 50 instead of 60 fps. However, tcam-capture will hang, if you change the frame rate. I do not know, why.

Therefore, use stay with v4l2 

Stefan

TIS-Edgar, 2024-04-12:
Closed due to inactivity.

closed 2024-04-12 by TIS-Edgar