TheImagingSource/tiscamera #449: gi.repository.GLib.Error: gst_parse_error: no element "Collection" (1)

vinit13792, 2022-02-15:
System specs:
Ubuntu 20.04
8GB RAM
DFK 37BUX178 6.3MP color camera.

I installed from source using all the instructions mentioned [here](https://github.com/TheImagingSource/tiscamera).

When i "Save Image" using tcam-capture, I get the following error:

`(tcam-capture:57615): GStreamer-CRITICAL **: 16:38:06.206: gst_element_make_from_uri: assertion 'gst_uri_is_valid (uri)' failed`
`Traceback (most recent call last):`
`  File "/home/ubuntu/tiscamera/tools/tcam-capture/tcam-capture", line 420, in save_image_action`
`    self.view.save_image(self.settings.get_image_type())`
`  File "/home/ubuntu/tiscamera/tools/tcam-capture/tcam_capture/TcamView.py", line 190, in save_image`
`    self.imagesaver.save_image(get_encoder_dict()[image_type])`
`  File "/home/ubuntu/tiscamera/tools/tcam-capture/tcam_capture/MediaSaver.py", line 246, in save_image`
`    self._create_pipeline()`
`  File "/home/ubuntu/tiscamera/tools/tcam-capture/tcam_capture/MediaSaver.py", line 146, in _create_pipeline`
`    self.pipeline = Gst.parse_launch(save_str)`
`gi.repository.GLib.Error: gst_parse_error: no element "Collection" (1)`

TIS-Stefan, 2022-02-15:
Hello

Maybe tcam-capture does not find any Gstreamer encoder modules. Which is strange. Please run
gst-inspect-1.0 pngenc. It should show the properties of the png encode.

Stefan

vinit13792, 2022-02-16:
> Hello
> 
> Maybe tcam-capture does not find any Gstreamer encoder modules. Which is strange. Please run gst-inspect-1.0 pngenc. It should show the properties of the png encode.
> 
> Stefan

Hi Stefan,

I get the following output after running `gst-inspect-1.0 pngenc`:

`Factory Details:`
`  Rank                     primary (256)`
`  Long-name                PNG image encoder`
`  Klass                    Codec/Encoder/Image`
`  Description              Encode a video frame to a .png image`
`  Author                   Jeremy SIMON <jsimon13@yahoo.fr>`

`Plugin Details:`
`  Name                     png`
`  Description              PNG plugin library`
`  Filename                 /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstpng.so`
`  Version                  1.16.2`
`  License                  LGPL`
`  Source module            gst-plugins-good`
`  Source release date      2019-12-03`
`  Binary package           GStreamer Good Plugins (Ubuntu)`
`  Origin URL               https://launchpad.net/distros/ubuntu/+source/gst-plugins-good1.0`

`GObject`
` +----GInitiallyUnowned`
`       +----GstObject`
`             +----GstElement`
`                   +----GstVideoEncoder`
`:

TIS-Stefan, 2022-02-16:
Hello

Lets see, whether you get an image at all in your other support case

Stefan

TIS-Stefan, 2022-02-16:
If you look into the source code of tcam-capture you find in /MediaSaver.py"

```
   self.encoder_dict = get_encoder_dict()
  self.image_encoder = self.encoder_dict["png"]
```
get_encoder_dict() uses:
```
    if Gst.ElementFactory.find("pngenc") is not None:
        encoder_dict["png"] = Encoder("png", "pngenc", MediaType.image, "png")
``` 
I conclude, that tcam-capture can not find pngenc. Therefore, It may is possible, that tcam-capture runs in a different environment, than gst-inspect. In case you use Anaconda or Pythonbox(?), you must add the GStreamer search paths to these environments. 

Stefan

vinit13792, 2022-02-16:
Hi Stefan,

I actually formatted the system, and re-installed Ubuntu and tiscamera, it started working. I guess there was some issue with the path

vinit13792, 2022-02-22:
Hi Stefan,

This is recurring issue, formatting a OS cannot be an option always. Is there any particular reason why this keeps happening?

TIS-Stefan, 2022-02-22:
You are nearly the only one with this issue. Therefore, I wonder, what is different on your computers compared to most others.

Do you use  Anaconda or Pythonbox?
Do you use a Docker container?
Did you change the standard path of Gstreamer etc?

Stefan

vinit13792, 2022-02-22:
> Do you use Anaconda or Pythonbox?

Hi Stefan,

Do you use Anaconda or Pythonbox? No

Do you use a Docker container? No

Did you change the standard path of Gstreamer etc? How to find the gstreamer path? and how to add it to path?

Since it worked once, what did you change, so it does not work anymore? I uninstalled the tiscamera, and reinstalled, and then this error ocurred.

TIS-Stefan, 2022-02-22:
> I am using ubuntu 20.04, followed tiscamera install guide on github There was no environment of anaconda involved here. This error occurred and i re-installed the tiscamera after uninstalling it from build directory

Did that work?

Stefan

vinit13792, 2022-02-22:
> > I am using ubuntu 20.04, followed tiscamera install guide on github There was no environment of anaconda involved here. This error occurred and i re-installed the tiscamera after uninstalling it from build directory
> 
> Did that work?
> 
> Stefan

Hi Stefan,

It didnt work.

TIS-Stefan, 2022-02-22:
But you wrote, it worked once at https://github.com/vinit13792. Therefore, what changed since then?

Stefan

vinit13792, 2022-02-23:
> But you wrote, it worked once at https://github.com/vinit13792. Therefore, what changed since then?
> 
> Stefan

So to make aravis work, i tried uninstalling, and then re-installed tiscamera again. And since then this error shows up everytime i try to click an image.

TIS-Stefan, 2022-02-23:
Please show me your cmake line. It should be 
`
cmake -DBUILD_ARAVIS=ON  -DTCAM_ARAVIS_USB_VISION=ON ..
`
Then rebuild and run sudo make install.

(Source : https://www.theimagingsource.com/documentation/tiscamera/building.html)

Stefan

vinit13792, 2022-02-23:
> Please show me your cmake line. It should be `cmake -DBUILD_ARAVIS=ON -DTCAM_ARAVIS_USB_VISION=ON ..` Then rebuild and run sudo make install.
> 
> (Source : https://www.theimagingsource.com/documentation/tiscamera/building.html)
> 
> Stefan

Hi Stefan,

I actually did a fresh ubuntu install, and then installed tiscamera using the tutorial mentioned [here].(https://www.theimagingsource.com/documentation/tiscamera/tutorial.html)

vinit13792, 2022-02-23:
> Please show me your cmake line. It should be `cmake -DBUILD_ARAVIS=ON -DTCAM_ARAVIS_USB_VISION=ON ..` Then rebuild and run sudo make install.
> 
> (Source : https://www.theimagingsource.com/documentation/tiscamera/building.html)
> 
> Stefan

Hi Stefan,

I just rebooted my computer, and i got the same error. `gi.repository.GLib.Error: gst_parse_error: no element "Collection" (1)`

So i followed the cmake line as you suggested, and i got the following:

There were two errors at the end.

`[  0%] Performing configure step for 'aravis'
`autoreconf: Entering directory `.'`
`autoreconf: configure.ac: not using Gettext
`autoreconf: running: aclocal --force -I m4`
`autoreconf: configure.ac: tracing`
`autoreconf: running: libtoolize --copy --force`
`libtoolize: putting auxiliary files in '.'.`
`libtoolize: copying file './ltmain.sh'`
`libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
`libtoolize: copying file 'm4/libtool.m4'
`libtoolize: copying file 'm4/ltoptions.m4'
`libtoolize: copying file 'm4/ltsugar.m4'
`libtoolize: copying file 'm4/ltversion.m4'
`libtoolize: copying file 'm4/lt~obsolete.m4'
`autoreconf: running: /usr/bin/autoconf --force`
`autoreconf: running: /usr/bin/autoheader --force`
`autoreconf: running: automake --add-missing --copy --force-missing`
`configure.ac:31: installing './compile'
`configure.ac:10: installing './missing'
`gst-0.10/Makefile.am: installing './depcomp'
`autoreconf: Leaving directory `.'
`checking for a BSD-compatible install... /usr/bin/install -c`
`checking whether build environment is sane... yes`
`checking for a thread-safe mkdir -p... /usr/bin/mkdir -p`
`checking for gawk... no`
`checking for mawk... mawk`
`checking whether make sets $(MAKE)... yes`
`checking whether make supports nested variables... yes`
`checking whether to enable maintainer-specific portions of Makefiles... no`
`checking whether make supports nested variables... (cached) yes`
`checking whether make supports the include directive... yes (GNU style)`
`checking for gcc... gcc`
`checking whether the C compiler works... yes`
`checking for C compiler default output file name... a.out`
`checking for suffix of executables... `
`checking whether we are cross compiling... no`
`checking for suffix of object files... o`
`checking whether we are using the GNU C compiler... yes`
`checking whether gcc accepts -g... yes`
`checking for gcc option to accept ISO C89... none needed`
`checking whether gcc understands -c and -o together... yes`
`checking dependency style of gcc... none`
`checking for library containing strerror... none required`
`checking for gcc... (cached) gcc`
`checking whether we are using the GNU C compiler... (cached) yes`
`checking whether gcc accepts -g... (cached) yes`
`checking for gcc option to accept ISO C89... (cached) none needed`
`checking whether gcc understands -c and -o together... (cached) yes`
`checking dependency style of gcc... (cached) none`
`checking for g++... g++`
`checking whether we are using the GNU C++ compiler... yes`
`checking whether g++ accepts -g... yes`
`checking dependency style of g++... none`
`checking how to run the C preprocessor... gcc -E`
`checking for grep that handles long lines and -e... /usr/bin/grep`
`checking for egrep... /usr/bin/grep -E`
[... 170 lines cut ...]
`  128 | }`
`      | ^`
`/home/ubuntu/tiscamera/external/aravis/src/arvuvcp.c: In function ‘arv_uvcp_command_to_string’:`
`/home/ubuntu/tiscamera/external/aravis/src/arvuvcp.c:134:1: warning: control reaches end of non-void function [-Wreturn-type]
`  134 | }`
`      | ^`
`make[5]: *** [Makefile:952: arvuvcp.lo] Error 1`
`make[5]: *** Waiting for unfinished jobs....`
`/home/ubuntu/tiscamera/external/aravis/src/arvuvinterface.c: In function ‘arv_uv_interface_init’:`
`/home/ubuntu/tiscamera/external/aravis/src/arvuvinterface.c:369:13: warning: Deprecated pre-processor symbol, replace with 
  369 |  uv_interface->priv = G_TYPE_INSTANCE_GET_PRIVATE (uv_interface, ARV_TYPE_UV_INTERFACE, ArvUvInterfacePrivate);`
`      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~`
`/home/ubuntu/tiscamera/external/aravis/src/arvuvdevice.c: In function ‘arv_uv_device_init’:`
`/home/ubuntu/tiscamera/external/aravis/src/arvuvdevice.c:748:13: warning: Deprecated pre-processor symbol, replace with 
`  748 |  uv_device->priv = G_TYPE_INSTANCE_GET_PRIVATE (uv_device, ARV_TYPE_UV_DEVICE, ArvUvDevicePrivate);`
`      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
`/home/ubuntu/tiscamera/external/aravis/src/arvuvstream.c: In function ‘arv_uv_stream_init’:
`/home/ubuntu/tiscamera/external/aravis/src/arvuvstream.c:418:13: warning: Deprecated pre-processor symbol, replace with 
  418 |  uv_stream->priv = G_TYPE_INSTANCE_GET_PRIVATE (uv_stream, ARV_TYPE_UV_STREAM, ArvUvStreamPrivate);`
`      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~`
`make[4]: *** [Makefile:702: all] Error 2`
`make[3]: *** [Makefile:583: all-recursive] Error 1`
`make[2]: *** [external/CMakeFiles/aravis.dir/build.make:113: external/aravis-prefix/src/aravis-stamp/aravis-build] Error 2`
`make[1]: *** [CMakeFiles/Makefile2:638: external/CMakeFiles/aravis.dir/all] Error 2`
`make: *** [Makefile:152: all] Error 2`

TIS-Stefan, 2022-02-23:
Hello

what happens here:

> /home/ubuntu/tiscamera/external/aravis/src/arvuvcp.c:133:29: error: ‘ARV_TYPE_UVCP_COMMAND’ undeclared (first use in this function); did you mean ‘ARV_TYPE_GVCP_COMMAND’?

Why is there an error, we do not get`?
what is the cmake command line you use?

vinit13792, 2022-03-08:
I give up.

I did a reset on Ubuntu system, updated my system. And then installed tiscamera. It worked well on day 1. Next day when I fired up my system, and clicked an image, it gave the same error. I do not understand why is this happening just over a reboot.

TIS-Stefan, 2022-03-08:
Hello

If a reboot destroys all, then the first question is, how did you install tiscamera?
How do you execute tcam-capture? From terminal or desktop?
what happens if open a terminal and enter
```
cd tiscamera/build
. ./env.sh
tcam-capture
```
You may ask yourself, what happens on reboot on your system, that does no happen on all other computers?
Do you have kind of reset program?
Do you run in a virtual machine?

Stefan

vinit13792, 2022-03-09:
> Hello
>
> If a reboot destroys all, then the first question is, how did you install tiscamera? How do you execute tcam-capture? From terminal or desktop? what happens if open a terminal and enter
>
> ``` [...]

how did you install tiscamera? with the help of instructions mentioned on Tiscamera github

How do you execute tcam-capture? using terminal

what happens if open a terminal and enter `cd tiscamera/build . ./env.sh tcam-capture` ? tcam-capture opens.

Do you have kind of reset program? No I do not.

Do you run in a virtual machine? No, its an old HP 7th gen 8gb laptop.

You may ask yourself, what happens on reboot on your system? I think there might be some disk errors. because before installing Ubuntu, the setup checks for disk errors, and there were some 6 errors found in disk checks. I think those disk checks has something to do withthis problem which is causing some important files to corrupt

vinit13792, 2022-03-09:
> If you look into the source code of tcam-capture you find in /MediaSaver.py"
>
> ```
>    self.encoder_dict = get_encoder_dict()
>   self.image_encoder = self.encoder_dict["png"]
> ```
>
> get_encoder_dict() [...]

you must add the GStreamer search paths to these environments... How to add Gstreamer search paths to environment?

TIS-Stefan, 2022-03-09:
Hi

> what happens if open a terminal and enter cd tiscamera/build . ./env.sh tcam-capture ? tcam-capture opens.

Do you get a live stream and are you able to save images?
the ". ./env.sh" sets include paths for the tiscamera files to your current build directory. 

> I think there might be some disk errors. because before installing Ubuntu, the setup checks for disk errors, and there were some 6 errors found in disk checks. I think those disk checks has something to do with this problem which is causing some important files to corrupt

I do not think, this could be an issue, but who knows? That could be possible. 

> How to add Gstreamer search paths to environment?

Use the GST_PLUGIN_PATH environment variable to point at the directory you want. Please see https://stackoverflow.com/questions/2120444/gstreamer-plugin-search-path
and https://gstreamer.freedesktop.org/documentation/gstreamer/gstregistry.html?gi-language=c

Stefan

vinit13792, 2022-03-09:
Hi Stefan,

I dont know the logic behind using gst plugin path and how that would make a difference since i am not installing anything else apart from tiscamera. The building from source does exactly the same thing. So I am not sure how this would make a difference.

vinit13792, 2022-03-09:
Also, I tried tcam-capture --reset, and the error disappeared and it started saving the image in the given folder.

TIS-Stefan, 2022-03-09:
Hi

> I dont know the logic behind using gst plugin path and how that would make a difference since i am not installing anything else apart from tiscamera.

Welcome to Linux. It is also about installing additional packages.  However, I am guessing. 

> Also, I tried tcam-capture --reset, and the error disappeared and it started saving the image in the given folder.

Where to go from here? I have no more ideas. 

Stefan

vinit13792, 2022-03-09:
> Hi
>
>
> Welcome to Linux. It is also about installing additional packages. However, I am guessing.
>
>
> Where to go from here? I have no more ideas.
>
> Stefan

`Welcome to Linux. It is also about installing additional packages. However, I am guessing.` I agree I am new to Linux, but i thought gstreamer was being installed as a dependency with tiscamera. 

`Where to go from here? I have no more ideas.` Well i think anyone who gets this similar error can do a `tcam-capture --reset` and check it out themselves if it works for them or not. We can close this issue for now.

TIS-Stefan, 2022-03-09:
Until now, you are the only one with this error. That makes it so hard to narrow down. 

Stefan

vinit13792, 2022-03-10:
Hi Stefan,

Yeah i guess you are right, because I searched for the same issue and i couldn't find anything anywhere. So I think its better if its open.

Vinit

TIS-Stefan, 2022-03-10:
Hi Vinit

ok. 

What do you want to do with the camera? Maybe programs like "qv4l2" are sufficient for you.

Stefan

vinit13792, 2022-03-10:
Hi Stefan,

I am working on a instance segmentation problem. I need the camera to click and save a color accurate image and save it in a folder. I want to take it a step further by automating this process soon by sending software triggers. But for software triggers i read in your documentation that all the automatics need to be `Turned OFF`. Hence i may have to just stream the video with all the automatics enabled and snap an image from the video stream. 

Could you please elaborate on the qv4l2 part? i didn't understand what that refers to.

Vinit

TIS-Stefan, 2022-03-10:
Hello Vinit

qv4l2 is a program. Install with 
sudo apt install qv4l2.

But it is a program, no SDK. It is just for testing and see, whether you can access the camera, even after a reboot.  

Stefan

vinit13792, 2022-03-10:
Hi Stefan,

So i shutdown my system yesterday, booted it today, and i am able to access camera. Snap and save an image using `tcam-capture`

Vinit

TIS-Stefan, 2022-03-10:
Hi Vinit

it seems, it is a good day to start programming!

Maybe you get some ideas from https://github.com/TheImagingSource/Linux-tiscamera-Programming-Samples

Stefan

vinit13792, 2022-03-10:
Yes. Definitely!!

Vinit

TIS-Edgar, 2024-04-12:
Closed due to inactivity.

closed 2024-04-12 by TIS-Edgar