TheImagingSource/tiscamera #451: Build fails with Fedora 35?

licorwork, 2022-02-16:
Hello,
I followed the cloning of tiscamera and installing what I thought are the needed dependencies, but still get the following errors running cmake. Any idea what I should try next? Mainly, I needed the tcam-uvc-extension-loader so I can access the full list of properties on dmk33ux174.

cmake ..
<string>:1: DeprecationWarning: The distutils package is deprecated and slated for removal in Python 3.12. Use setuptools or check PEP 632 for potential alternatives
<string>:1: DeprecationWarning: The distutils.sysconfig module is deprecated, use sysconfig instead
CMake Error at /usr/share/cmake/Modules/FindPkgConfig.cmake:603 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake/Modules/FindPkgConfig.cmake:825 (_pkg_check_modules_internal)
  cmake/modules/FindGObjectIntrospection.cmake:39 (pkg_check_modules)
  src/gobject/CMakeLists.txt:20 (find_package)

-- Configuring incomplete, errors occurred!

Thank you

TIS-Stefan, 2022-02-17:
Hello

you may look at https://github.com/TheImagingSource/tiscamera/issues/426. It is a similar issue.

Stefan

licorwork, 2022-02-17:
Hello Stefan.
Thank you for the feedback! I managed to install the "tcam-uvc-extension-loader" only so far. But, I need the extension file to load to the dmk33ux174 camera. Is that something you can send me?
Thank you, again.
Ahmed

TIS-Edgar, 2022-02-18:
Hi Ahmed,

You need three parts for the uvc-extension units.
1. tcam-uvc-extension-loader
2. the extension description
3. the udev file for automatic loading

All of these are included in tiscamera. You can find the configured udev file in your build folder under data/udev and the extensions under tiscamera/data/uvc-extensions. For your camera only the 33.json is relevant.

All of these files should be automatically installed. It may very well be that Fedora handles some paths differently, that I currently do not know.

If you really only need the uvc-extensions you can use the cmake flag `TCAM_BUILD_UVC_EXTENSION_LOADER_ONLY` and set it to `ON`. That way only the relevant parts will be built and installed.

licorwork, 2022-02-18:
Hi Edgar again.
I managed to get the remaining packages needed to build tiscamera and was able to use the following successfully (video displayed fine):

gst-launch-1.0 tcambin ! videoconvert ! ximagesin

But when I ran tcam-capture, I got the following "Type error"?!
Traceback (most recent call last):
  File "/usr/bin/tcam-capture", line 870, in <module>
    tcam_capture = TCamCapture(app, serial, caps_str, fullscreen)
  File "/usr/bin/tcam-capture", line 194, in __init__
    self.open_device(self.serial, self.device_type, caps_str)
  File "/usr/bin/tcam-capture", line 669, in open_device
    self.props_widget = PropertyDialog(self.data, self.view, self.props)
  File "/usr/lib/python3.10/site-packages/tcam_capture/PropertyDialog.py", line 159, in __init__
    self.setup_ui()
  File "/usr/lib/python3.10/site-packages/tcam_capture/PropertyDialog.py", line 217, in setup_ui
    self.tab_dict[category].add_property(prop)
  File "/usr/lib/python3.10/site-packages/tcam_capture/PropertyDialog.py", line 58, in add_property
    self.prop_dict[prop.name] = PropertyWidget(self.data, prop)
  File "/usr/lib/python3.10/site-packages/tcam_capture/PropertyWidget.py", line 56, in __init__
    self.setup_ui()
  File "/usr/lib/python3.10/site-packages/tcam_capture/PropertyWidget.py", line 183, in setup_ui
    self.__setup_ui_integer()
  File "/usr/lib/python3.10/site-packages/tcam_capture/PropertyWidget.py", line 104, in __setup_ui_integer
    self.sld.setValue(self.prop.value)
  File "/usr/lib/python3.10/site-packages/tcam_capture/TcamSlider.py", line 122, in setValue
    super().setValue(self.calc_log_val(value))
**TypeError: setValue(self, int): argument 1 has unexpected type 'float'**

TIS-Edgar, 2022-02-21:
> Thank you for the feedback. I build with the "tcam-uvc-extension-loader only" option successfully and loaded 33.json file. I'm not I needed to do that because even before loading the file, I did get all controls to show up when I do:v4l2-ctl -d /dev/video0 --list-ctrls. 

UVC cameras will always offer basic standard properties like gain or exposure. The extension unit is there to offer additional properties that are not in the UVC standard (e.g. Trigger Polarity). Are you sure you are seeing all properties and not just the basic ones? Doesn't really matter if you have access to all properties you need, I just want to avoid confusion here.

> I can change the format to "Y16", but the values I get still have a ceiling of 255 ?! Don't know if that's expected with that kind of camera or not?

What properties are your talking about? Properties tend to have one range only and map internally should that be necessary.
This is to ensure that user applications don't have to be more complicated than necessary.

> But when I ran tcam-capture, I got the following "Type error"?!

I am trying to reproduce this problem, but I am running into different tcam-capture problems, so I am unsure if I can offer a quick solution for that. Is this a blocking issue for you?

licorwork, 2022-02-21:
Sorry for the confusion and Thank you for your patience!
I'd like to provide further details about what I did and the problem I thought I ran into... When I first connected the camera and ran "v4l2-ctl --list-ctrls:, I got the following:
v4l2-ctl --list-ctrls --device /dev/video0
[image]
When I used Octave, I got 15 properties/controls:
[image]

And since I needed to image with low-light levels, I changed brightness, gamma, and gain to low values. All seemed to be accepted fine except gain. It gets accepted initially, but when I start streaming, it reverts back up to a higher gain value?! Also, the exposure_time_absolute has a minimum of 1, which I think has steps of 100us (I think I saw that somewhere and I can't remember where now?). So, setting to 1 was not low enough in my case, especially since gain kept kicking back up to 479 or so?! So, maybe my first question should have been about that: What can I do to make gain stick to value of 0 or so and how I can get lower exposure than 100us (I believe the camera allows down to 30ums?!)?
And because I ran into the same issue few years ago and went through a similar exercise where, with Stefan's help and/or stumbling on some tools/utilities, I was able to gain access to extended list of properties that did accept low gain values and "exposure_time_us" which allowed me to set exposures less than 100us.
So, this time around, as I mentioned in my previous posting, I stumbled gain on some utilities that allowed v4l2 to access the usb-extensions and now I can see some 51 properties/controls that include the previous 15 +  "exposure_time_us", "auto_shutter", and "gain_auto" which together gave me what I needed. Please see the following capture:
[image]

licorwork, 2022-02-21:
Yeah, that rings a bell! the linuxtv.org postings! Thanks for pointing that out. I'll read through it again and remember it for future needs. 
I can set auto exposure and makes sense to me. But, is there I could set "gain_auto" to 0 without the accessing the extended list? Maybe that would been enough to solve my problem?!
Thank you again, Edgar, for your help!

TIS-Edgar, 2022-02-22:
If you want to configure properties in the extended list there is no way around having that list loaded. When tcam-uvc-extension-loader is correctly installed the correlated udev rules file should ensure that the extension unit is automatically enabled every time the camera is connected.

licorwork, 2022-02-22:
Thank you. That explains what actually caused the extended list to show up. I did a bunch of updates and installed the dependencies that tiscamera needed and then compiled the tcam-uvc-extension-loader only and after that I noticed the extended list of controls. I now see that a "/etc/udev/rules.d/80-theimagingsource-cameras.rules" gets created and it points to the json files. Thank you, this is very helpful. The camera works fine now.
Best regards!

closed 2022-02-22 by licorwork