TheImagingSource/tiscamera #452: zoom in QT

HabermannR, 2022-02-21:
Hi, in 
https://github.com/TheImagingSource/tiscamera/blob/master/tools/tcam-capture/tcam_capture/TcamScreen.py
you show how you can zoom into a live picture.
I try to add this to your stereo example:
https://github.com/TheImagingSource/IC-Imaging-Control-Samples/blob/master/Python/tisgrabber/samples/40-qt-stereo.py
but I do not understand qt enough.
For me it seems I need to use a QGraphicsView:
#LeftVideo = qt5.QWidget()
LeftVideo = PyQt5.QtWidgets.QGraphicsView()
And I do not get an error when I change the scale, but nothing happens.
Do I need to map the live picture to something else?
Where is this in tcam_capture?
Thanks!

TIS-Stefan, 2022-02-21:
Hello

That is a question for QT programmers, not a question for the camera. Therefore, I can am not really of help here. I do not even know, what you mean by "change the scale". I looked at Stackoverflow for how to show an image in a Widget, without going deeper into that QT graphics possibilities. 
I am sorry for not being the QT expert.

Stefan

HabermannR, 2022-02-21:
Thanks for the quick answer.
I just want to zoom into the picture in order to focus manually.

I think what happens is that qt just takes the camera feed and overlays it into its widget and thus cannot scale/zoom it.
# Pass the window handles to the left and right camera.
        ic.IC_SetHWnd(LeftCamera,  LeftVideo.winId())

TIS-Stefan, 2022-02-21:
Hello

> I think what happens is that qt just takes the camera feed and overlays it into its widget and thus cannot scale/zoom it.

This is correct. 
That means, you must get the images, convert into a qpixmap or how it is named and display it on your own. Using the callback is tricky. because the callback runs in a different thread, than the main program. But the sample https://github.com/TheImagingSource/IC-Imaging-Control-Samples/blob/master/Python/tisgrabber/samples/41-qt-triggering.py shows, how to convert the image to a cv::Mat and display it in the main thread. Maybe that is a good start.

I suppose, you use Python in Windows, right? Not in Linux?

Stefan

HabermannR, 2022-02-21:
Thank you both very much!
If I fail, I can still use your software to zoom and focus, and then use the python digital app to take numerated pictures.

TIS-Stefan, 2022-02-21:
ok. Please let me know, when you fail. Maybe I can find some spare time for the QT programming.

TIS-Edgar, 2024-04-12:
Closed due to inactivity.

closed 2024-04-12 by TIS-Edgar