TheImagingSource/tiscamera #455: tis-capture Camera driver locks ubuntu 20.04

midjji, 2022-03-01:
So I have had this problem for a long time, but could not share the code or isolate it. Fortunately the problem spread to your sample applications. 

Running tis-capture connects to the cameras and they work fine. 
Shutting down tis-capture, then running it again, and the cameras wont start. 
Problem might be that gstreamer cant change from ready to play(but I am not sure). 
Disconnecting the camera and reconnecting it, then allows tis-capture to work as normal. 

Install the latest version of this repo, then run tis-capture, select a camera, then see an image, then shut down the program, then start it again. it will find the camera, but it wont connect to it. 
Using the latest version of the library, and default repo dependences.

TIS-Stefan, 2022-03-01:
You can start
tcam-capture --reset. 
Then tcam-capture does not try to restore the last used camera. 

Stefan

midjji, 2022-03-01:
Tried it, 
tcam-capture  
select camera, show video, 
tcam-capture --reset, 
select camera, 
show video (hangs)

Tried it,
gst-launch-1.0 tcambin ! videoconvert ! xvimageink
Shows video,
close by x or ctrl-c, 
gst-launch-1.0 tcambin ! videoconvert ! xvimageink
just shows black image: 
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
INFO:
Working with src caps: video/x-raw, framerate=(fraction)75/1, width=(int)2048, height=(int)1536, format=(string)GRAY8
Setting pipeline to PLAYING ...
New clock: GstSystemClock
ERROR: from element /GstPipeline:pipeline0/GstTcamBin:tcambin0/GstTcamSrc:tcambin-source/GstTcamMainSrc:tcamsrc-mainsrc: Internal data stream error.
Additional debug info:
gstbasesrc.c(3072): gst_base_src_loop (): /GstPipeline:pipeline0/GstTcamBin:tcambin0/GstTcamSrc:tcambin-source/GstTcamMainSrc:tcamsrc-mainsrc:
streaming stopped, reason not-negotiated (-4)
Execution ended after 0:00:05.181811769
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXxxxx
gst-launch-1.0 tcambin ! videoconvert ! xvimageink
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
INFO:
Working with src caps: video/x-raw, framerate=(fraction)75/1, width=(int)2048, height=(int)1536, format=(string)GRAY8
Setting pipeline to PLAYING ...
New clock: GstSystemClock
ERROR: from element /GstPipeline:pipeline0/GstTcamBin:tcambin0/GstTcamSrc:tcambin-source/GstTcamMainSrc:tcamsrc-mainsrc: Internal data stream error.
Additional debug info:
gstbasesrc.c(3072): gst_base_src_loop (): /GstPipeline:pipeline0/GstTcamBin:tcambin0/GstTcamSrc:tcambin-source/GstTcamMainSrc:tcamsrc-mainsrc:
streaming stopped, reason not-negotiated (-4)
Execution ended after 0:00:05.079143910
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX

Tried unplugging the cameras and plugging them back in, 
gst-launch-1.0 tcambin ! videoconvert ! xvimageink
Same error, needed to reboot to fix it. 

I have two cameras connected, but trying with just one makes no difference or so it seems.

TIS-Stefan, 2022-03-02:
I forgot to ask, which camera model you use, I am sorry for that.
Also, please let me know, which computer model you use.

The error is " not-negotiated (-4)". That means, that somewhere the format provided by the camera could not be converted to what the other GStreamer modules needs.

You may install qv4l2 as simple video for Linux application. That must work anyway.

TIS-Stefan, 2022-03-02:
We may try some more debugging on your computer. Please run

```
export TCAM_LOG=DEBUG
gst-launch-1.0 --gst-debug=tcamsrc:5,tcambin:5 tcambin ! videoconvert ! xvimageink
```
With some luck, it gives more information.

midjji, 2022-03-15:
Hi,
Appending the output from the first time when it worked, then the second
when it hangs,

0:00:00.009707082  5191 0x55ae8bc23f30 DEBUG                tcambin ***@***.***> init
Setting pipeline to PAUSED ...
0:00:00.015706628  5191 0x55ae8bc23f30 INFO                 tcambin gsttcambin.cpp:1149:gst_tcam_bin_change_state: NULL_TO_READY
0:00:00.015714697  5191 0x55ae8bc23f30 DEBUG                tcambin gsttcambin.cpp:421:gst_tcambin_create_source: Creating source...
0:00:00.016058907  5191 0x55ae8bc23f30 DEBUG                tcamsrc gsttcamsrc.cpp:791:gst_tcam_src_change_state: State change: NULL -> READY
2022.03.15T18:24:23:635        <DEBUG> Logger:108: 
The following library versions are used:
	Tcam:	0.14.0_master/baacb4cf_rev_3059
	Aravis:	not_configured_version_not_configured
2022.03.15T18:24:23:788        <INFO> V4l2Device.cpp:410: Current framerate is 30 / 1 fps
2022.03.15T18:24:23:806        <WARNING> v4l2_utils.cpp:145: Unable to find std property. Passing raw property identifier through. 'Exposure Time (us)'(199e201)
2022.03.15T18:24:23:820        <INFO> V4l2Device.cpp:1190: Encountered write only control.
2022.03.15T18:24:23:888        <WARNING> v4l2_utils.cpp:145: Unable to find std property. Passing raw property identifier through. 'Trigger Delay (100ns)'(199e272)
2022.03.15T18:24:23:888        <INFO> V4l2Device.cpp:859: Detected multiple exposure interfaces. Simplifying
2022.03.15T18:24:23:888        <DEBUG> V4l2Device.cpp:693: Found format: Y800
2022.03.15T18:24:23:888        <DEBUG> V4l2Device.cpp:693: Found format: Y16
2022.03.15T18:24:23:888        <DEBUG> V4l2Device.cpp:243: Returning 2 formats.
2022.03.15T18:24:23:888        <DEBUG> PipelineManager.cpp:127: Received 2 formats.
2022.03.15T18:24:23:888        <DEBUG> PropertyHandler.cpp:273: Created 38 properties
0:00:00.269324147  5191 0x55ae8bc23f30 DEBUG                tcamsrc gsttcamsrc.cpp:699:open_source_element: Ghost pad target set
0:00:00.269426119  5191 0x55ae8bc23f30 INFO                 tcamsrc gsttcamsrc.cpp:772:open_source_element: Opened device has serial: '27810456' type: 'v4l2'
0:00:00.269450646  5191 0x55ae8bc23f30 INFO                 tcamsrc gsttcamsrc.cpp:800:gst_tcam_src_change_state: Opened source element
0:00:00.269545793  5191 0x55ae8bc23f30 INFO                 tcambin gsttcambin.cpp:460:gst_tcambin_create_source:<tcambin0> Opened device has serial: '27810456' type: 'v4l2'
0:00:00.269766560  5191 0x55ae8bc23f30 INFO                 tcambin gsttcambin.cpp:463:gst_tcambin_create_source:<tcambin0> caps of src: video/x-raw, format=(string)GRAY8, width=(int)640, height=(int)480, framerate=(fraction){ 600/1, 480/1, 5000000/20833, 120/1, 60/1, 30/1, 15/1, 5/1, 1/1 }; video/x-raw, format=(string)GRAY8, width=(int)1920, height=(int)1080, framerate=(fraction){ 180/1, 120/1, 60/1, 30/1, 15/1, 5/1, 1/1 }; video/x-raw, format=(string)GRAY8, width=(int)2048, height=(int)1536, framerate=(fraction){ 75/1, 60/1, 30/1, 15/1, 5/1, 1/1 }; video/x-raw, format=(string)GRAY16_LE, width=(int)640, height=(int)480, framerate=(fraction){ 600/1, 480/1, 5000000/20833, 120/1, 60/1, 30/1, 15/1, 5/1, 1/1 }; video/x-raw, format=(string)GRAY16_LE, width=(int)1920, height=(int)1080, framerate=(fraction){ 90/1, 60/1, 30/1, 15/1, 5/1, 1/1 }; video/x-raw, format=(string)GRAY16_LE, width=(int)2048, height=(int)1536, framerate=(fraction){ 35/1, 30/1, 15/1, 5/1, 1/1 }
0:00:00.273350513  5191 0x55ae8bc23f30 INFO                 tcambin gsttcambin.cpp:583:gst_tcambin_create_elements: creating elements
0:00:00.274636767  5191 0x55ae8bc23f30 DEBUG                tcambin gsttcambin.cpp:562:create_and_add_element: Adding bayer2rgb(0x55ae8bcc6140) to pipeline
0:00:00.274748070  5191 0x55ae8bc23f30 DEBUG                tcambin gsttcambin.cpp:562:create_and_add_element: Adding videoconvert(0x55ae8bcc4880) to pipeline
0:00:00.274982394  5191 0x55ae8bc23f30 INFO                 tcambin gsttcambin.cpp:1202:gst_tcam_bin_change_state: READY_TO_PAUSED
0:00:00.275518725  5191 0x55ae8bc23f30 INFO                 tcambin gsttcambin.cpp:1227:gst_tcam_bin_change_state:<tcambin0> caps of sink: video/x-raw, format=(string)YUY2, width=(int)[ 1, 32768 ], height=(int)[ 1, 32768 ], framerate=(fraction)[ 0/1, 2147483647/1 ]; video/x-raw, format=(string)YV12, width=(int)[ 1, 32768 ], height=(int)[ 1, 32768 ], framerate=(fraction)[ 0/1, 2147483647/1 ]; video/x-raw, format=(string)UYVY, width=(int)[ 1, 32768 ], height=(int)[ 1, 32768 ], framerate=(fraction)[ 0/1, 2147483647/1 ]; video/x-raw, format=(string)I420, width=(int)[ 1, 32768 ], height=(int)[ 1, 32768 ], framerate=(fraction)[ 0/1, 2147483647/1 ]; video/x-raw, width=(int)[ 1, 32768 ], height=(int)[ 1, 32768 ], framerate=(fraction)[ 0/1, 2147483647/1 ], format=(string){ I420, YV12, YUY2, UYVY, AYUV, VUYA, RGBx, BGRx, xRGB, xBGR, RGBA, BGRA, ARGB, ABGR, RGB, BGR, Y41B, Y42B, YVYU, Y444, v210, v216, Y210, Y410, NV12, NV21, GRAY8, GRAY16_BE, GRAY16_LE, v308, RGB16, BGR16, RGB15, BGR15, UYVP, A420, RGB8P, YUV9, YVU9, IYU1, ARGB64, AYUV64, r210, I420_10BE, I420_10LE, I422_10BE, I422_10LE, Y444_10BE, Y444_10LE, GBR, GBR_10BE, GBR_10LE, NV16, NV24, NV12_64Z32, A420_10BE, A420_10LE, A422_10BE, A422_10LE, A444_10BE, A444_10LE, NV61, P010_10BE, P010_10LE, IYU2, VYUY, GBRA, GBRA_10BE, GBRA_10LE, BGR10A2_LE, GBR_12BE, GBR_12LE, GBRA_12BE, GBRA_12LE, I420_12BE, I420_12LE, I422_12BE, I422_12LE, Y444_12BE, Y444_12LE, GRAY10_LE32, NV12_10LE32, NV16_10LE32, NV12_10LE40 }
[... cut ...]
2022.03.15T18:27:36:840        <WARNING> V4l2Device.cpp:1480: Did not receive image for long time.
2022.03.15T18:27:40:844        <ERROR> V4l2Device.cpp:1459: Timeout while waiting for new image buffer.
2022.03.15T18:27:44:846        <ERROR> V4l2Device.cpp:1459: Timeout while waiting for new image buffer.
2022.03.15T18:27:48:851        <ERROR> V4l2Device.cpp:1459: Timeout while waiting for new image buffer.
2022.03.15T18:27:52:855        <ERROR> V4l2Device.cpp:1459: Timeout while waiting for new image buffer.

midjji, 2022-03-15:
The cameras are:
dmk33ux252, but the problem also occurs on
dfk37bux178

I have noted that there are conflicts between the camera properties and the
gstreamer properties before, trigger bounce something which had a name
collision.
Also is there a way to get the frame number on linux? Preferably using a
proper C++ interface rather than the crummy v4l/gstreamer backend.

Something along the way is dropping frames on rare occasions, it does so
regardless of what settings I give gstreamer, etc.
though it seems more common if the load is high, in particular USB bus
load, but cpu load too,
and it makes it impossible to know which external trigger event then
corresponds to which frame. Hilariously this problem exact problem almost
crashed the mars drone, :p so its not just me, and not just your cameras
that suffer from it.

Alternatively, could you add a device mode where the top left 4 pixels are
overwritten with the id number?

Unrelated bugs, querying the camera shows that it supports binning, and a
mode can be set, but looking at the images that does not appear to be the
case.
This implies both that the device is incorrectly reporting its
capabilities, and that gstreamer is incorrectly swallowing the error which
the device should be outputting.

TIS-Stefan, 2022-03-16:
Hello

The debug output says, that there are simply no more images receives by v4l2. It seems, it is not a software related issue.

Sometime ago, I suggest to install qv4l2 program
sudo apt install qv4l2
and run it and check, whether the same happens. qv4l2 is a good test program for checking video for linux devices, such as cameras. 

Please le me know your results. 

Stefan

midjji, 2022-04-05:
Hi there,
yes, the exact same thing happens. No visible error message.

However, if I select close device I also get:
QSocketNotifier: Invalid socket 19 and type 'Exception', disabling...

willingtonvr, 2022-06-07:
Hello  same happening here.

At first time it works,  next time  have to shutoff the camera,
frequently  shutoff the camera doesn't work.

I have to  do the following:
open with gst-launch if it works then close it, power off the camera and power on.
open with tis-capture or my application.
test for another features ( migrating to the new API)

system info.

Camera DFK 33GX249; net.core.rmem_default=10485760; mtu = 1500 (not working with 9000)

gst-launch-1.0 --version
gst-launch-1.0 version 1.16.2
GStreamer 1.16.2
https://launchpad.net/distros/ubuntu/+source/gstreamer1.0

latest tiscamera

Thanks !

 log here
-----------------------------------------------------------------------------------------------------------------------
TCAM_LOG=DEBUG gst-launch-1.0 --gst-debug=tcamsrc:5,tcambin:5 tcambin ! videoconvert ! xvimagesink
Estableciendo el conducto a PAUSA …
0:00:00.035996826 17691 0x55a8fbc45990 INFO                 tcambin gsttcambin.cpp:364:select_transform_element: Auto selected transform element: tcamconvert
0:00:02.927995126 17691 0x55a8fbc45990 INFO                 tcamsrc gsttcamsrc.cpp:339:open_source_element:<tcambin-source> Opened device with serial: '45514186' type: 'aravis'
0:00:02.928069797 17691 0x55a8fbc45990 INFO                 tcambin gsttcambin.cpp:170:tcambin_create_source:<tcambin0> Opened device has serial: '45514186' type: 'aravis'
0:00:02.928284102 17691 0x55a8fbc45990 INFO                 tcambin gsttcambin.cpp:411:tcambin_create_elements:<tcambin0> creating all tcambin pipeline elements
0:00:02.929834944 17691 0x55a8fbc45990 DEBUG                tcambin gsttcambin.cpp:379:create_and_add_element:<tcambin0> Adding tcamconvert(0x55a8fbda0130) to pipeline
0:00:02.933674463 17691 0x55a8fbc45990 DEBUG                tcambin gsttcambin.cpp:530:tcambin_create_elements:<tcambin0> Internal pipeline: tcamsrc ! capsfilter ! tcamconvert
0:00:02.934009597 17691 0x55a8fbc45990 INFO                 tcambin gsttcambin.cpp:635:gst_tcam_bin_change_state:<tcambin0> caps of sink: video/x-raw, format=(string)YV12, width=(int)[ 1, 8192 ], height=(int)[ 1, 8192 ], framerate=(fraction)[ 0/1, 2147483647/1 ]; video/x-raw, format=(string)I420, width=(int)[ 1, 8192 ], height=(int)[ 1, 8192 ], framerate=(fraction)[ 0/1, 2147483647/1 ]; video/x-raw, width=(int)[ 1, 8192 ], height=(int)[ 1, 8192 ], framerate=(fraction)[ 0/1, 2147483647/1 ], format=(string){ I420, YV12, YUY2, UYVY, AYUV, VUYA, RGBx, BGRx, xRGB, xBGR, RGBA, BGRA, ARGB, ABGR, RGB, BGR, Y41B, Y42B, YVYU, Y444, v210, v216, Y210, Y410, NV12, NV21, GRAY8, GRAY16_BE, GRAY16_LE, v308, RGB16, BGR16, RGB15, BGR15, UYVP, A420, RGB8P, YUV9, YVU9, IYU1, ARGB64, AYUV64, r210, I420_10BE, I420_10LE, I422_10BE, I422_10LE, Y444_10BE, Y444_10LE, GBR, GBR_10BE, GBR_10LE, NV16, NV24, NV12_64Z32, A420_10BE, A420_10LE, A422_10BE, A422_10LE, A444_10BE, A444_10LE, NV61, P010_10BE, P010_10LE, IYU2, VYUY, GBRA, GBRA_10BE, GBRA_10LE, BGR10A2_LE, GBR_12BE, GBR_12LE, GBRA_12BE, GBRA_12LE, I420_12BE, I420_12LE, I422_12BE, I422_12LE, Y444_12BE, Y444_12LE, GRAY10_LE32, NV12_10LE32, NV16_10LE32, NV12_10LE40 }
El conducto está vivo y no necesita PREPARARSE …
INFORMACIÓN:
Working with src caps: video/x-bayer, framerate=(fraction)30/1, width=(int)1920, height=(int)1200, binning=(string)1x1, skipping=(string)1x1, format=(string)gbrg
Estableciendo el conducto a REPRODUCIENDO …
New clock: GstSystemClock
--------------------------

TIS-Stefan, 2022-06-07:
Hello

Do you use tiscamera 0.14 or 1.0? I am not clear about this.
Also, which computer model do you use?
And please let me know the pipeline.

Stefan

willingtonvr, 2022-06-07:
Hello, thanks in advance 

I've tested on 2 machines (production and develop) see the configurations of both at the end.

the tiscamera 0.14 seems more reliable the tcam-capture app has more chances to work. sometimes get a withe screen with the message "In trigger mode waiting for image", but keep waiting forever.

All my test are on 1920x1200 BGRx. (gbrg)
when start with 30/1 framerate there are 100% chances to hang., so 20/1 works better.

when it works the FPS counter on the app only gets a 12 FPS max, with 6 fps average

runing tcam-ctrl to read properties some times "unlock the drivers"

tiscamera 1.0 hangs a lot, i mean for every 10 attempts (power off/on cycle) to connect via tcam-caputure 2 goes well
for this it seems that putting full GST_DEBUG=5 seems to "unlock" the problem. also here is a interesting extract of the full debug log: 

0:00:04.285385753  0x55cf716d5a40 DEBUG              GST_PADS gstpad.c:4076:gst_pad_query:<tcambin-src_caps:sink> sent query 0x7fb3040046d0 (allocation), result 1
0:00:04.285395360  0x55cf716d5a40 DEBUG              GST_PADS gstpad.c:4076:gst_pad_query:<src:proxypad1> sent query 0x7fb3040046d0 (allocation), result 1
0:00:04.285416319  0x55cf716d5a40 WARN              structure gststructure.c:1861:priv_gst_structure_append_to_gstring: No value transform to serialize field 'allocator' of type 'GArray'
0:00:04.285424674  0x55cf716d5a40 WARN              structure gststructure.c:1861:priv_gst_structure_append_to_gstring: No value transform to serialize field 'pool' of type 'GArray'
0:00:04.285433410  0x55cf716d5a40 DEBUG               basesrc gstbasesrc.c:3260:gst_base_src_prepare_allocation:<tcammainsrc0> ALLOCATION (1) params: allocation query: 0x7fb3040046d0, GstQueryAllocation, caps=(GstCaps)"video/x-bayer\,\ format\=\(string\)gbrg\,\ width\=\(int\)1920\,\ height\=\(int\)1200\,\ framerate\=\(fraction\)20/1\,\ pixel-aspect-ratio\=\(fraction\)1/1", need-pool=(boolean)true, allocator=(GArray)NULL, pool=(GArray)NULL;
0:00:04.285447205  0x55cf716d5a40 DEBUG               basesrc gstbasesrc.c:2519:gst_base_src_get_range:<tcammainsrc0> calling create offset 18446744073709551615 length 4096, time 0
0:00:04.477432701  0x7fb304011000 DEBUG          tcam-libtcam AravisDeviceStream.cpp:530:aravis_new_buffer_callback: arvBufferStatus: Timeout has been reached before all packets were received
0:00:04.527264608  0x7fb304011000 DEBUG          tcam-libtcam AravisDeviceStream.cpp:530:aravis_new_buffer_callback: arvBufferStatus: Timeout has been reached before all packets were received
0:00:04.577151455  0x7fb304011000 DEBUG          tcam-libtcam AravisDeviceStream.cpp:530:aravis_new_buffer_callback: arvBufferStatus: Timeout has been reached before all packets were received

My machines and configs:

**machine production**
HP Slim Desktop S01-pF1xxx/8768, BIOS F.12 06/08/2020

Ubuntu 20.04.4 LTS
kernel 5.13.0-44-generic
Versions:
	Tcam:	0.14.0_master/e1fa7b21_rev_3056
	Aravis:	0.7_version_e1fa7b21
	
	
gst-launch-1.0 version 1.16.2
GStreamer 1.16.2
Processor : Intel® Core™ i5-10400 CPU @ 2.90GHz × 12 
RAM: 8GB
Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 15)

relevant network params: (same for production and develop)

net.core.rmem_default = 33554432
net.core.rmem_max = 33554432
net.core.rps_sock_flow_entries = 0
net.core.somaxconn = 4096
net.core.tstamp_allow_data = 1
net.core.warnings = 0
net.core.wmem_default = 33554432
net.core.wmem_max = 33554432

**Machine develop - laptop**
HP Laptop 15-gw0xxx/87D1, BIOS F.25 08/18/2021

5.11.0-43-generic
Ubuntu 20.04.4 LTS

Processor AMD® Ryzen 7 3700u with radeon vega mobile gfx
RAM 24 GB
Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 15)

Versions: 
	Tcam:	1.0.0_master/e24e0aef_rev_4006
	Aravis:	0.8_version_cc14855
	Modules:	gst_aravis_arvu3v_v4l2_libusb_tools_doc

**Camera**	
	
Model: DFK 33GX249 Serial: 45514186 Type: aravis

tcam-gigetool info 45514186
Model:          DFK 33GX249
Serial:         45514186
Firmware:       Rev 1063, FPGA Rev 21/16

TIS-Stefan, 2022-06-08:
Hello
"aravis_new_buffer_callback: arvBufferStatus: Timeout has been reached before all packets were received" means, that you did not receive all packages of a frame from the camera. This should not be a problem of 1920x1200@20fps, because this is really slow. Sounds a little bit for a computer problem 

Please have a look at
https://www.theimagingsource.com/documentation/tiscamera/optimizations.html
https://www.theimagingsource.com/documentation/tiscamera/troubleshooting.html

https://www.theimagingsource.com/documentation/tiscamera/environment.html at topic "TCAM_ARV_STREAM_OPTIONS"

In my experience on different hardware platforms tiscamera 1.0 is much more stable than 0.14, in particular with stream restarting. 

Stefan

midjji, 2022-06-08:
Dont buy it, its a tcam problem, not a computer one and the proof is
simple. There are two possibilities. Either the problem is in the camera
firmware, or the problem is in the tcam, specifically the driver, as the
only thing on the host computer which can affect the next time you run a
problem is a locked file handle(kernel mem resource).

Just to clarify,
The problem persists, it happens on multiple devices, and it is not helped
by using a more powerful one with lower load.

The error is a something related to the low level driver, a likely a file
handle( usb kernel resource handle?)  which isnt properly closed, almost
certainly due to bugs in the tcam, but could be an edgecase bug in
gstreamer. Another one of my issues which points out that the tcam invokes
undefined behaviour( explicitly calling a base class destructor on a
derived class for string) which is an outright compile error on clang, but
is just removed by gcc, was never solved, just closed as we dont support
clang. Note that this is a pure error and cannot ever,  under any compiler,
have done what they thought.  There is no coherent intrepetation of it in
any C language or compiler, except as ub.  Unfortunately that isnt the
problem causing it, but it sure does highlight how badly tcam is coded/
missmanaging resources, and thats what not working and what causes this
kind of problem. Nothing except busy file handles behaves this way. If you
need this to work reliably odds are good it will be faster to just reverse
engineer the protocol using the functional windows implementation and
discard the the shitty tcam stuff.  That could also provide critical
features like persistent trigger/image id, missing from the linux toolkit.
Note that i have been pointing out this problem across multiple devices and
systems for years, they just dont give a enough of a shit about the linux
version.

willingtonvr, 2022-06-08:
Well, i have to investigate, since i've worked with tiscameras and linux since 2009 (firewire times) and libunicap source code
and never had critical problems.

I've ported my source to gstreamer/tcam/aravis sice 0.12.0 version and works great on GiE cameras, probably is not the case with the USB cameras.

I never worked with USB devices, really don't like the way USB works and its limitations: cable length and crappy connectors.
lookig to your issues makes me leave the temptation to try the USB subsystem and stick to the gie interface. Windows drivers/software wasn't an option in 2009 in 2022 won't be either.

Willington

midjji, 2022-06-08:
Yeah usb is a mess, especially so for video, but there are few good
alternatives sometimes.

TIS-Stefan, 2022-06-09:
Hmmm, I am lost. Thus lets try a different approach in order to rule out hardware.

Please install qv4l2. It is a standard v4l2 program. No Gstreamer, no software from us. 

Run the program, open the camera, show live video.

Expected Result: All works perfectly. 

Please let me know your result.

willingtonvr, 2022-06-10:
Hello,

I changed the Ethernet adapter it works as expected.

my problem with the driver locks was due to the RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 15), 

just put an intel pro 1000 device and voila! got 1920x1200 @30 fps

thank you

TIS-Stefan, 2022-06-13:
Thank you for letting me know. And sorry about my nonsense about qv4l2. I mixed up support cases.

Stefan

closed 2022-06-13 by TIS-Stefan