TheImagingSource/tiscamera #456: DZK 33UX250 - Ubuntu 18 - No avilable caps

bruane-path, 2022-03-08:
Hello,

I'm trying to get a DZK 33UX250 - USB 3.0 Polarsens camera working on my system (Ubuntu 18.04 [64-bit]), but I am running into a problem.  If I try to use the camera with the example C code, Tcam-capture, or IC Measure software, I can't succesfully capture any images or see any video output.  It looks like the issue is that there are no supported video formats found for the device.  IC Measure, for example, logs the following error to the the console: 
```
0:02:04.841290379 23088 0x7f9844004d60 ERROR            tcammainsrc gsttcammainsrc.cpp:727:gst_tcam_mainsrc_get_all_camera_caps: Device did not provide ANY valid caps. Refusing playback.
[2022-03-08 10:47:31.002] [IC Measure] [error] Failed to open device. what(): Failed to open device: No supported videoformats.
```

A couple of notes that will  hopefully help in diagnosing the issue: I also have a DFK 33UX273 camera and have no problems using it with the example C code, Tcam-capture, or IC Measure on the same system.  I also tried using the DZK 33UX250 on a Windows 8 machine, and it _DOES_ work on that computer.  I installed IC Measure / device driver to test it out, and was able to see the live video feed.

If I use tcam-ctrl on the Ubuntu 18.04 computer, I get the following output:

```
$ tcam-ctrl -l
Model: DZK 33UX250 Serial: 37120237 Type: v4l2
```
```
$ tcam-ctrl -c 37120237
Available gstreamer-1.0 caps:
EMPTY
```

Any ideas on how to get the DZK 33UX250 working on my Ubuntu 18.04 based computer?  Let me know if you need any other information to help diagnose the problem.
Thanks!

TIS-Stefan, 2022-03-09:
Hello!

The polarization cameras are not supported by our tiscamera SDK. But you can use Aravis in order to access them directly.

[polarcamera.zip](https://github.com/TheImagingSource/tiscamera/files/8213125/polarcamera.zip)

This C++ sample separates the four polarization images from the sensor. 
It uses Aravis and OpenCV for displaying. 

I hope, this helps.

Stefan

bruane-path, 2022-03-10:
Thanks for the example files, Stefan!  I had some issues with the makefile on my system (some undefined references).  After getting through some of these by linking some more libraries explicitly, it was just leading to more and more undefined refs.  I ended up opting for creating a `CmakeLists.txt` file based on the makefile to help w/ the dependencies, which ended up working out.  Just for ref:  this is what the file I got to make the code compile:
```
project(polarcamera)

cmake_minimum_required(VERSION 2.6)
set(CMAKE_CXX_STANDARD 11)

find_package(PkgConfig REQUIRED)
find_package(OpenCV 4.1.0 REQUIRED)
find_package(Qt5 COMPONENTS Core Gui Widgets Test Concurrent)
find_package(Eigen3 COMPONENTS Eigen)
pkg_check_modules(deps REQUIRED IMPORTED_TARGET glib-2.0)
pkg_check_modules(arv REQUIRED IMPORTED_TARGET aravis-0.6)

add_executable(arv-polarcamera main.cpp arvhelper.cpp )

target_link_libraries(arv-polarcamera
    PkgConfig::deps
    PkgConfig::arv
    ${OpenCV_LIBS}
)
```
When I run the arv-polarcamera program, it does detect the camera and the pixel format options (which I can select through the console options).  However, once I select the cam/format, it pops a GUI window up, but I never see the live feed (it's just a black rectangle).  I put some print statements in different parts of the code, and it appears the callback (AcquisitionCallback) is never being invoked.  I do not see any errors in the console output.

Any ideas on what the issue would be?  Thanks!

TIS-Stefan, 2022-03-11:
Hello

First of all, I am sorry, that I did not made a build instruction for the sample and pointed to Aravis Library and OpenCV being needed. 

I ran the program on my desktop computer with Ubuntu 20.04. After start I get

```
Aravis OpenCV
Enumerating GigE cameras
	1 :  TIS-32919960
Select : 1
Camera : DZK 33UX250
Available pixel formats:
	1: Polarized Mono 8   0x8108000a
	2: Polarized Mono 12 Packed   0x810c000b
	3: Polarized Mono 16   0x8110000c
Select : 1
```

After selecting the pixel format, 5 OpenCV windows open. A big one showing the whole image and 4 small ones, that show the polarized images.

Unfortunately I must guess now, why you do not get images. For example, the trigger mode of the camera is enabled. Therefore, you may add the line

`arv_camera_set_trigger(Camera,"Off");`

Maybe the frame rate is too high. You would get the "Image does not fit to buffer size in that case" error message. You may try
`arv_camera_set_frame_rate(Camera, 15.0);`
instead of
`arv_camera_set_frame_rate(Camera, 30.0);`

You can check with lsub -t, whether the camera is connected to an USB 3 port:

```
$ lsusb -t
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/10p, 5000M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 5000M
        |__ Port 1: Dev 4, If 0, Class=Miscellaneous Device, Driver=, 5000M
        |__ Port 1: Dev 4, If 1, Class=Miscellaneous Device, Driver=, 5000M
        |__ Port 1: Dev 4, If 2, Class=Miscellaneous Device, Driver=, 5000M
        |__ Port 1: Dev 4, If 3, Class=Video, Driver=uvcvideo, 5000M
        |__ Port 1: Dev 4, If 4, Class=Video, Driver=uvcvideo, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/16p, 480M
```
Look for "uvcvideo". 5000M indicates USB 3, 480M USB 2.0. The camera does not run on USB 2.0 ports.

A sample like the one above is the only way getting the camera running in Linux. An alternative is changing the firmware in the camera to DMK 33UX250. In your case you wont loose anything, because the special video formats are used in Windows for visualizing AoLP and DoLP modes. With the DMK 33UX250 firmware you can use v4l2, aravis, tiscamera, Gstreamer and all these little packages. 

If you want to change the firmware I would like you to make a contact at https://www.theimagingsource.com/company/contact/.
I am available until 6pm UTC+1. (Germany)

Stefan

bruane-path, 2022-03-11:
Hi Stefan,

Thanks for the information!  I confirmed the camera is on a USB 3 port.  I set trigger mode to "Off" and dropped the frame rate to 15fps (also tried 5fps), but still getting the same result: no video/images & no console prints from the callback.

Your suggestion about switching the firmware sounds promising.  We have already done development work with the DFK 33UX273 (using GStreamer and the tiscamera SDK), so if we could reuse that work with the DZK 33UX250, that would be great. 

Just to confirm, if we switched the firmware to DMK 33UX250 on the DZK, we would not lose any of the camera's polarization data, correct?  Is the special Windows visualization format you mentioned based on calculations using the raw data that we would have available to us in Linux?

Thanks!
Bill

TIS-Stefan, 2022-03-14:
Bill

> Just to confirm, if we switched the firmware to DMK 33UX250 on the DZK, we would not lose any of the camera's polarization data, correct? Is the special Windows visualization format you mentioned based on calculations using the raw data that we would have available to us in Linux?

Correct. The Windows visualization is based on the raw data of the camera. There is no special processing in the camera.

Please make a contact at  https://www.theimagingsource.com/company/contact/.

Stefan

bruane-path, 2022-03-14:
Thanks, Stefan.  I just filled out the contact form you linked.

TIS-Edgar, 2024-04-12:
Closed due to inactivity.

closed 2024-04-12 by TIS-Edgar