TheImagingSource/tiscamera #457: camera settings via TIS.py

vinit13792, 2022-03-10:
Hi,

I would like to know if i can set the following properties of camera via TIS class

`1 : Model: DFK 37BUX178 Serial: 10910616 `
`Select : 1`
`1: rggb`
`2: rggb16`
`3: {RGBx,xRGB,BGRx,xBGR,RGBA,ARGB,BGRA,ABGR}`
`4: {RGBx,xRGB,BGRx,xBGR,RGBA,ARGB,BGRA,ABGR}`
`5: {RGBx,xRGB,BGRx,xBGR,RGBA,ARGB,BGRA,ABGR}`
`6: {RGBx,xRGB,BGRx,xBGR,RGBA,ARGB,BGRA,ABGR}`
`Select : 2`
`1:  640x480`
`2:  1920x1080`
`3:  2048x2048`
`4:  3072x2048`
`Select : 4`
`1:  30/1`
`2:  15/1`
`3:  5/1`
`4:  1/1`
`Select : 1`

Vinit

TIS-Stefan, 2022-03-10:
The TIS class, as I implemented it, determines the output format only. Please see the constructor documentation at https://github.com/TheImagingSource/Linux-tiscamera-Programming-Samples/tree/master/python-common
I suggest to avoid 16 bit color formats, in case you do not have a very high dynamic. Without tiscamera-dutils, there is no conversion from rbbg16 to RGBx64. The tiscamera-dutils can be downloaded from https://www.theimagingsource.com/support/downloads-for-linux/
(But RGB64 is not added to the TIS.py.)

It would be for 8bit per color channel:

```
Tis = TIS.TIS()
Tis = Tis.openDevice("10910616", 3072, 2048, "30/1",TIS.SinkFormats.BGRA, True)
```
The last parameter "True" opens a video window (ximagesink) for showing the live display. If False is passed, no live video is shown. 

Stefan

vinit13792, 2022-03-10:
Hi Stefan,

Just to be clear, whats the difference between rggb, rggb16 and RGBx? And if i want a RGB image from the videostream which format should i go ahead with?

TIS-Stefan, 2022-03-10:
Hello
rggb is Bayer Raw data 8 bit. Looks grayscale with faint checkboard pattern.
rggb16 is the same as rggb, but 16 bit per pixel.
RGBx / BGRx is same as RGB32 color image. The Bayer Raw data has been debayered.

For an explanation of Bayer Raw, please have a look at https://en.wikipedia.org/wiki/Bayer_filter

Stefan

vinit13792, 2022-03-10:
> 

Hi Stefan,

Is it mandatory to close the device after opening it? Although i do not see the function for closing the device using TIS class

TIS-Stefan, 2022-03-10:
Hello Vinit

the Stop_Pipeline() call closed the device.

"Tis.Set_Property()" is used in order to set a property. For example:

```
Tis.Set_Property("Whitebalance Auto", False)
Tis.Set_Property("Whitebalance Red", 64)
Tis.Set_Property("Whitebalance Green", 50)
Tis.Set_Property("Whitebalance Blue", 64)

# Disable gain and exposure automatic
Tis.Set_Property("Gain Auto", False)
Tis.Set_Property("Gain",0)
Tis.Set_Property("Exposure Auto", False)
Tis.Set_Property("Exposure", 24000)
```
I recommend to set the properties, after you called Tis.Start_pipeline() . (This is changed in next tiscamera version 1.0, which is available as beta now)

Stefan

vinit13792, 2022-03-23:
Hi Stefan, 

Thank you for this. 

I would also like to know if there are any settings where I can improve the sharpness of the image. It seems the images are not sharp enough compared to the mobile camera. 

I do understand that the mobile camera and these industrial cameras are leagues apart, but I just want a way to sharpen the image captured from my TIS camera to be sharper. Any pointers?

Regards,
Vinit.

TIS-Stefan, 2022-03-23:
Hello Vinit

we do not have a sharpness property. You may use OpenCV and create an own one.

In case the image looks a little bit "outwashed" or "milky", then the lens does not fit to the sensor. 

Compare to a mobile camera is futile, because:
- mobil sensor has higher resolution
- lens was created for the mobile sensor in particular
- there is massive image processing on mobil device. You do not see the original image from the sensor. 

Stefan

vinit13792, 2022-03-23:
Hi Stefan,

What did you mean by "then the lens does not fit to the sensor."?

Vinit

TIS-Stefan, 2022-03-23:
Hi Vinit

> What did you mean by "then the lens does not fit to the sensor."?

If the lens is made for bigger pixels on the sensor, then you get some issues with outworn images. 

Stefan

TIS-Edgar, 2024-04-12:
Closed due to inactivity.

closed 2024-04-12 by TIS-Edgar