TheImagingSource/tiscamera #459: unstable frame rate

Hypothesis-Z, 2022-03-26:
HI! 

I'm using DFK 33GP1300 on Ubuntu 18.04LTS. I found some timeout when running C++ program like [Linux-tiscamera-Programming-Samples](https://github.com/TheImagingSource/Linux-tiscamera-Programming-Samples/tree/master/Save%20an%20image%20using%20OpenCV%20in%20C%2B%2B). Most attempts on snapping an image return in 50ms, but sometime the duration time can reach up to 250ms. 

Here is my code:
``` C++
Camera::Camera(std::string address){
    gst_init(NULL, NULL);
    cam_ptr_ = std::make_shared<TcamImage>(address);
}

bool Camera::Open(){
    // Set a color video format, resolution and frame rate
    cam_ptr_->set_capture_format("BGRx", gsttcam::FrameSize{1280,960}, gsttcam::FrameRate{30,1});

    // Comment following line, if no live video display is wanted.
    // cam_ptr_->enable_video_display(gst_element_factory_make("ximagesink", NULL));
    
    // Start the camera
	if (!cam_ptr_->start()){
        sleep(1);
        LOG(ERROR) << "Openning camera failed!";
        return false;
    }
    LOG(INFO) << "Camera opened.";
    return true;
}

bool Camera::Release(){
    return cam_ptr_->stop();
}

bool Camera::Snap(cv::Mat& image){
    if (cam_ptr_->snapImage(60))
    {
        image.create( cam_ptr_->getHeight(), cam_ptr_->getWidth(), CV_8UC(cam_ptr_->getBytesPerPixel()));
        memcpy( image.data, cam_ptr_->getImageData(), cam_ptr_->getImageDataSize());
    }
    else
    {   // Timeout
        LOG(WARNING) << "Timeout on snapping a image.";
        return false;
    }
    cv::cvtColor(image, image, cv::COLOR_BGRA2BGR);
    // cv::imshow("show", image);
    // cv::waitKey(0);
	return true;
}
```

Here is std output
``` log
I20220326 18:00:55.983924 26339 m_cam_node.cpp:38] duration time: 30ms
I20220326 18:00:56.017194 26339 m_cam_node.cpp:38] duration time: 32ms
I20220326 18:00:56.049361 26339 m_cam_node.cpp:38] duration time: 31ms
I20220326 18:00:56.083317 26339 m_cam_node.cpp:38] duration time: 32ms
I20220326 18:00:56.166404 26339 m_cam_node.cpp:38] duration time: 82ms
W20220326 18:00:56.232672 26339 camera.cpp:41] Timeout on snapping a image.
I20220326 18:00:56.255024 26339 m_cam_node.cpp:38] duration time: 21ms
I20220326 18:00:56.288712 26339 m_cam_node.cpp:38] duration time: 32ms
I20220326 18:00:56.324084 26339 m_cam_node.cpp:38] duration time: 33ms
I20220326 18:00:56.352793 26339 m_cam_node.cpp:38] duration time: 27ms
I20220326 18:00:56.387781 26339 m_cam_node.cpp:38] duration time: 33ms
I20220326 18:00:56.452998 26339 m_cam_node.cpp:38] duration time: 64ms
I20220326 18:00:56.525981 26339 m_cam_node.cpp:38] duration time: 71ms
I20220326 18:00:56.560192 26339 m_cam_node.cpp:38] duration time: 32ms
I20220326 18:00:56.587836 26339 m_cam_node.cpp:38] duration time: 26ms
I20220326 18:00:56.622365 26339 m_cam_node.cpp:38] duration time: 33ms
I20220326 18:00:56.655375 26339 m_cam_node.cpp:38] duration time: 31ms
I20220326 18:00:56.690043 26339 m_cam_node.cpp:38] duration time: 33ms
[... 211 lines cut ...]
I20220326 18:01:04.611413 26339 m_cam_node.cpp:38] duration time: 33ms
I20220326 18:01:04.642424 26339 m_cam_node.cpp:38] duration time: 29ms
I20220326 18:01:04.673943 26339 m_cam_node.cpp:38] duration time: 30ms
I20220326 18:01:04.707561 26339 m_cam_node.cpp:38] duration time: 32ms
I20220326 18:01:04.741683 26339 m_cam_node.cpp:38] duration time: 33ms
I20220326 18:01:04.775983 26339 m_cam_node.cpp:38] duration time: 33ms
I20220326 18:01:04.808300 26339 m_cam_node.cpp:38] duration time: 30ms
I20220326 18:01:04.845266 26339 m_cam_node.cpp:38] duration time: 35ms
I20220326 18:01:04.875844 26339 m_cam_node.cpp:38] duration time: 29ms
I20220326 18:01:04.915138 26339 m_cam_node.cpp:38] duration time: 38ms
I20220326 18:01:04.943303 26339 m_cam_node.cpp:38] duration time: 26ms
W20220326 18:01:05.004616 26339 camera.cpp:41] Timeout on snapping a image.
W20220326 18:01:05.065773 26339 camera.cpp:41] Timeout on snapping a image.
W20220326 18:01:05.126075 26339 camera.cpp:41] Timeout on snapping a image.
I20220326 18:01:05.179083 26339 m_cam_node.cpp:38] duration time: 52ms
I20220326 18:01:05.211239 26339 m_cam_node.cpp:38] duration time: 31ms
I20220326 18:01:05.244377 26339 m_cam_node.cpp:38] duration time: 32ms
I20220326 18:01:05.277804 26339 m_cam_node.cpp:38] duration time: 32ms
I20220326 18:01:05.313324 26339 m_cam_node.cpp:38] duration time: 34ms
I20220326 18:01:05.346658 26339 m_cam_node.cpp:38] duration time: 31ms
I20220326 18:01:05.377821 26339 m_cam_node.cpp:38] duration time: 29ms
I20220326 18:01:05.411360 26339 m_cam_node.cpp:38] duration time: 32ms
I20220326 18:01:05.444922 26339 m_cam_node.cpp:38] duration time: 32ms
I20220326 18:01:05.478562 26339 m_cam_node.cpp:38] duration time: 32ms
```

TIS-Stefan, 2022-03-28:
Hello

Setting the TCAM_LOG environment variable could be helpful here. (https://www.theimagingsource.com/documentation/tiscamera/environment.html)

Also the TCAM_ARV_PACKET_REQUEST_RATIO could be helpful

`export TCAM_ARV_PACKET_REQUEST_RATIO=1.0
`
Which computer model do you use? 

Stefan

TIS-Edgar, 2024-04-12:
Closed due to inactivity.

closed 2024-04-12 by TIS-Edgar