TheImagingSource/tiscamera #462: Camera returning black screen

singlakdeepak, 2022-04-26:
Hi, 
I installed the tiscamera library on my Linux PC with Ubuntu 20.04 and using the basic command 
`gst-launch-1.0 tcambin ! videoconvert ! ximagesink` to run the camera. However, my camera is returning a **black screen** with the following message in the background
`gst-launch-1.0 tcambin ! videoconvert ! ximagesink
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
INFO:
Working with src caps: video/x-raw, framerate=(fraction)539/1, width=(int)720, height=(int)540, format=(string)GRAY8
Setting pipeline to PLAYING ...
New clock: GstSystemClock
ERROR: from element /GstPipeline:pipeline0/GstXImageSink:ximagesink0: Output window was closed
Additional debug info:
ximagesink.c(696): gst_x_image_sink_handle_xevents (): /GstPipeline:pipeline0/GstXImageSink:ximagesink0
Execution ended after 0:00:05.060191483
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...`
The camera mode is DMK 37AUX287. I also installed IC Measure from https://www.theimagingsource.com/support/downloads-for-linux/ and my camera is displaying frames in this application. 
I also tried to use the camera on Windows PC with its windows software and it works fine there with any frame rate.

It would be great if someone can let me know some workarounds for the problem since I would like to use the camera from Python for my project. Thank you!

--
Deepak

TIS-Stefan, 2022-04-27:
I would start with lower frame rates in the GStreamer pipeline. 

`gst-launch-1.0 tcambin ! video/x-raw, format=BGRx, width=720, height=540, framerate=30/1 ! videoconvert ! ximagesink`

If that works, you may use "300/1" instead of "30/1". You can try up to "539/1" until your computer can not handle the high frame rate anymore.

Stefan

singlakdeepak, 2022-04-27:
Hi Stefan,

Thank you for your reply. I tried the following command on my PC. So the following happens: As soon as I plugged in the camera, and ran the command, it displayed me some frames when I moved the camera, however within 10 seconds the frame update stopped and it kept on displaying the last frame captured. After canceling this and running the command again, it started displaying the black screen again. 
To be fair, it seems IC-Measure is a bit buggy too on my PC and it keeps on crashing sometimes when I try to open my device there. Any help would be greatly appreciated. Thanks!

Deepak

TIS-Stefan, 2022-04-28:
Hello

Is the USB connector plugged well into the camera?
Which computer model and brand do you use? In particular, which USB controller chip do you use? If it is AMD/ASMEDIA, then you must expected issues.

Stefan

singlakdeepak, 2022-04-28:
Hi Stefan, 
Yes, the USB connector is well plugged in. Here are some of the details of the processors in PC:
vendor_id	: AuthenticAMD
cpu family	: 25
model		: 33
model name	: AMD Ryzen 9 5900X 12-Core Processor x24
It also has a graphics card by NVIDIA. 
I bought the PC from iBuyPower. 
To check the USB controller chip, I used `sudo lspci|grep -i usb` and it returned me this:
02:00.0 USB controller: Advanced Micro Devices, Inc. [AMD] Device 43ee
2d:00.3 USB controller: Advanced Micro Devices, Inc. [AMD] Matisse USB 3.0 Host Controller

It seems from here, it is AMD. 
By the way, it seems I was connecting the camera on the front 3.0 ports earlier and it showed black screen but on the port at the back, it is showing live stream well. However, I am not able to understand why it didn't work on the front port since other devices work well on the front ports. 
Thanks for your help. 
--
Deepak

TIS-Edgar, 2024-04-12:
Closed due to inactivity.

closed 2024-04-12 by TIS-Edgar