TheImagingSource/tiscamera #464: For camera(DFK33UP1300) on Jetson Nano

FantasticCHH, 2022-05-13:
Würden Sie mir bitte mitteilen, wie ich das Jetson Nano IDK wie das PC IDK lassen kann?
Ich habe einige Einstellungen für mein Experiment
Ich hatte mein PC-Experiment beendet, aber jetzt muss ich auf die Jetson Nano-Plattform umsteigen.
Unten sind meine schematischen Diagramme, die die PC-Plattform und die Jetson-Nano-Plattform darstellen. Mein erwarteter Effekt ist, dass die Jetson Nano-Plattform mit den gleichen Einstellungen wie die PC-Plattform angepasst werden kann. Sie wissen nicht, ob es ein Paket oder Update gibt, das diese Funktion unterstützt?
Meine Jetpack-Version ist 4.3. Ich habe versucht, das Paket der Exe-Datei über Wine zu installieren. Offensichtlich gibt es einige Probleme bei der Installation. Ich frage mich, ob es andere Lösungen gibt?

would you please let me know how to let the Jetson Nano IDK as like the  PC IDK?
I have some settings to setting for my experiment
I had finished my PC experiment,but now I have to transplant to  Jetson Nano platform.
Below are my schematic diagrams, representing the personal computer platform and the jetson nano platform. My expected effect is that the jetson nano platform can be adjusted with the same settings as the PC platform. Don't know if there is a package or update that supports this feature?
My jetpack version is 4.3. I have tried to install the package of the exe file through wine. Obviously, there are some problems in the installation. I wonder if there are other solutions?
[image]
[image]
?

TIS-Stefan, 2022-05-13:
Hallo

Es tut mir leider, aber die Kameraeigenschaften von Windows und Linux unterscheiden sich sehr, da unterschiedliche Treiber verwendet werden. 
Es bleibt Ihnen nur die Möglichkeit, die Kameraeigenschaften so einzustellen, daß Sie ein ähnliches Bild wie in Windows erhalten.  Bei weißabgleich wäre das
Linux 64 = 1.0 in Windows.
Daraus folgt:

Rot 64 * 1,59 = 101
Grün 64 *1,08 = 69
Blau 64 *1,596 = 101

Ich hoffe, daß das weiterhilft.
------------
Hello

I am sorry, but the camera properties of Windows and Linux are very different, because different drivers are used. 
Your only option is to adjust the camera properties to get a similar image as in Windows.  For white balance this would be
Linux 64 = 1.0 in Windows.
From this follows:

Red 64 * 1.59 = 101
Green 64 *1.08 = 69
Blue 64 *1,59 = 101

I hope that this helps.

Stefan

FantasticCHH, 2022-05-13:
Tcam Capture - Resolution 

[image]

excuse me, for this setting about the resolution on Jetson Nano. 
I get the setting on my PC platform. I used the choice about RGB 1024*768
But on Jetson Nano do not have this choice , and also the format do not have RGB,do I have another plan to cover this question?
Thanks you for the your answer

TIS-Stefan, 2022-05-13:
Using v4l2 for USB cameras has a limited list of video formats only. You may access the camera by USB3Vision using Aravis (use tiscamera 1.0 (git checkout development) for this) and build with Aravis support. Then you can set many formats.

FantasticCHH, 2022-05-13:
sorry for bothering,about the aravis: I didn't understand some news what you teach me. :(
can you teach me how can I do for that?
I have no idea about this .
I just try to install 
TCAM_ARAVIS_USB_VISION 
DBUILD_LIBUSB
and 
DBUILD_V4L2
But the question do not been fix. poorly...

TIS-Stefan, 2022-05-13:
Some background information:
v4l2 reads the available video formats and their frame rates from the USB descriptor in the camera. The USB descriptor in the camera has limited space, therefore, we can store only some video formats and frame rate in there. The camera itself has millions of formats and frame rate combinations, but you simple can not store them in the USB descriptor of the camera.
Here we use USB3Vision instead of v4l2. With USB3Vision we can specify the width and height and the frame rate we want to use. The camera tells us, whether this is a valid format. For USB3Vision, we use the open source Aravis library.

Did you build tiscamera on your own or did you install the package?

Package: Aravis / USB3Vision support is built in.

Build on your own:
Run cmake as shown in the "Readme" at https://github.com/TheImagingSource/tiscamera:
```
git clone https://github.com/TheImagingSource/tiscamera.git
cd tiscamera
# only works on Debian based systems like Ubuntu
sudo ./scripts/install-dependencies.sh --compilation --runtime
mkdir build
cd build

# With ARAVIS:
cmake -DBUILD_ARAVIS=ON ..

make
sudo make install
```

After you did so, tcam-capture will show your camera twice with the backend "v4l2" and "aravis" shown after the serial number.

Give that a try.

Stefan

FantasticCHH, 2022-05-14:
Thanks for your teaching.
I were try to install as like your teaching steps.
It's looks like as you say.
[image]
But at the format and resulation,it did not have the choice as like PC platform
[image]
[image]
[image]
How can I create it ,or update this function?

TIS-Stefan, 2022-05-16:
The serial number with extension "-aravis" is not offered?

The do following:

```
cd tiscamera
cd build
rm * -rf
# With ARAVIS:
cmake -DBUILD_ARAVIS=ON ..

make
sudo make install
```
The "rm * -rf" is new. Sometimes cmake does not overwrite as wanted...

Stefan

FantasticCHH, 2022-05-17:
I were tried it again,but the UI interface do not be exchange.as like I posted the img
format and resulation still not have the choice which I want to choose(format:RGB,Resulation:1024*768).
[image]

TIS-Stefan, 2022-05-17:
Hello
The camera is listed only once? 
Please show me the output of 
tcam-ctrl -l

Stefan

FantasticCHH, 2022-05-17:
this is the img of "tcam-ctrl -l"
but looks not install aravis well
I'm trying another time.but at the step of "make" ,looks not successfully .
I don't know why.poorly
[image]

TIS-Stefan, 2022-05-17:
WTF??
The instructions I gave work on all other computers. Did you also add the ".." to the cmake command?

Stefan

FantasticCHH, 2022-05-17:
yes~ I also add ".." but it show m
[image]
e

TIS-Stefan, 2022-05-17:
ok, there is an error, so it can not work

please
sudo rm * -rf
in the build
run cmake ...... again and show the output.

Stefan

FantasticCHH, 2022-05-17:
I did it again and again, but it still error again and again.
Do you know how can I fix it ?or download it again?

TIS-Stefan, 2022-05-17:
You have a "permission denied" error. So it is something with user rights on your computer.
You may try 
sudo cmake .....
Which usually is not a good idea.

TIS-Stefan, 2022-05-17:
Arrrgh! I forgot one point.
In order to use usb3vision, tiscamera must be built with that.  So the cmake command expands to
```
cd tiscamera
cd build
rm * -rf
# With ARAVIS:
cmake -DBUILD_ARAVIS=ON TCAM_ARAVIS_USB_VISION=ON ..

make
sudo make install
```

Please apologize, that I missed that one and wasted your time.

Stefan

FantasticCHH, 2022-05-17:
I were followed the tips ,but it still have a big trouble.
I don't know why.poorly
Do you use the same jetson system machine to install?
I have no idea now....
[image]
[image]
[image]
[image]

TIS-Stefan, 2022-05-17:
Run "sudo make". Then "sudo make install

TIS-Stefan, 2022-05-17:
Jetson nano is not the problem. Ubuntu 18.04 is not the problem. 
Please change into
tiscamera/build/tools/firmwareupdate
run
sudo ./firmwareupdate -i -d nnnnnnnn

Show me the output. "nnnnnnnn" is to be replaced by your camera's serial number. You camera is from 2017, so we may have an old firmware.

FantasticCHH, 2022-05-17:
this photo is from tiscamera/build/tools/firmware-update

[image]

TIS-Stefan, 2022-05-17:
Version 781 is recent. It implements USB3Vision protocol. 

ok. last try

```
cd tiscamera
cd build
rm * -rf
cd ..
git checkout development
cd build
cmake -DTCAM_ARAVIS_USB_VISION=ON -DTCAM_BUILD_ARAVIS=ON -DTCAM_DOWNLOAD_MESON=ON ..
make
sudo make install
```
It uses a different Aravis version. And it is the tiscamera 1.0 release candidate

FantasticCHH, 2022-05-17:
it is looks like some proble
[image]
[image]
m

TIS-Stefan, 2022-05-17:
"Permission denied" always points to user right issues on your system. 
maybe sudo rm *-rf in build directory helps.

FantasticCHH, 2022-05-17:
I already do that operate but it looks not fix the problem. :(
[image]
[image]

TIS-Stefan, 2022-05-17:
It seems, your Linux knowledge is a little bit limited, right? 

I suggest to continue tomorrow. 

Since you seem to be from Asia, you may contact my colleagues in Taipeh https://www.theimagingsource.com/company/contact/
Maybe they can help directly on your computer using Anydesk or Teamviewer.

FantasticCHH, 2022-05-19:
excuse me for the same question 
before I ask your colleagues in Taipei
your colleagues had already try to install on the jetpack version 4.5
I got to try jetpack package version: 4.3 4.4
It's also do not have aravis development version 
could you please let me know how to find the aravis version?
I've been stuck on this issue for a long time.. :(

TIS-Stefan, 2022-05-19:
I have the suspect, you already asked my colleagues in Taipeh. 

However, we found, the DFG 33UP1300 does not work with Aravis, the frame rates are not listed. We do not know why that happens, all other cameras work.

Please go in contact with my Asian colleagues, as I mentioned. 

Also: tiscamera installs the correct Aravis version. You do not need to bother about this. 

Stefan

FantasticCHH, 2022-05-19:
As I said, I have already consulted with your colleagues in Taipei.
[image]

TIS-Stefan, 2022-05-19:
I am sorry for misunderstanding. 

However, they asked me too and we found the issue of the DFK 33UP1300 with Aravis.

TIS-Stefan, 2022-05-25:
Hello

you can try this:
Please download the file(s)
 - tiscamera_1.0.0.3995~development_3e7c334a_arm64_ubuntu_1804.deb
 from https://we.tl/t-2qypxKFAaK
The link is valid for 7 days.

Until now, I had time to test it on AMD64. The DFK 33UP1300 works well with v4l2 and aravis.

Install with
sudo apt install ./tiscamera_1.0.0.3995~development_3e7c334a_arm64_ubuntu_1804.deb

Then run tcam-capture.

However, release tests are pending. 

Stefan

FantasticCHH, 2022-05-26:
Thanks for your help,this gave me a lot of support.
By the way,could you help me to add the choice
Include(Format:RGB,Resolution:1024*768,FPS:120)

[image]
[image]
[image]

TIS-Stefan, 2022-05-26:
​ 

Hello

The list in the program tcam-capture is limited, we made a selection of the millions of formats you can set.
In your own program you can use a pipeline like
tcambin ! video/x-raw, format=BGRx, width=1024, height=768, framerate=120/1 ! videoconvert ! ximagesink

You can test this on the command line with
gst-launch-1.0 tcambin ! video/x-raw, format=BGRx, width=1024, height=768, framerate=120/1 ! videoconvert ! ximagesink

 [https://user-images.githubusercontent.com/63339132/170415954-b12bfd6d-accd-4a9c-979b-8e42a8509df6.jpg]
 [https://user-images.githubusercontent.com/63339132/170415959-e4af6d47-acaa-4f37-abcd-153ab18e30a1.jpg]
 [https://user-images.githubusercontent.com/63339132/170415962-3c0ccd0c-503f-440e-b632-9ae34732b74e.jpg] 

  
 Was this message helpful?   Yes   [https://theimagingsource.deskpro.com/ticket-rate/VNFT-2227-QDAF/WA3B2QAM7CDTSZN/400484?setrating=1]   It was OK   [https://theimagingsource.deskpro.com/ticket-rate/VNFT-2227-QDAF/WA3B2QAM7CDTSZN/400484?setrating=0]   No   [https://theimagingsource.deskpro.com/ticket-rate/VNFT-2227-QDAF/WA3B2QAM7CDTSZN/400484?setrating=-1] 

View and manage this ticket online: https://theimagingsource.deskpro.com/tickets/83729 

  The Imaging Source [https://www.theimagingsource.com] · https://theimagingsource.deskpro.com/ 

 (#ETWJWA3B2QAM7CDTSZN)

FantasticCHH, 2022-05-27:
I got to use"gst-launch-1.0 tcambin ! video/x-raw, format=BGRx, width=1024, height=768, framerate=120/1 ! videoconvert ! ximagesink " ,it couldn't be added.
By the way, I got to try the format choice and it can't be exchanged. resolution can be exchanged successfully,and also I let the FPS choice be exchanged sometimes it will let the Jetson Nano error , I have to restarted the machine.But it can't be troubleshooting.

[image]
.

TIS-Stefan, 2022-05-30:
Hello

you must pass the serial number and the backend to the tcambin, otherwise, it does not know, how to use the camera;

gst-launch-1.0 tcambin serial=18714599-aravis ! video/x-raw, format=BGRx, width=1024, height=768, framerate=120/1 ! videoconvert ! ximagesink

Stefan

FantasticCHH, 2022-06-01:
It couldn't be installed.
[image]

TIS-Stefan, 2022-06-01:
Please try the release version:

https://we.tl/t-yCzjdpfuOG

Download and install.

FantasticCHH, 2022-06-01:
I got already do it.And try to do it (gst-launch-1.0 tcambin ! video/x-raw, format=BGRx, width=1024, height=768, framerate=120/1 ! videoconvert ! ximagesink) again.But the problem isn't been fixed.

TIS-Stefan, 2022-06-01:
The link above is a newer version, than the one I sent previously. Therefore, please dowload & install.

FantasticCHH, 2022-06-01:
sure.I know~I got already download it,and finished install .After that ,I got already try it again.But it isn't been fixed.

TIS-Stefan, 2022-06-01:
Runs perfectly on my desktop computer.

please run tcam-ctrl --version and show the output. It must be 
```
Versions: 
	Tcam:	1.0.0_development/12cac40d_rev_3997
	Aravis:	0.8_version_12cac40d
	Modules:	gst_aravis_arvu3v_v4l2_libusb_tools_doc_tests
```
run tcam-capture and make sure, the trigger mode is disabled.

FantasticCHH, 2022-06-01:
I run the code tcam-ctrl --version
The answer is as like yours (I try it on my Jetson Nano)
And also try this(gst-launch-1.0 tcambin！video/x-raw，format=BGRx，width=1024，height=768，framerate=120/1！videoconvert！ximagesink)
It can't  be installed

TIS-Stefan, 2022-06-01:
Its a Nano issue. Working on that.

FantasticCHH, 2022-06-02:
we use the picture (1024*768) to do our research but  we use it successfully on our windows platform.
But I think maybe the picture size isn't a problem.
I want to understand how can I try to add the format  RGB32 choice ,I had read the windows platform again ,I don't know two platform gap.
We need to read the photo by RGB ,I got already research how to let the gray8 picture become RGB photo ,but it didn't be find.
[image]

By the way, I got another problem, after I install the news version ,I try to open it but it means error?
[image]

TIS-Stefan, 2022-06-02:
Hello

> we use the picture (1024768) to do our research but we use it successfully on our windows platform.

of course on Windows, because you do not run Windows on the Nano, right? The issue is caused by the combination of DFK 33UP1300 and Nano. DFK 33UP1300, Ubuntu and Desktop computer runs perfectly too!
The problem is deep into USB transfer stuff on the Nano!

> I want to understand how can I try to add the format RGB32 choice ,I had read the windows platform again ,I don't know two platform gap.

you do that by setting:
 gst-launch-1.0 tcamsrc serial=18714599 ! video/x-raw, format=BGRx, width=1280,height=1024 ,framerate=30/1 ! appsink
You will receive RGB32 in byte order BGRA in the sink in appsink.

Stefan

TIS-Stefan, 2022-06-02:
And the other error is already discussed above a few times......

Again you tried to set a video format, while using the camera default v4l2 backend, with a not available video format of 1024x768.

FantasticCHH, 2022-06-02:
Oops I think that is language problem.That's my fault.
Picture size do not have problem now.(thanks for your help)
Now , I want to add the format "RGB"(BGRA)
looks error?
[image]

TIS-Stefan, 2022-06-02:
My fault. Must be "tcambin" instead of "tcamsrc":

gst-launch-1.0 tcambin serial=18714599  ! video/x-raw, format=BGRx, width=1280,height=1024 ,framerate=30/1 ! appsink

And you wont see an image, because the images go into memory into the appsink.

FantasticCHH, 2022-06-03:
[image]

TIS-Stefan, 2022-06-03:
It is again in your command line window:
wrong : tcam**src**
correct tcam**bin**

FantasticCHH, 2022-06-03:
Sry,wrong picture.But, I got already try the tcambin version.But it make the Jetson Nano restarted,after that it did't do anything change. 
[image]

TIS-Stefan, 2022-06-03:
Looks perfectly correct for me. You wanted RGB32 in memory, you got RGB32 in memory. That is the appsink. The behavior is exactly as I wrote. If you want a live image:
gst-launch-1.0 tcambin serial=18714599 ! video/x-raw, format=BGRx, width=1280,height=1024 ,framerate=30/1 ! videoconvert ! ximagesink

BTW: The DFK 33UP1300 does not run with Aravis on the Nano happens, because the DFK 33UP1300 is sending the image data too fast. Aravis is not that performant and on a slower ARM CPU it can not get all image data packets in time.

FantasticCHH, 2022-06-03:
Would you please let me know where the Format BGRXx and rggb different from?
Because of I enter the BGRx as like you gave ,but it run out the picture by rggb.
And I try to find the content about them OPENcv(rggb to RGBx),but it did not show anything.
By the way , can I use the picture by format  RGBx  not BGRx ,because I use the RGBx training on our windows platform.

TIS-Stefan, 2022-06-03:
BGRx = RGB32
rggb = Bayer Raw data

If set BGRx as format, the camera runs with rggb. Debayering is done on the computer. 

OpenCV can convert BGRx to RGBx.

FantasticCHH, 2022-06-03:
This picture at the left side means the Jetson Nano version,the other one means the Windows version. Both of them got much more different , the format looks have big problem, I got already try to use the BGR2RGB, it didn't be changed.Even I think this one is not a BGRx photo,is like the gray one.This photo came from my code , and I did not let photo be edited.How can I use BGRx format while I'm using my code.I got this question because of when I open the Tcam-Capture softeare ,the format setting always turn to the gray choice automatically , also the BGRx diid't be added in my Tcam-Capture.Should I open my code before I take this "gst-launch-1.0 tcambin serial=18714599 ! video/x-raw, format=BGRx, width=1280,height=1024 ,framerate=30/1 ! videoconvert ! ximagesink"every time? Or HOW CAN I use this camera convenient ?
[image]

TIS-Stefan, 2022-06-03:
Hello 

looks for saturation set to a low value. Try tcam-capture and set the properties.

FantasticCHH, 2022-06-03:
Even I think this one is not a BGRx photo,is like the gray one!!!!!
This photo came from my code , and I did not let photo be edited.
How can I use BGRx format while I'm using my code???????
I got this question because of when I open the Tcam-Capture softeare ,the format setting always turn to the gray choice automatically , also the BGRx diid't be added in my Tcam-Capture.
Should I open my code before I take this "gst-launch-1.0 tcambin serial=18714599 ! video/x-raw, format=BGRx, width=1280,height=1024 ,framerate=30/1 ! videoconvert ! ximagesink"every time??? Or HOW CAN I use this camera convenient ????

TIS-Stefan, 2022-06-03:
The "gray" one does not look gray to me. 
You do not have the choice for BGRx, because the format of your graphics board is used by default. 
How does the image look in tcam-capture?

Is it possible. the white balance made the image gray? If the image is very homogeneous, the white balance "Grayworld" makes the image gray.

FantasticCHH, 2022-06-03:
After I gave you the message, I got to fix the white balance, I think you are right.
the second photo of the bottom as like you said.
I get another problem, can I turn on any setting to let it get bigger vision or more higher FPS? 
I got let it close to mine hope (1024*768), but it can do it .Even I try it to show 720*480 but it can't be work.
And I got to try it show 640*480(this setting can show on 30 FPS successfully ) out 40FPS but it can't be work too .
By the way why the interface only show three not add BGRx?
Thanks for your help. 👍 
[image]

[image]

TIS-Stefan, 2022-06-03:
run
tcam-ctrl -c 18714599 
and check the list of provided frame rates. Do not expect the maximum frame running, because the computer is too slow. The other one should do.

FantasticCHH, 2022-06-06:
I got another question about the Tcam-Capture.
When I were trying conctrl the white balance, the setting need "1.59".But this value will be change to 1.58 after I conctrl it.I want to know how can I use 1.59 again.Because I got use 1.59 once,and it make the photo close to our windows version.After that ,I haven't seen it.I will care about that "0.01".
[image]

[12 more replies]