TheImagingSource/tiscamera #465: Cannot stream with high resolution when using the library "TIS.py"

YumaKoyamatsu, 2022-05-14:
Hi.
I'm using DFK33UX273 with Aravis on Ubuntu 18.04.LTS.
I tried "tcam-ctrl -c serial-aravis" on terminal, and found that I could set a resolution as 1440 x 1080 at 200 FPS.
I used this settings for my Python code as below.

```
import TIS
import cv2
import time

TIS.openDevice("serial-aravis", 1440, 1080, "200/1", TIS.SinkFormats.BGRA,True) 
TIS.Start_pipeline()  

time.sleep(10)

TIS.Stop_pipeline()
cv2.destroyAllWindows()
```
But a black screen was displayed and the following message was output to std.
```
tcambin name=source ! capsfilter name=caps ! tee name=t t. ! queue ! videoconvert ! ximagesink t. ! queue ! appsink name=sink
Error starting pipeline. 

```
When I set lower resolution as below, It worked fine.
```
TIS.openDevice("serial-aravis", 640, 480, "200/1", TIS.SinkFormats.BGRA,True) 
```
Furthermore I ran GStreamer directly in the terminal as below.
```
gst-launch-1.0 tcambin serial=serial-aravis ! video/x-raw, format=BGR, width=1440, height=1080, framerate=200/1 ! videoconvert ! ximagesink

```
 It worked fine and output the following message.
```
Working with src caps: video/x-bayer, framerate=(fraction)200/1, width=(int)1440, height=(int)1080, format=(string)rggb
New clock: GstSystemClock
```
How can I get it to work properly with high resolution using Python code?

Thank you.

TIS-Stefan, 2022-05-16:
It seems, that the image data transfer is interrupted on too high frame rates. This leads to incomplete images, which are discarded. Therefore, you may start at

IS.openDevice("serial-aravis", 1440, 1080, "100/1", TIS.SinkFormats.BGRA,True) 

and increase the frame rate until you do not get frames any more.

Aravis is not as performat as v4l2.  You need a good computer for this high frame rates. 

Stefan

TIS-Edgar, 2024-04-12:
Closed due to inactivity.

closed 2024-04-12 by TIS-Edgar