TheImagingSource/tiscamera #466: Which packet in ros melodic

havvanurbozomeroglu, 2022-05-19:
Hello, 

ı have DFK 33GX236 GigE color industrial camera. i have to use this camera in ros. which package should I install?
(uvc_camera, uvc_cam, usb_cam, gscam, gencam_cu etc.). ı can't find any documentation about it. ı will be very happy if you help.

TIS-Stefan, 2022-05-19:
Hello

I would use this: http://wiki.ros.org/camera_aravis

Stefan

havvanurbozomeroglu, 2022-05-19:
Thank you very much for your reply @TIS-Stefan
there is only noetic tag for this package. ı use melodic. no problem, would it? ı don't have the camera with me right now. so i can't test it.

TIS-Stefan, 2022-05-19:
Hello

I am not an ROS expert nor an ROS supporter. Therefore, I can not answer your question. I searched for "ROS" and "Aravis" and found the link. "Aravis" is the Gige Vision library we use for the cameras in Linux.

Stefan

closed 2022-05-19 by havvanurbozomeroglu