TheImagingSource/tiscamera #468: MIPI CSI 3 cameras drivers

hidon, 2022-05-26:
hello Stefan
is there MIPI CSI 2 cameras drivers for jetpack 4.6 version?

TIS-Stefan, 2022-05-26:
​ 

Hello

Yes. Please make a contact on https://www.theimagingsource.com/company/contact/.

Since we have holidays, I can send the files next Monday.
------------------------------------------------------------------------------------------------------------------------------ 

  
 Was this message helpful?   Yes   [https://theimagingsource.deskpro.com/ticket-rate/BDRF-1629-KOKC/WQ755D9KMAPP64R/400487?setrating=1]   It was OK   [https://theimagingsource.deskpro.com/ticket-rate/BDRF-1629-KOKC/WQ755D9KMAPP64R/400487?setrating=0]   No   [https://theimagingsource.deskpro.com/ticket-rate/BDRF-1629-KOKC/WQ755D9KMAPP64R/400487?setrating=-1] 

View and manage this ticket online: https://theimagingsource.deskpro.com/tickets/83748 

  The Imaging Source [https://www.theimagingsource.com] · https://theimagingsource.deskpro.com/ 

 (#ETXCWQ755D9KMAPP64R)

hidon, 2022-05-30:
Hello Stefan,

Can mipi csi 2 camera ribbon Flexible Flat Cable length  be maximum?

Halil.

TIS-Stefan, 2022-05-30:
Hello Halil

Typically 20cm length for MIPI flat cable connection

hidon, 2022-05-31:
I need 50-60 cm length MIPI flat cable connection. How can I do?

TIS-Stefan, 2022-05-31:
You can not. Either use USB or FPD link with serializer and deserializer boards. BTW: this is a sales question, not a tiscamera issue. Please make a contact at https://www.theimagingsource.com/company/contact/

hidon, 2022-05-31:
ok thanks Stefan. 
tiscamera + MIPI flat cable connection --> configuration I 
tiscamera + MIPI flat cable connection +FPD link with serializer and deserializer board  --> configuration II
Is Configuration II of delay  more than configuration?  
The configureation II is how much time delay than configure I?

TIS-Stefan, 2022-05-31:
The delay is the same for both, as far as I know. BTW: this is a sales question, not a tiscamera issue. Please make a contact at https://www.theimagingsource.com/company/contact/

TIS-Edgar, 2024-04-12:
Closed due to inactivity.

closed 2024-04-12 by TIS-Edgar