TheImagingSource/tiscamera #469: tcamsrc not found

kmcgrathgenerate, 2022-06-09:
I am working with a DMK37U camera and followed the instructions but for some reason I am unable to run the python wrapper code. I get this error from TIS.py

```
Traceback (most recent call last):
  File "test.py", line 8, in <module>
    if not Tis.selectDevice():
  File "/home/admin_temp/repos/image-based-cell-sorting/cellsorter/TIS.py", line 281, in selectDevice
    serials = source.get_device_serials()
AttributeError: 'GstTcamSrc' object has no attribute 'get_device_serials'
```
After some further digging, I realized that tcamsrc is not found on my cli and I think that has something to do with the installation. Here are the steps I performed:

```
sudo ./scripts/dependency-manager install
mkdir build
cd build

cmake -DTCAM_BUILD_ARAVIS=ON -DTCAM_BUILD_ARAVIS=ON -DTCAM_BUILD_TOOLS=ON -DTCAM_BUILD_V4L2=ON -DTCAM_BUILD_LIBUSB=ON -DTCAM_BUILD_DOCUMENTATION=ON

make
sudo make install
```
Here is showing I have the tcam-ctrl active and I do have devices that are available:

```
$tcam-ctrl -l
Model: DMK 37BUX287 Serial: 36120054 Type: v4l2
Model: DMK 37BUX287 Serial: 36120054 Type: aravis
```
And here is tcamsrc not found
```
$ tcamsrc
tcamsrc: command not found
```
Any idea where I went wrong?

TIS-Stefan, 2022-06-10:
Hello

The tis.py is current written for tiscamera 0.14. I have no time to update it for tiscamera 1.0. I have a small sample only, that has an adapted TIS.py: 
[tiscamera1.0-opencv-trigger.zip](https://github.com/TheImagingSource/tiscamera/files/8876883/tiscamera1.0-opencv-trigger.zip)
You may try this.

Stefan

kmcgrathgenerate, 2022-06-10:
Thank you for the quick response, should be able to get this work pretty easily.

TIS-Stefan, 2022-06-10:
Simply take the class from the Zip file :-)
Have a nice weekend!

TIS-Edgar, 2024-04-12:
Closed due to inactivity.

closed 2024-04-12 by TIS-Edgar