TheImagingSource/tiscamera #470: Various patches required to package tiscamera-v1 on nixos

jraygauthier, 2022-06-20:
Hi,

we just packaged and merged latest version of tiscamera into nixpkgs `master` branch.

This is nice and seems to work well.

However, in order to package the v1 software, multiple patches were required. See the `*.patch` files under:

https://github.com/jraygauthier/nixpkgs/tree/11af729cd0535c48b22f65edc5112ec1fb04d7c7/pkgs/os-specific/linux/tiscamera

I think a lot of these could be easily applied to your `master` branch and will simplify packaging of further versions of the software.

Thanks in advance.

TIS-Edgar, 2022-06-22:
Hi,

Sorry for the late reply. The patches seem fine to me. Do you want to create a merge request or would it be ok if i simply committed the changes myself?

jraygauthier, 2022-06-22:
Hi again, @TIS-Edgar. I will gladly open a merge request. Coming back with it soon.

jraygauthier, 2022-06-22:
PR #471 applying the patches on top of the master branch ready to be merged.

closed 2022-06-22 by TIS-Edgar