TheImagingSource/tiscamera #472: Based on Ubuntu18.04 state question

FantasticCHH, 2022-06-26:
Sry for bothering for the second time.
I got another question about let the Tcam-capture catch the setting value(.json)file.We are trying making easily do our code,and also finding how to let the file add by python cv.Now we are trying to let the settting exchange by the state as like the picture.But we saw that, when we want to open the new file which we were saved without .josn , We saw this menu was been exchanged,but after that we try to see the view which did not be changed. Could you please let me now How can I fixed it?
[image]

Allen-Yang-github, 2022-06-28:
Dear sir,
抱歉因為看不太懂您的需求是什麼，是在python中讀取.json檔案並寫入到相機中嗎?

FantasticCHH, 2022-06-30:
我想詢問一下關於您提到的透過CODE導入相機參數的部分。有一個確切的網址提供我參考嗎?感謝您
還是是這兩個嗎 是否有其他的呢
https://github.com/TheImagingSource/Linux-tiscamera-Programming-Samples/blob/master/Auto%20Focus%20On%20Push%20in%20Python/Program.py
https://github.com/TheImagingSource/Linux-tiscamera-Programming-Samples/tree/master/Auto%20Focus%20On%20Push%20in%20Python#auto-focus-on-push-in-pyhton

Allen-Yang-github, 2022-06-30:
您好,
設Python Properties請參考以下連結:
https://github.com/TheImagingSource/tiscamera/blob/master/examples/python/02-set-properties.py

TIS-Stefan, 2022-06-30:
Loading a json type property file

tcam-capture shows in the "info" function a json style property set of the currently used camera. This string can be saved to a file and loaded in an own Python script later. The tcambin property which receives the string is [tcam-properties-json](https://www.theimagingsource.com/documentation/tiscamera/tcam-gstreamer.html?highlight=tcambin#tcam-properties-json).

Based on the "03-live.py" sample the saved file "state1.json" is loaded into a string:
```Python
state=""

with open("state2.json","r") as f:
    state = f.read()
```
Then the pipeline is created and a serial number assigned:
```Python
 serial = "45219982"
 pipeline = Gst.parse_launch("tcambin name=bin "
                                " ! videoconvert"
                                " ! ximagesink sync=false")

    # retrieve the bin element from the pipeline
    camera = pipeline.get_by_name("bin")

    # serial is defined, thus make the source open that device
    if serial is not None:
        camera.set_property("serial", serial)
```
Next line is passing the string in "state", which contains the property settings, to the tcambin:

```Python
camera.set_property("tcam-properties-json",state)
```
The pipeline can start now:
```Python
    pipeline.set_state(Gst.State.PLAYING)
```
Attention: It seems, there is a bug in tcam-capture 1.0. The device state in the info tab is not correctly updated, therefore, restart tcam-capture, before creating a new state.

FantasticCHH, 2022-07-04:
I have no idea about this part.
Could you please let me know how can I do it?

[image]

TIS-Stefan, 2022-07-04:
That is the Python source code, which loads the saved properties. The properties have be saved from tcam-capture.

It is, what you asked for.

I ask my Chinese colleague to go in contact with you.

Stefan

FantasticCHH, 2022-07-09:
could you please provide me a python code about the sample code (NO.12 tcam-properties.py) and (NO.3 live-stream.py)
As like when I were using the camera ,I can use my function contral

TIS-Stefan, 2022-07-11:
Hello

I suppose, you had a phone call with my colleague about software development already.

Stefan

TIS-Stefan, 2022-07-20:
Can you please show the error messages, when white balance is missing?
Also can you show your script?
Did you set the pipeline into READY state, before setting properties?

TIS-Stefan, 2022-07-20:
Do you have exact steps to reproduce the issue?

FantasticCHH, 2022-07-22:
I'm sorry about I forgot to update the SDK version.
After I sent the mail to you ,I got try to update it .
I think the question had been fixed .

FantasticCHH, 2022-10-11:
您好 感謝您的回覆
幾天前有在github上方針對在Ubuntu平台的tiscamera軟體
在state介面的部分，我們有嘗試著把先前使用的參數檔進行檔案留存，並進行參數導入及更新成我們所需要的設定。發現這部分似乎是有點bug。在使用上發現
我們的確能夠將上面的值更新到我們要的狀態，但是跑出來的畫面似乎是沒辦法連動的。
您提到的部分是我們想完成的目標，但目前在軟體上方發現一些問題，不曉得如果透過外界隱函數導入，不曉得是否又能夠執行。

TIS-Stefan, 2022-10-11:
Hello

I tried to translate your text with different translators, but I am sorry, I do not really understand, what you want to do. (Allen is not available at this point of time.)

I understand, you use tcam-capture in order to save camera property values into a json file. In your own script, you read that json file again, after you changed some values. You find that somewhat buggy, but you did not describe what is buggy. The translation I got is " but the screen that runs out seems to be unable to be linked." That translation does not really makes sense. I assume, the image you see in the live view is not, what you expect. 
I suggest:
Please post the json file.
Please show the result, you get
Please show the result, you want.

Thank you in advance!

Stefan

FantasticCHH, 2022-10-11:
All of the problems had been solved.
Thanks for your help.