TheImagingSource/tiscamera #473: pipeline does't want to pause

rangeetmanna, 2022-07-20:
i use a camera DMK37BUX273 and I have ubuntu OS. I can run tcam-caputure and gst-launch-1.0 tcambin ! videoconver! ximagesink, but I can't able to save that stream . When I use the lauch code from save_stream_file (like gst-lauch-1.0 tcambin ! video.... ! tee ......queue......ximagesink t.!......avimux!..... filesink name=fsink): this one is provided in the tis docu in example section, it says that error: pipeline does't want to pause, error: from element.....No file name specified for writing. ....Freeing pipeline.......

How can I save that live stream video with a small size in a avi file format? Please help me out? I mean I can watch the video and can't save it using the provided example code by tis docu. What may be the issues? Any leads?

Thank you very much!

Best,
Rangeet

TIS-Stefan, 2022-07-21:
As I wrote by email to you already:

I found in the internet this:
https://gist.github.com/liviaerxin/bb34725037fd04afa76ef9252c2ee875#mux-video-and-audio

# save to `*.mp4` file
gst-launch-1.0 -v -e videotestsrc \
  ! x264enc \
  ! mp4mux name=mux \
  ! filesink location="bla.mp4" audiotestsrc \
  ! lamemp3enc \
  ! mux.
You do not have audio, so following should be fine:
# save to `*.mp4` file
gst-launch-1.0 -v -e tcambin \
  ! videoconvert \
  ! x264enc \
  ! mp4mux name=mux \
  ! filesink location="bla.mp4"

rangeetmanna, 2022-07-21:
Yes, thank you very much Stefan, you are a lifesaver. Sorry, I did not check my email. 

It works now. Thanks a lot!

TIS-Stefan, 2022-07-21:
:-)

closed 2022-07-21 by TIS-Stefan