TheImagingSource/tiscamera #474: How to record and save the stream in mp4

rangeetmanna, 2022-07-23:
Hi Stefan,

I am now able to save the stream in mp4 format, but do you know how to view and save the stream simultaneously? I have tried the available code with videotestsrc  and it worked, but when I changed the videotestsrc to tcambin, it is not working anymore.

Need your help again!

Thanks in advance!

TIS-Stefan, 2022-07-25:
You use a [tee ](https://gstreamer.freedesktop.org/documentation/coreelements/tee.html?gi-language=c) module for splitting the stream:

```
tcambin name=tcam0 ! video/x-raw,format=BGRx ! tee name=teemodule teemodule.
! queue max-size-time=0 max-size-bytes=0 max-size-buffers=0 ! videoconvert n-threads=4
! ximagesink sync=false teemodule.
! queue name=save-queue max-size-time=0 max-size-bytes=0 max-size-buffers=0
! videoconvert n-threads=4
! queue
! x264enc
! mp4mux ! queue max-size-time=0 max-size-bytes=0 max-size-buffers=0
! filesink async=true sync=false name=save-sink location=~/awesome-stream.mp4
```
`max-size-time=0 max-size-bytes=0 max-size-buffers=0` disables queues leaking. This is necessary because x264enc itself can cache up to 3-4 seconds (depending on the system), which can cause problems in other sinks.

Please be aware that encoder and filesink always require an explicit EOS to correctly write a valid file. when testing with gst-launch use -e to send an EOS when pressing ctrl-c.

Stefan

FantasticCHH, 2022-07-25:
I just use the newest SDK to contral the camera parameter and also store
the picture(1picture/1sec) ,when I store that I also use the python
code(flask) to let it get that picture keep showing on our html.
Sry, tomorrow is our Research Paper testing , Maybe I can help you after
tomorrow.

By the way, after finishing fixing SDK. I did not use the code as like you
gave .
Maybe I can research that after tomorrow.

rangeetmanna, 2022-07-25:
it is not running completely, it is showing that : bash: !queue: event not found and thread is not getting converted from 0 to 4.

rangeetmanna, 2022-07-25:
gst-launch-1.0 -vvv videotestsrc
! tee name=t
t. ! queue ! x264enc ! mp4mux ! filesink location=xyz.mp4 -e
t. ! queue leaky=1 ! autovideosink sync=false

This runs well for videotestsrc but when I changed it to tcambin, error is prompting (.ERROR: from element /GstPipeline:pipeline0/GstTcamBin:tcambin0/GstTcamSrc:tcambin-source/GstTcamMainSrc:tcammainsrc0: Internal data stream error.
Additional debug info:)
 Any solution?

TIS-Stefan, 2022-07-25:
At "!queue" a space is missing. Should be "! queue".

rangeetmanna, 2022-07-25:
Thanks Stefan, it has worked perfectly. Danke schön!

closed 2022-07-25 by TIS-Stefan