TheImagingSource/tiscamera #475: supported caps not working for DFM 72BUC02-ML

djozwiak, 2022-08-02:
tcam-ctrl shows that the following caps are available for DFM 72BUC02-ML - 
Available gstreamer-1.0 caps:
video/x-raw,format=GRAY8,width=2592,height=1944,framerate={15/2,7/1,5/1,4/1};
video/x-raw,format=GRAY8,width=1920,height=1080,framerate={15/1,10/1,15/2,5/1};
video/x-raw,format=GRAY8,width=1280,height=720,framerate={30/1,25/1,15/1,10/1};
video/x-raw,format=GRAY8,width=640,height=480,framerate={60/1,30/1,25/1,15/1};

However, only 1920x1080 and 640x480 actually work. 2592x1944 shows an error and, while 1280x720 does not show an error, there is no video displayed. This was tested with versions 0.12.0 and 0.14.0. The following pipeline was used for testing (note that I'm interested only in the 1280x720 format so that I can avoid scaling) -

gst-launch-1.0 tcambin ! video/x-raw,format=GRAY8,width=1280,height=720,framerate=15/1 ! glimagesink

TIS-Kevin, 2022-08-04:
Dear djozwiak,
we are currently working on your case.
I tried to reproduce your problem and did not receive an error while working with the format 2592x1944. So what is the error you receive?

For 1280x720 it seems to not work for me too with tiscamera 0.14 and 1.0 (newest Version). So we have to investigate on that.

djozwiak, 2022-08-05:
With GST_DEBUG=4 I see the following error:

0:00:07.587261471  1339 0x555f4cd8b800 ERROR                tcambin gsttcambin.cpp:1284:gst_tcam_bin_change_state:<tcambin0> Unable to work with given caps: video/x-raw,
format=(string)GRAY8, width=(int)2592, height=(int)1944, framerate=(fraction)15/1

However note that I'm not trying to use that resolution...that was just to do a thorough test. The resolution that I'd like to work is 1280x720.

djozwiak, 2022-08-17:
Hi Kevin:
Thanks very much. I have tested this new firmware and the 1280x720 caps are now working.

closed 2022-08-17 by djozwiak