TheImagingSource/tiscamera #478: TIS con Jetson Nano

Ukhupacha, 2022-08-22:
Dear TIS,

I got a DFK 33UX273 camera connected by usb to a jetson nano.
I've been trying to use it with the jetson https://github.com/dusty-nv/jetson-inference that uses v4l2 for the camera but I can't seem to make it work. I took a look at https://github.com/TheImagingSource/nvidia-jetson-samples but they seem to work only with MIPI cameras.

Do you have any advice on how to make it work?

Kindly,

Ukhu

TIS-Alexander, 2022-08-23:
Dear Ukhu,

in fact, the examples delivered with our Jetson development kits, which are hosted on github, are designed to work with our MIPI cameras, and utilize the ISP of the NVIDIA platform. However, it should be possible to use the examples with slight modifications with other camera types as well. Basically, the GStreamer pipeline must be changed for this to use another camera type. If you have tiscamera installed, you should be able to replace the GStreamer element nvarguscamerasrc (which is used to deliver images from our MIPI cameras and preprocess the images using the builtin ISP) with tcambin, which delivers images from all our industrial cameras.

Please note that the images from nvarguscamerasrc are delivered in graphics memory (NVMM) whereas tcambin can only deliver images in CPU memory. The already present GStreamer element nvvidconv should take care of the required conversion. Could you please try out this change in the context of the face detection example?

Best regards
Alexander

TIS-Edgar, 2024-04-12:
Closed due to inactivity.

closed 2024-04-12 by TIS-Edgar