TheImagingSource/tiscamera #479: Stream into Web Browser and Setting camera properties

dinhtuanan, 2022-09-08:
Hi,
I'm using Model: DFK 33GX265 Serial: 31120204 aravis on Ubuntu 18.04 LTS
I tried https://github.com/TheImagingSource/Linux-tiscamera-Programming-Samples/tree/master/python/webserver to stream into web browser.
I'm having 2 problems right now:
1. How can i start and stop the stream whenever i want
My **camera_setting.py** (server)
```
import app.camera_setting.tcam_camera as tcam_camera
import logging

camera = tcam_camera.tcam_camera()
def imagegenerator():
    try:
        while True:
            frame = camera.snap_image(1000)
            if frame is not None:
                # Create the boundary between the sent jpegs
                yield (b'--imagingsource\r\n'
                    b'Content-Type: image/jpeg\r\n\r\n' + frame + b'\r\n')
            else:
                logging.info("No frame received.")

            # Send 10 frames per second only
            time.sleep(0.1) 
    finally:
        logging.info("Stream closed")

@camera_setting.route('/mjpeg_stream')
def imagestream():
    logging.info("Starting mjpeg stream")
    return Response( imagegenerator() , mimetype='multipart/x-mixed-replace; boundary=imagingsource')
```
My **camera_setting.html**
```
<img src="/camera_setting/mjpeg_stream" class="" alt="" />
```
2. I would like to know if i can set the properties of camera (brightness, contrast, gain, exposure, auto reference, auto max value, highlight reduction,...) via **tcam_camera** class or **somehow to set properties** (below is my website so you can imagine)
`If it's possible to set the properties while streaming, that's great, if not, set the properties while stopping the stream and then starting the stream again, that's great too`

[image]

Regards,
An

TIS-Stefan, 2022-09-08:
Hello An

Yes, you can set these properties while streaming. For a start, you may look into the tiscamera samples at https://github.com/TheImagingSource/tiscamera/tree/master/examples/python 

I do not dare to... but I would like to point to the tiscamera documentation at https://www.theimagingsource.com/documentation/tiscamera/ too.

Stefan

dinhtuanan, 2022-09-08:
Hi Stefan,
I set properties of camera via TIS class (set_property) but it doesn't have brightness, contrast, exposure properties. It said that these properties hasn't implemented. Where can I set those attributes?

TIS-Stefan, 2022-09-08:
Please run the program "tcam-capture". There you can see the properties and their names. E.g. "exposure" is "Exposure Time". 
You can also run tcam-ctrl -p xxxxxxxxx (serial number) in order to get a list of properties and their correct names.

TIS-Stefan, 2022-09-08:
Some more properties are available after installing tcam-dutils from https://www.theimagingsource.com/support/downloads-for-linux/

dinhtuanan, 2022-09-09:
Hi Stefan,
I tried tcam-ctrl -p xxxxxxxxx (serial number) and get the list of properties but there's no brightness and contrast properties. 
Thank you.

TIS-Stefan, 2022-09-09:
Brightness is "Blacklevel". Do not mix it with Gain and Exposure. Usually you use Gain and ExposureTime to set the image brightness, not the "Blacklevel" property. 

Contrast is not necessarily available and I must admit, I do not know, whether it is built into tcamdutils. If contrast is not listed, then it is simply not available and you can not get it.

Stefan

TIS-Edgar, 2024-04-12:
Closed due to inactivity.