TheImagingSource/tiscamera #480: Ubuntu 20.04.4: "./scripts/dependency-manager install" fails

StohanzlMart, 2022-09-19:
If you scroll down on the following link, you will see, that the version your dependency manager is trying to download does not exist: http://security.ubuntu.com/ubuntu/pool/main/g/gst-plugins-base1.0/ has no libgstreamer-plugins-base1.0-dev_1.16.2-4ubuntu0.1_amd64.deb

What is the quickest way to fix this? Thank you!

```
Es müssen noch 383 kB von 19,2 MB an Archiven heruntergeladen werden.
Nach dieser Operation werden 87,2 MB Plattenplatz zusätzlich benutzt.
Möchten Sie fortfahren? [J/n] j
Ign:1 http://de.archive.ubuntu.com/ubuntu focal-updates/main amd64 libgstreamer-plugins-base1.0-dev amd64 1.16.2-4ubuntu0.1
Fehl:1 http://security.ubuntu.com/ubuntu focal-updates/main amd64 libgstreamer-plugins-base1.0-dev amd64 1.16.2-4ubuntu0.1
  404  Not Found [IP: 141.30.62.25 80]
E: Fehlschlag beim Holen von http://security.ubuntu.com/ubuntu/pool/main/g/gst-plugins-base1.0/libgstreamer-plugins-base1.0-dev_1.16.2-4ubuntu0.1_amd64.deb 404  Not Found [IP: 141.30.62.25 80]
E: Einige Archive konnten nicht heruntergeladen werden; vielleicht »apt-get update« ausführen oder mit »--fix-missing« probieren?
```

TIS-Stefan, 2022-09-19:
Hallo 

Haben sie denn mal da 
sudo apt-get update
ausgeführt? Damit werden die Linux Paketquellen erneuert.

Did you run 
sudo apt-get update
? This updates the Linux packet sources.

Stefan

StohanzlMart, 2022-09-19:
Hello and thank you!

Yes I did `sudo apt update && sudo apt upgrade -y` but the upgrade also returned the same errors on some packages. `sudo apt-get update --fix-missing` didn't help too.

TIS-Edgar, 2022-09-20:
I am unsure where this error comes from. The dependency manager tells you what it is executing:

> Executing: sudo apt install git g++ cmake pkg-config libgstreamer1.0-dev libgstreamer-plugins-base1.0-dev libglib2.0-dev libgirepository1.0-dev libzip-dev libzip4 libglib2.0-0 libgirepository-1.0-1 libusb-1.0-0-dev libusb-1.0-0 uuid-dev libuuid1 libudev-dev libudev1 python3-sphinx libc6 libgstreamer1.0-0 gstreamer1.0-plugins-base gstreamer1.0-plugins-good gstreamer1.0-plugins-bad gstreamer1.0-plugins-ugly qtgstreamer-plugins-qt5 libxml2-dev libxml2 ninja-build libqt5widgets5 libqt5gui5 qtbase5-dev graphviz

As you can see there are no versions (they are only used when creating packages).
So the question is where those come from. Lets start with cleaning the cache

Can you give me the output of the following commands:

> sudo apt-cache show libgstreamer-plugins-base1.0-dev
> sudo apt clean
> sudo apt update
> scripts/dependency manager

StohanzlMart, 2022-09-29:
Thank you for your help! I got it working by installing the precompiled x86 Ubuntu 20.04 .deb file.

closed 2022-09-29 by TIS-Stefan