TheImagingSource/tiscamera #481: Missing dependencies when installing tiscamera for Ubuntu 16.04

jdog816, 2022-09-20:
Hello,

I'm trying to run a DFM 22BUC03-ML camera on a 32-bit Odroid XU4 with Ubuntu 16.04. I followed the instructions here https://github.com/TheImagingSource/tiscamera to install tiscamera, but when I go to install the ./scripts/dependency-manager I get the following message:

          Aborted loading dependencies. Reason: Unable to find file: '/usr/tiscamera/scripts/../dependencies/ubuntu_1604.dep'

Can you provide the ubuntu_1604.dep file, or update the repo so it can be installed?

Thanks

jdog816, 2022-09-22:
Yes, I'm switching to a C4 odroid (64 bit) to see if it works on that, will let you know.

TIS-Stefan, 2022-09-22:
Very well!

jdog816, 2022-09-22:
Got a little further with the C4 in Ubuntu 20.04. I was able to clone tiscamera and install the dependencies. But I get an error when I try to cmake the V4L2 option. I'll post another ticket for that, since it's a separate issue. Thanks

closed 2022-09-22 by jdog816