TheImagingSource/tiscamera #482: Error when trying to cmake V4L2 option for tiscamera

jdog816, 2022-09-22:
I'm running Ubuntu 20.04 and was able to follow the instructions here for installing tiscamera, up until the cmake option - where I get the following error when trying to add the V4L2 option:

root@odroid:/etc/tiscamera/build# cmake -DTCAM_BUILD_V4L2=ON ..
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
CMake Error at CMakeLists.txt:21 (project):
  No CMAKE_C_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.

CMake Error at CMakeLists.txt:21 (project):
  No CMAKE_CXX_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

-- Configuring incomplete, errors occurred!
See also "/etc/tiscamera/CMakeFiles/CMakeOutput.log".
See also "/etc/tiscamera/CMakeFiles/CMakeError.log".

So I checked the output and error logs and got this:

root@odroid:/etc/tiscamera/CMakeFiles# cat CMakeOutput.log
The system is: Linux - 4.9.277-82 - aarch64
root@odroid:/etc/tiscamera/CMakeFiles# cat CMakeError.log
Compiling the C compiler identification source file "CMakeCCompilerId.c" failed.
Compiler: CMAKE_C_COMPILER-NOTFOUND 
Build flags: 
Id flags:  

The output was:
No such file or directory

Compiling the C compiler identification source file "CMakeCCompilerId.c" failed.
Compiler: CMAKE_C_COMPILER-NOTFOUND 
Build flags: 
Id flags: -c 

The output was:
No such file or directory

Compiling the C compiler identification source file "CMakeCCompilerId.c" failed.
Compiler: CMAKE_C_COMPILER-NOTFOUND 
Build flags: 
Id flags: -Aa 

The output was:
No such file or directory

Compiling the C compiler identification source file "CMakeCCompilerId.c" failed.
Compiler: CMAKE_C_COMPILER-NOTFOUND 
Build flags: 
Id flags: -D__CLASSIC_C__ 

The output was:
No such file or directory

Compiling the C compiler identification source file "CMakeCCompilerId.c" failed.
Compiler: CMAKE_C_COMPILER-NOTFOUND 
Build flags: 
Id flags: --target=arm-arm-none-eabi;-mcpu=cortex-m3 

The output was:
No such file or directory

Any ideas?

TIS-Stefan, 2022-09-23:
My first question is: Did you run 
./scripts/dependency-manager install
from the tiscamera directory?
Stefan

jdog816, 2022-09-23:
Yes I did. This is what I got:

          root@odroid:/etc/tiscamera# ./scripts/dependency-manager install
          
          Executing: sudo apt install git g++ cmake pkg-config libgstreamer1.0-dev libgstreamer-plugins-base1.0-dev libglib2.0-dev libgirepository1.0-dev libusb-1.0-0-dev libzip-dev libzip5 libglib2.0-0 libgirepository-1.0-1 libusb-1.0-0 libuuid1 uuid-dev libudev-dev libudev1 python3-sphinx libgstreamer1.0-0 gstreamer1.0-plugins-base gstreamer1.0-plugins-good gstreamer1.0-plugins-bad gstreamer1.0-plugins-ugly libxml2-dev libxml2 ninja-build libqt5widgets5 libqt5gui5 qtbase5-dev graphviz
          
          Reading package lists... Done
          Building dependency tree       
          Reading state information... Done
          cmake is already the newest version (3.16.3-1ubuntu1).
          libusb-1.0-0 is already the newest version (2:1.0.23-2build1).
          libusb-1.0-0 set to manually installed.
          pkg-config is already the newest version (0.29.1-0ubuntu4).
          pkg-config set to manually installed.
          libzip5 is already the newest version (1.5.1-0ubuntu1).
          libzip5 set to manually installed.
          git is already the newest version (1:2.25.1-1ubuntu3.5).
          libgirepository-1.0-1 is already the newest version (1.64.1-1~ubuntu20.04.1).
          libgirepository-1.0-1 set to manually installed.
          libglib2.0-0 is already the newest version (2.64.6-1~ubuntu20.04.4).
          libglib2.0-0 set to manually installed.
          libuuid1 is already the newest version (2.34-0.1ubuntu9.3).
          Some packages could not be installed. This may mean that you have
          requested an impossible situation or if you are using the unstable
          distribution that some required packages have not yet been created
          or been moved out of Incoming.
          The following information may help to resolve the situation:
          
          The following packages have unmet dependencies:
           gstreamer1.0-plugins-bad : Depends: libfluidsynth2 (>= 2.0.5) but it is not going to be installed
          E: Unable to correct problems, you have held broken packages.

And then I tried cmake with the errors I pasted above. Plugins-bad won't install, but I don't think I need it for my application.

jdog816, 2022-09-23:
Ok, I got a little further by running:

          apt install build-essential

and now it recognizes the C and C++ compilers, but I run into some more errors:

          root@odroid:/usr/tiscamera/build# cmake -DTCAM_BUILD_V4L2=ON ..
          -- The C compiler identification is GNU 9.4.0
          -- The CXX compiler identification is GNU 9.4.0
          -- Check for working C compiler: /usr/bin/cc
          -- Check for working C compiler: /usr/bin/cc -- works
          -- Detecting C compiler ABI info
          -- Detecting C compiler ABI info - done
          -- Detecting C compile features
          -- Detecting C compile features - done
          -- Check for working CXX compiler: /usr/bin/c++
          -- Check for working CXX compiler: /usr/bin/c++ -- works
          -- Detecting CXX compiler ABI info
          -- Detecting CXX compiler ABI info - done
          -- Detecting CXX compile features
          -- Detecting CXX compile features - done
          -- Setting build type to 'Release' as none was specified.
          -- TCAM_VERSION: 1.0.0, Git: commit#=4009, branch=master hash=2225d34d
          Package gobject-introspection-1.0 was not found in the pkg-config search path.
          Perhaps you should add the directory containing `gobject-introspection-1.0.pc'
          to the PKG_CONFIG_PATH environment variable
          No package 'gobject-introspection-1.0' found
          Package gobject-introspection-1.0 was not found in the pkg-config search path.
          Perhaps you should add the directory containing `gobject-introspection-1.0.pc'
          to the PKG_CONFIG_PATH environment variable
          No package 'gobject-introspection-1.0' found
          Package gstreamer-1.0 was not found in the pkg-config search path.
          Perhaps you should add the directory containing `gstreamer-1.0.pc'
          to the PKG_CONFIG_PATH environment variable
          No package 'gstreamer-1.0' found
          Package gstreamer-1.0 was not found in the pkg-config search path.
          Perhaps you should add the directory containing `gstreamer-1.0.pc'
          to the PKG_CONFIG_PATH environment variable
          No package 'gstreamer-1.0' found
          -- Build spdlog: 1.8.2
          -- Looking for pthread.h
          -- Looking for pthread.h - found
          -- Performing Test CMAKE_HAVE_LIBC_PTHREAD
          -- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
          -- Looking for pthread_create in pthreads
          -- Looking for pthread_create in pthreads - not found
          -- Looking for pthread_create in pthread
          -- Looking for pthread_create in pthread - found
          -- Found Threads: TRUE  
          -- Build type: Release
          -- Version: 7.1.3
          -- Build type: Release
          -- CXX_STANDARD: 17
          -- Performing Test has_std_17_flag
          -- Performing Test has_std_17_flag - Success
          -- Performing Test has_std_1z_flag
          -- Performing Test has_std_1z_flag - Success
          -- Performing Test SUPPORTS_USER_DEFINED_LITERALS
          -- Performing Test SUPPORTS_USER_DEFINED_LITERALS - Success
          -- Performing Test FMT_HAS_VARIANT
          -- Performing Test FMT_HAS_VARIANT - Success
          -- Required features: cxx_variadic_templates
          -- Looking for strtod_l
          -- Looking for strtod_l - not found
          -- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
          -- Checking for module 'gstreamer-1.0'
          --   No package 'gstreamer-1.0' found
          CMake Error at /usr/share/cmake-3.16/Modules/FindPkgConfig.cmake:463 (message):
            A required package was not found
          Call Stack (most recent call first):
[... 78 lines cut ...]
              8 |   return ((int*)(&strtod_l))[argc];
                |                   ^~~~~~~~
                |                   strtoull
          /usr/tiscamera/build/CMakeFiles/CMakeTmp/CheckSymbolExists.c:8:19: note: each undeclared identifier is reported only once for each function it appears in
          make[1]: *** [CMakeFiles/cmTC_0cadb.dir/build.make:66: CMakeFiles/cmTC_0cadb.dir/CheckSymbolExists.c.o] Error 1
          make[1]: Leaving directory '/usr/tiscamera/build/CMakeFiles/CMakeTmp'
          make: *** [Makefile:121: cmTC_0cadb/fast] Error 2
          
          
          File /usr/tiscamera/build/CMakeFiles/CMakeTmp/CheckSymbolExists.c:
          /* */
          #include <stdlib.h>
          
          int main(int argc, char** argv)
          {
            (void)argv;
          #ifndef strtod_l
            return ((int*)(&strtod_l))[argc];
          #else
            (void)argc;
            return 0;
          #endif
          }

The output log has so much data I can't scroll all the way to the top to copy from the beginning.

TIS-Stefan, 2022-09-26:
Unfortunately I do not have an Odroid C4. Therefore, I can not really be of help
What you can do is search for the error message in the internet, e.g.
"Odroid C4 Ubuntu 20.04 Package gobject-introspection-1.0 was not found"
You will get results like this: https://forums.linuxmint.com/viewtopic.php?t=95067.

That is the only hint, I can give.

Stefan

jdog816, 2022-09-26:
I was able to install tiscamera on an Odroid XU4 running Ubuntu 18.04.  However, I get this error when trying to run tcam-capture:

     root@odroid:/home/odroid/Camera/tiscamera# tcam-capture
     2022-09-26T17:35:09 - unknown:0: warning QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
     2022-09-26T17:35:11 - unknown:0: info New device: 46410017-v4l2
     2022-09-26T17:35:14 - unknown:0: info device selected: DFM 22BUC03-MLA-46410017-v4l2

     2022-09-26T17:35:14 - unknown:0: debug Setting 'conversion-element' property to 'auto'
     2022-09-26T17:35:15 - unknown:0: info settings caps to : video/x-raw, format=(string)GRAY8, width=(int)640, height=(int)480, 
     framerate=(fraction)30/1
     2022-09-26T17:35:16 - unknown:0: info Error while retrieving property names: Error: Device not opened

     ** (tcam-capture:2574): CRITICAL **: 17:35:16.358: tcam_property_enumeration_get_value: assertion 'self != NULL' failed

I was able to launch the pipeline directly though:

     root@odroid:/home/odroid/Camera/tiscamera# gst-launch-1.0 tcambin ! videoconvert ! ximagesink
     Setting pipeline to PAUSED ...
     Pipeline is live and does not need PREROLL ...
     INFO:
     Working with src caps: video/x-raw, framerate=(fraction)60/1, width=(int)744, height=(int)480, binning=(string)1x1, skipping= 
     (string)1x1, format=(string)GRAY8
     Setting pipeline to PLAYING ...
     New clock: GstSystemClock
     ERROR: from element /GstPipeline:pipeline0/GstXImageSink:ximagesink0: Output window was closed
     Additional debug info:
     ximagesink.c(697): gst_x_image_sink_handle_xevents (): /GstPipeline:pipeline0/GstXImageSink:ximagesink0
     Execution ended after 0:01:35.259153108
     Setting pipeline to PAUSED ...
     Setting pipeline to READY ...
     Setting pipeline to NULL ...
     Freeing pipeline ...

How do you take a picture in the GUI that pops up? And any ideas on why I'm getting the device not opened error with tcam-capture?

TIS-Stefan, 2022-09-27:
It seems, you got a disconnect during tcam-capture enumerated the camera properties. Does this always happen, when you start tcam-capture?

To your second question: With gst-launch-1.0 you can not save an image in a nice way. But there are samples at https://github.com/TheImagingSource/Linux-tiscamera-Programming-Samples, that show, how to do that.

Stefan

TIS-Stefan, 2022-09-27:
Please open a terminal and enter
```
export GST_DEBUG=tcammainsrc:5,tcam-libtcam:5
tcam-capture
```
Then show me the output. 

Thank you in advance!
Stefan

jdog816, 2022-09-27:
Yes, that always happens. When I enter tcam-capture in the terminal, two windows pop up - one for me to chose the camera, and the other has tabs (Quit, Open Device, Properties, Format, Info, Options). I select the camera in the first window (it's the only option), and then I get the following messages in the terminal:

     root@odroid:/home/odroid/Camera/tiscamera# export GST_DEBUG=tcammainsrc:5,tcam-libtcam:5
     root@odroid:/home/odroid/Camera/tiscamera# tcam-capture
     2022-09-27T17:45:17 - unknown:0: warning QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
     0:00:00.649677498 18180   0x5adb40 INFO            tcam-libtcam libtcam_base.cpp:69:print_setup_to_logger: 
     The following library versions are used:
	Tcam:	1.0.0_master/2225d34d_rev_4009
	Aravis:	0.8_version_2225d34d
	Modules:	gst_aravis_arvu3v_v4l2_libusb_tools_doc
     0:00:00.667616089 18180 0xb16044c0 INFO            tcam-libtcam aravis_utils.cpp:187:fetch_gige_daemon_device_list: 
     Failed to create shmkey. Not using gige-daemon to enumerate devices.
     2022-09-27T17:45:20 - unknown:0: info New device: 46410017-v4l2
     2022-09-27T17:45:23 - unknown:0: info device selected: DFM 22BUC03-MLA-46410017-v4l2

     2022-09-27T17:45:23 - unknown:0: debug Setting 'conversion-element' property to 'auto'
     0:00:05.726256096 18180   0x5adb40 DEBUG            tcammainsrc mainsrc_device_state.cpp:207:open_camera: 
     <tcammainsrc0> Trying to open device with serial='46410017 and type='v4l2'.
     0:00:06.488486279 18180   0x5adb40 DEBUG           tcam-libtcam V4l2Device.cpp:991:index_formats: Found format: 
     Mono8
     0:00:06.488554280 18180   0x5adb40 INFO            tcam-libtcam V4l2Device.cpp:261:get_framerate: Current framerate is 60 / 1 fps
     0:00:06.488814160 18180   0x5adb40 DEBUG           tcam-libtcam V4l2Device.cpp:178:get_available_video_formats: 
     Returning 1 formats.
     0:00:06.489049665 18180   0x5adb40 INFO            tcam-libtcam 
     SoftwarePropertiesExposureAuto.cpp:38:generate_exposure_auto: Adding software ExposureAuto.
     0:00:06.489100374 18180   0x5adb40 INFO            tcam-libtcam 
     SoftwarePropertiesExposureAuto.cpp:104:generate_gain_auto: Adding software GainAuto.
     0:00:06.489150750 18180   0x5adb40 INFO            tcam-libtcam 
     SoftwarePropertiesExposureAuto.cpp:208:generate_auto_functions_roi: Adding software AutoFunctionsROI
     0:00:06.489260835 18180   0x5adb40 DEBUG           tcam-libtcam V4l2Device.cpp:178:get_available_video_formats: 
     Returning 1 formats.
     0:00:06.489327212 18180   0x5adb40 DEBUG           tcam-libtcam V4l2Device.cpp:178:get_available_video_formats: 
     Returning 1 formats.
     0:00:06.489379504 18180   0x5adb40 DEBUG           tcam-libtcam V4l2Device.cpp:178:get_available_video_formats: 
     Returning 1 formats.
     0:00:06.489491757 18180   0x5adb40 DEBUG            tcammainsrc mainsrc_device_state.cpp:230:open_camera: 
    <tcammainsrc0> Device provides the following caps: video/x-raw, format=(string)GRAY8, width=(int)320, height=(int)240, 
     framerate=(fraction){ 15/1, 25/1, 30/1, 60/1 }; video/x-raw, format=(string)GRAY8, width=(int)640, height=(int)480, 
     framerate=(fraction){ 15/1, 25/1, 30/1, 60/1 }; video/x-raw, format=(string)GRAY8, width=(int)744, height=(int)480, 
     framerate=(fraction){ 15/1, 25/1, 30/1, 60/1 }
     2022-09-27T17:45:24 - unknown:0: info settings caps to : video/x-raw, format=(string)GRAY8, width=(int)640, height=(int)480, framerate=(fraction)30/1
     0:00:06.531447389 18180   0x5adb40 DEBUG           tcam-libtcam V4l2Device.cpp:395:stop_stream: Stopping stream
     0:00:06.531491681 18180   0x5adb40 DEBUG           tcam-libtcam V4l2Device.cpp:418:stop_stream: Stopped stream
     0:00:06.531542516 18180   0x5adb40 DEBUG           tcam-libtcam V4l2Device.cpp:395:stop_stream: Stopping stream
     0:00:06.531576100 18180   0x5adb40 DEBUG           tcam-libtcam V4l2Device.cpp:418:stop_stream: Stopped stream
     2022-09-27T17:45:25 - unknown:0: info Error while retrieving property names: Error: Device not opened

     ** (tcam-capture:18180): CRITICAL **: 17:45:25.262: tcam_property_enumeration_get_value: assertion 'self != NULL' failed

TIS-Stefan, 2022-09-28:
See in issue https://github.com/TheImagingSource/tiscamera/issues/483

jdog816, 2022-09-28:
When I run 
     gst-launch-1.0 tcambin ! videoconvert ! ximagesink
I get the same grey video but no apparent way to capture/save an image from it.

When i run
     gst-launch-1.0 tcambin ! videoconvert ! xvimagesink
after adding that line to that file, I get:

     Setting pipeline to PAUSED ...
     ERROR: Pipeline doesn't want to pause.
     ERROR: from element /GstPipeline:pipeline0/GstXvImageSink:xvimagesink0: Could not initialise Xv output
     Additional debug info:
     xvimagesink.c(1773): gst_xv_image_sink_open (): /GstPipeline:pipeline0/GstXvImageSink:xvimagesink0:
     No Xv Port available
     Setting pipeline to NULL ...
     Freeing pipeline ...

jdog816, 2022-09-28:
I probably should've closed this issue earlier because I fixed the original problem (installing tiscamera) by switching to Ubuntu 18.04. I think I'll open a separate issue because I'm trying to run a custom script that was originally written for gstreamer/tiscamera 0.10 but I'm trying to update it to run on the latest. It's able to capture and save images like before, except it's still throwing some errors which I think are related to plugins that are no longer supported (tisvideobufferfilter, tiscolorize, tis_auto_exposure, bayer2rgb).

hanlray, 2023-03-29:
> tcam_property_enumeration_get_value

Have you fixed this 'tcam_property_enumeration_get_value' error? I have the exact problem.

TIS-Stefan, 2023-03-29:
Unfortunately, I lost the context of that error. Also I do not get this error on my Ubuntus. 
Do you still see that error?
If yes, would you please post the output of 
tcam-ctrl --version
? Thank you in advance!

Stefan

TIS-Edgar, 2024-04-12:
Closed due to inactivity.

closed 2024-04-12 by TIS-Edgar