TheImagingSource/tiscamera #483: UBUNTU20.04 72BUC02 Python set-properties

Max313233, 2022-09-23:
Hi,
I just started using TIS cameras in Linux
UBUNTU20.04 
72BUC02 
Python 

getting a picture works fine using "VideoCapture(0)" in opencv

but I can not change properties and make them last, using the example 02-set-properties.py
the only value I can change and it keeps the changes is ExposureTimer

code:
    camera.set_tcam_enumeration("ExposureAuto", "Off")
    camera.set_tcam_enumeration("GainAuto", "Off")
    camera.set_tcam_integer("Contrast", 1)
    camera.set_tcam_integer("Sharpness", 2)
    camera.set_tcam_float("Gamma", 100)
    camera.set_tcam_float("Gain", 5)
    camera.set_tcam_float("ExposureTime", 10000)

why does it lose most of its settings each time the script stops ?

Thx 4 Help
Max

[image]

TIS-Stefan, 2022-09-23:
Hello

why does it lose most of its settings each time the script stops ?

Because they are software properties. The camera is not that smart, is has only exposure and gain. 

But:
you can setup your camera with tcam-capture and save the properties as json file in the "info" function.  The saved file can be loaded with a Python script like:

```
# Copyright 2019 The Imaging Source Europe GmbH
#
# Licensed under the Apache License, Version 2.0 (the "License");
# you may not use this file except in compliance with the License.
# You may obtain a copy of the License at
#
# http://www.apache.org/licenses/LICENSE-2.0
#
# Unless required by applicable law or agreed to in writing, software
# distributed under the License is distributed on an "AS IS" BASIS,
# WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
# See the License for the specific language governing permissions and
# limitations under the License.

import time
import sys
import gi

gi.require_version("Gst", "1.0")

from gi.repository import Gst

def main():

    Gst.init(sys.argv)  # init gstreamer

    # this line sets the gstreamer default logging level
    # it can be removed in normal applications
    # gstreamer logging can contain verry useful information
    # when debugging your application
    # see https://gstreamer.freedesktop.org/documentation/tutorials/basic/debugging-tools.html
    # for further details
    # Gst.debug_set_default_threshold(Gst.DebugLevel.WARNING)

    state=""

    with open("state2.json","r") as f:
        state = f.read()

    print(state)

    serial = "45219982"

    pipeline = Gst.parse_launch("tcambin name=bin "
                                " ! videoconvert"
                                " ! ximagesink sync=false")

    # retrieve the bin element from the pipeline
    camera = pipeline.get_by_name("bin")

    # serial is defined, thus make the source open that device
    if serial is not None:
        camera.set_property("serial", serial)

    camera.set_property("tcam-properties-json",state)

    pipeline.set_state(Gst.State.PLAYING)

    print("Press Ctrl-C to stop.")

    # We wait with this thread until a
    # KeyboardInterrupt in the form of a Ctrl-C
    # arrives. This will cause the pipline
    # to be set to state NULL
    try:
        while True:
            time.sleep(1)
    except KeyboardInterrupt:
        pass
    finally:
        pipeline.set_state(Gst.State.NULL)

if __name__ == "__main__":
    main()
```
That means, only loading the property setup from file. You do not need to handle them directly in your code any more.

Stefan

Max313233, 2022-09-23:
Hi thx a lot, I will try it.
Can the json file als be created with ic-capture in win ?

TIS-Stefan, 2022-09-23:
> Can the json file als be created with ic-capture in win ?

I am very sorry, but no. IC Capture uses more or less own names and types for properties, while tiscamera 1.0 and tcam-capture try to implement the Genicam property names and types. 

Stefan

Max313233, 2022-09-27:
ok thx a lot. I'm trying right now

Max313233, 2022-09-28:
Hi,

for some reasons my tcam-capture is not working any more. just reinstalled it
Ubuntu 20.04
72buc02

the cam can be chosen in startupmenu but then shows 0fps

but when I grab an image with opencv it works like a charme

I only need tcam to generate a json file of the properties. I guess I can edit the values in this file by hand, so if someone can provide me a json file of the cam 72buc02 I could go on.
Or gimmi a hind how to fix my problem with tcam

thx a lot!
[image]

Max313233, 2022-09-28:
sry, it didn't work

[image]

TIS-Stefan, 2022-09-28:
I miss the camera serial number. ik, it was in "<" and ">", which were swallowed by markow..

tcam-ctrl --save-json 12345678

where 12345678 ist to be replaced by the serial number of the camera.

Max313233, 2022-09-28:
hi it seems to work, I can turn off the automatic funktions but to set gain and exposure it seems like I need to have the exact values. so I set up another VM, installed a fresh ubuntu and tiscamera by following these instructions
https://www.theimagingsource.com/documentation/tiscamera/tutorial.html#installation

I also installed the 2 packages shown in the pic
but still the same. 
TCAM finds the cam, let me chose it but gives no pic and says 0fps
could the camera be misconfigured in some way?
ICcapture under win works just fine
opencv also gives me a pic in linux

[image]
[image]

TIS-Stefan, 2022-09-28:
You are running in a VM?
Which computer do you use?

Maybe it is caused by the xvimagesink.

Please try
gst-launch-1.0 tcambin ! videoconvert ! ximagesink
That should work
gst-launch-1.0 tcambin ! videoconvert ! xvimagesink
If this does not work, you edit the file "~/.config/the_imaging_source/tcam-capture.conf" and add the line
"video-sink-element=ximagesink"

I must admit, I have no idea, what happens on your system. I am sorry.

Stefan

Max313233, 2022-09-28:
gst-launch-1.0 tcambin ! videoconvert ! xvimagesink
this works !

TIS-Stefan, 2022-09-29:
Hello

I had a customer, who tested tiscamera 1.1 and it worked, withou the "Device not opened" situation. The customer had the same error before.
I suggest, you do that too:
```
git clone https://github.com/TheImagingSource/tiscamera.git
cd tiscamera
git checkout development
./scripts/dependency-manager install
mkdir build
cd build
make -j8
sudo make install
tcam-capture
```

"make -j8" tells make to use 8 threads for compiling. That is faster, than running "make" only. 

Please let me know your results!

Stefan

Max313233, 2022-09-30:
good morning,

here it failed
[image]

TIS-Stefan, 2022-09-30:
Oh
I forgot the cmake ..
as documented at https://github.com/TheImagingSource/tiscamera, scroll down.

Max313233, 2022-09-30:
ok now it works !!
but it can handle only 640x480.
This is not a problem, I could make a json file and edit it to my desire. 
python scrips do I need. opencv works like a charme

thx u very much
[image]

TIS-Stefan, 2022-09-30:
The json file does not contain the video format. You pass the format, resolution and frame rate to the pipeline (as shown in many other samples and issues here)
```
tcambin ! video/x-raw, format=BGRx, width=2592,height=1944,framerate=5/1 ! videoconvert ! ximagesink.
```
You may use tcam-capture to check the available resolutions and framerate combinatios.

Max313233, 2022-09-30:
ok thanks.
here is another question:
I have 2 cameras in the system I'm building right now. I grab a picture from them in opencv by the command cv2.VideoCapture(0) or cv2.VideoCapture(1)
can I somehow force the cameras depending on there serial nr to be nr 0 or 1 ? not shure if they alsways keep there order

TIS-Stefan, 2022-09-30:
I suggest to use OpenCV with GSstreamer support. Then you can use the camera's serial number in the pipeline, as well as specify the pixel format, resolution and frame rates.

TIS-Stefan, 2022-09-30:
It would be something like this:

```C++
const char *pipeline1 = "tcambin serial=33910094 ! video/x-raw, format=BGRx, width=290,height=1944, framerate=3/1 !  appsink";
const char *pipeline2 = "tcambin serial=33910095 ! video/x-raw, format=BGRx, width=290,height=1944, framerate=3/1 !  appsink";

cv::VideoCapture *cap1 = new cv::VideoCapture(pipeline1, cv::CAP_GSTREAMER);
cv::VideoCapture *cap2 = new cv::VideoCapture(pipeline2, cv::CAP_GSTREAMER);
```
Using the serial numbers, you can always be sure, that each videocapture object is using the correct camera.
If both cameras are connected to the same USB controller, you must use the half of the maximum allowed frame rate. Otherwise , there is not enough bandwidth on the USB bus.

Max313233, 2022-10-04:
Hi,
I'm testing the pipeline tip, but it gives me an error for syntax
sadly it doesn't tell where exact the problem is
I  tried:
const char *pipeline1 = "tcambin serial=39120848! video/x-raw, format=GRAY8, width=2592, height=1944, framerate=4/1 !  appsink";

is there any example with TIS and pipeline available ?

thx 4 ur constant help !!

TIS-Kevin, 2022-10-04:
Hello Max,
just mant to make sure which programming language you use. My Colleague send you the code for C programming but at beginning you noted Python.

That is c code:
> 
> I tried:
> const char *pipeline1 = "tcambin serial=39120848 ! video/x-raw, format=GRAY8, width=2592, height=1944, framerate=4/1 ! appsink";

Please use the necessary code from our python example:
The Example for Python would be here:
https://github.com/TheImagingSource/tiscamera/blob/master/examples/python/07-appsink.py

Example where my Colleague extracted from ( C ):
https://github.com/TheImagingSource/tiscamera/blob/master/examples/c/07-appsink.c

can you run 
`gst-launch-1.0 tcambin serial=39120848 ! video/x-raw, format=GRAY8, width=2592, height=1944, framerate=4/1 ! appsink` 
in your Terminal and tell me if that works?

Max313233, 2022-10-05:
Hi,

this is the answer from the terminal:

gst-launch-1.0 tcambin serial=39120848 ! video/x-raw, format=GRAY8, width=2592, height=1944, framerate=4/1 ! appsink
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
INFO:
Working with src caps: video/x-raw, format=(string)GRAY8, width=(int)2592, height=(int)1944, framerate=(fraction)4/1
Setting pipeline to PLAYING ...
New clock: GstSystemClock

then nothing happens

I also used the examplecode 07appsink but couldn't perform cv2.VideoCapture.

Here is my code, it says: TypeError: an integer is required (got type Pipeline) 

THX 4 ur help again

> import sys
> import gi
> import cv2
> 
> 
> gi.require_version("Gst", "1.0")
> gi.require_version("GstVideo", "1.0")
> 
> from gi.repository import Gst, GstVideo
> 
> framecount = 0
> 
> 
> def callback(appsink, user_data):
>     """
>     This function will be called in a separate thread when our appsink
>     says there is data for us. user_data has to be defined
>     when calling g_signal_connect. It can be used to pass objects etc.
>     from your other function to the callback.
>     """
>     sample = appsink.emit("pull-sample")
>             
>     if sample:
> 
>         caps = sample.get_caps()
> 
>         gst_buffer = sample.get_buffer()
> 
>         try:
>             (ret, buffer_map) = gst_buffer.map(Gst.MapFlags.READ)
> 
>             video_info = GstVideo.VideoInfo()
>             video_info.from_caps(caps)
> 
>             stride = video_info.finfo.bits / 8
> 
>             pixel_offset = int(video_info.width / 2 * stride +
>                                video_info.width * video_info.height / 2 * stride)
> 
> 
>             pixel_data = buffer_map.data[pixel_offset]
>             timestamp = gst_buffer.pts
> 
>             global framecount
> 
>             output_str = "Captured frame {}, Pixel Value={} Timestamp={}".format(framecount,
>                                                                                  pixel_data,
>                                                                                  timestamp)
> 
>             print(output_str, end="\r")  # print with \r to rewrite line
> 
>             framecount += 1
> 
>             
>         finally:
>             gst_buffer.unmap(buffer_map)
> 
>     return Gst.FlowReturn.OK
> 
> 
> def main():
> 
>     Gst.init(sys.argv)  # init gstreamer
> 
> 
>     Gst.debug_set_default_threshold(Gst.DebugLevel.WARNING)
> 
>     serial = 39120848
> 
>     pipeline = Gst.parse_launch("tcambin serial=39120848 ! video/x-raw, format=GRAY8, width=2592, height=1944, framerate=4/1"
>                                 " ! videoconvert"
>                                 " ! appsink name=sink")
> 
>     # test for error
>     if not pipeline:
>         print("Could not create pipeline.")
>         sys.exit(1)
> 
>     # The user has not given a serial, so we prompt for one
>     if serial is not None:
>         source = pipeline.get_by_name("source")
> #        source.set_property("serial", serial)
> 
>     sink = pipeline.get_by_name("sink")
> 
>     # tell appsink to notify us when it receives an image
>     sink.set_property("emit-signals", True)
> 
>     user_data = "This is our user data"
> 
>     # tell appsink what function to call when it notifies us
>     sink.connect("new-sample", callback, user_data)
> 
>     pipeline.set_state(Gst.State.PLAYING)
> 
>     print("Press Ctrl-C to stop.")
> 
> 
>     vid = cv2.VideoCapture(pipeline)
>     ret, img_gray = vid.read()
>     cv2.imshow("show_me", img_gray)
>     cv2.waitKey(0)
>             
> 
>     pipeline.set_state(Gst.State.NULL)
> 
> 
> if __name__ == "__main__":
>     main()

TIS-Kevin, 2022-10-06:
Dear Max,
i have to admit that i tried that on my own and could not find a solution. Currently our Linux Developer is in holidays and comes back at Monday. Probably he knows what to do.

Maybe a program like this one is what you are looking for?
https://github.com/TheImagingSource/Linux-tiscamera-Programming-Samples/blob/master/python/Snap%20an%20Image%20and%20convert%20to%20OpenCV/Program.py

Kevin

Max313233, 2022-10-17:
Hi,
we could somehow mange to use the pipeline with the following code.
the problem is, that the settings applyed with the json files will get lost.
We tried to merge the loading of the json and grabing a pic from a Cam by SN but it takes a lot of time and doesn't work at all.

hoping you can help us once more :)
THX a lot !

Load JSON:
{
import sys
import gi

gi.require_version("Gst", "1.0")

from gi.repository import Gst

def load(serial,file):

    Gst.init(sys.argv)

    state=""

    with open(file,"r") as f:
        state = f.read()

    print(state)

    #serial = "39120848"

    pipeline = Gst.parse_launch("tcambin name=bin "
                                " ! videoconvert"
                                " ! ximagesink sync=false")

    # retrieve the bin element from the pipeline
    camera = pipeline.get_by_name("bin")

    # serial is defined, thus make the source open that device
    if serial is not None:
        camera.set_property("serial", serial)
        
        

    camera.set_property("tcam-properties-json",state)

    pipeline.set_state(Gst.State.PLAYING)

    pipeline.set_state(Gst.State.NULL)
    
    return 0

}

grab pic and do things with it

{
import cv2

import numpy as np

import TIS

Cam_X = 2592

Cam_Y = 1944

def template_matching(sizeOUT,gamma,Serial,ImgPath):

    #39120848

[... 1040 lines cut ...]

        filename = ImgPath + '/Cam.png'

        cv2.imwrite(filename, sized)

        

        cv2.destroyAllWindows()

        

    #    cv2.imshow("tada", sized)

    #    cv2.waitKey(0)

    #    cv2.destroyAllWindows()

    

       

        arrayOUT = np.array([[XnozzleL,YnozzleL,max_val_N_L],[XpinL,YpinL,max_val_P_L],[XnozzleR,YnozzleR,max_val_N_R],[XpinR,YpinR,max_val_P_R]])

        return arrayOUT
}

TIS-Stefan, 2022-10-18:
Hello

I think, the issue is in the line
`
pipeline.set_state(Gst.State.NULL)
`
This deletes the camera and all the properties and the pipeline, so all settings you made are gone.

You must create one pipeline and keep it, as long as you use it. You can not open and configure the camera and then delete the pipeline, you used for that and create a new one. 

I hope, this gives an idea to you. If not, please come back to me. Also please add your Python code as zip file, because the one above is not nicely readable with all the empty lines...

Stefan

TIS-Stefan, 2022-10-18:
I looked somewhat deeper...
You start with:

```Python
Gst.init(sys.argv)

with open(JsonPath,"r") as f:

    state = f.read()
 
print(state)

pipeline = Gst.parse_launch("tcambin name=bin "
                            " ! videoconvert"
                            " ! ximagesink sync=false")

# retrieve the bin element from the pipeline
camera = pipeline.get_by_name("bin")

pipeline = Gst.parse_launch("tcambin name=bin "
                        " ! videoconvert"
                        " ! ximagesink sync=false")

camera.set_property("tcam-properties-json",state)
```
Then you continue with 
```Python
Tis1 = TIS.TIS()
Tis1.openDevice(Serial, Cam_X, Cam_Y, "4/1", TIS.SinkFormats.GRAY8, False)
```
That creates all new. Therefore the properties are gone. It is better to add a function in the TIS class, that reads the json string after the openDevice() call on the self.source 

Can you do that?

Stefan

TIS-Stefan, 2022-10-18:
You may enhance the TIS.py as follows:

```Python
def openDevice(self,serial, width, height, framerate, sinkformat: SinkFormats, showvideo: bool, statefile):
        ''' Inialize a device, e.g. camera.
        :param serial: Serial number of the camera to be used.
        :param width: Width of the wanted video format
        :param height: Height of the wanted video format
        :param framerate: Numerator of the frame rate. /1 is added automatically
        :param color: True = 8 bit color, False = 8 bit mono. ToDo: Y16
        :param statefile: json file, that contains the paramters to be set in the camera
        :return: none
        '''

        self.serialnumber = serial
        self.height = height
        self.width = width
        self.framerate = framerate
        self.sinkformat = sinkformat
        self.livedisplay = showvideo
        self._createPipeline()
        self.source.set_property("serial", self.serialnumber)
        self.pipeline.set_state(Gst.State.READY)
        self.pipeline.get_state(40000000)

        self.loadstatefile(statefile)

    def loadstatefile(self, filename):
        state = ""
        with open(JsonPath,"r") as f:
            state = f.read()
        self.source.set_property("tcam-properties-json",state)
```

You call the TIS openDevice() with:
``` Python
Tis1 = TIS.TIS()
Tis1.openDevice(Serial, Cam_X, Cam_Y, "4/1", TIS.SinkFormats.GRAY8, False, "state.json")
```

Later you can even call loadstatefile() for loading a different property set:
``` Python
Tis1.loadstatefile("state2.json")
```
This works, as long as the pipeline in Tis1 is not NULL.

Stefan

TIS-Edgar, 2024-04-12:
Closed due to inactivity.

closed 2024-04-12 by TIS-Edgar