TheImagingSource/tiscamera #484: No camera detected on raspberry pi 4 bullseye

gp4ges, 2022-09-28:
I want to have bullseye on my raspberry.
I sucessfully compiled and installed tiscamera.
However, the camera is never detected. I turned on legacy camera support in the raspi-config, I also tried to copy the /boot/config.txt from the provided image on the imaging source website.
This is not a hardware problem as the camera works fine with the buster image provided on the imaging source website.
I tried many different methods to check if the camera is detected including:
`vcgencmd get_camera`
`i2cdetect -y 1`
`libcamera-hello`
`v4l2-ctl --list-formats`
`cam -l`
None of these method showed a detected camera

Did anyone manage to make Tiscamera work on Bullseye?

TIS-Stefan, 2022-09-28:
Please let me know, which camera you use.

Stefan

gp4ges, 2022-09-28:
The camera is this one:
[https://www.theimagingsource.com/embedded-vision/mipi-csi-2-cameras/mipi-csi-2-board-cameras/dfm36mx290ml/](url)

TIS-Stefan, 2022-09-28:
Did you install the https://www.theimagingsource.com/support/downloads-for-linux/install/pimipisrc/ file?

gp4ges, 2022-09-28:
I will try it, and let you know, thanks.

gp4ges, 2022-09-28:
I installed this and tcamprop, then I recompiled tiscamera. 
I am still not able to detect the camera.

gp4ges, 2022-09-28:
It is still not working, I'm wondering if it could be a /boot/config.txt issue. My current file looks like this:

```
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Automatically load overlays for detected cameras
start_x=1

# Automatically load overlays for detected DSI displays
#display_auto_detect=1

# Enable DRM VC4 V3D driver
#dtoverlay=vc4-kms-v3d
max_framebuffers=2

# Disable compensation for displays with overscan
disable_overscan=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[pi4]
dtoverlay=vc4-fkms-v3d
max_framebuffers=2
# Run as fast as firmware / board allows
# arm_boost=1

[all]
start_x=1
#dtoverlay=imx477,media-controller=0
gpu_mem=128
#dtoverlay=vc4-fkms-v3d
start_x=1
dtparam=i2c_vc=on
```

TIS-Stefan, 2022-09-28:
Hello

> vcgencmd get_camera
> i2cdetect -y 1
> libcamera-hello
> v4l2-ctl --list-formats
> cam -l

These commands do not work at all on PI. I learned, version 1.2 does not work on Bullseye. There is an alpha version:
[tcampimipisrc_1.3.0.656_armhf.zip](https://github.com/TheImagingSource/tiscamera/files/9665637/tcampimipisrc_1.3.0.656_armhf.zip)
That is a GStreamer driver, so only GStreamer works.

gp4ges, 2022-09-28:
Thanks a lot, it worked!
I just had to install tcamproperty from https://github.com/TheImagingSource/beta-packages/releases

palacidr, 2023-03-27:
I am having the similar issue as OP.

I have the following hardware:

RPi 4 Model B

Camera DMM 36AX290-ML

Attached is image of my connection to the RPI4. I have confirmed the cable works by verifying it with a RPi camera module.
[image]

I have installed the latest Demo image from Imaging Source from https://www.theimagingsource.com/en-us/support/download/imagetispimipi-1.0.0/

The camera does not show up as recognized by the TCam-Capture software
[image]

I download and attempt to install the latest versions of tcamdutils and tcampimipisrc but there are dependency issues.
[image]

Any help with this would be greatly appreciated. Ideally, I should be able to get the software installed on a RPi 3 running the latest version of Raspbian OS (Bullseye 11) with the same camera DMM 36AX290-ML, but I should at least get it to work on the RPi4 first. I see that the last link in this thread leads to beta-packages that are no longer available.

TIS-Stefan, 2023-03-28:
Hello

I am very sorry, but PI 3 is not supported. At least, I never tried that. 

For PI4: Please use a standard Raspbian, install tiscamera and the PIMIPIsrc from https://www.theimagingsource.com/en-de/support/download/ under "GStreamer Modules" and install them. (Maybe it works for PI3 too)

Stefan

palacidr, 2023-03-28:
> Hello
>
> I am very sorry, but PI 3 is not supported. At least, I never tried that.
>
> For PI4: Please use a standard Raspbian, install tiscamera and the PIMIPIsrc from [...]

It worked! I used RPi4, Raspbian GNU/Linux 11 (bullseye). Downloaded and installed
tiscamera_1.0.0.4005_armhf_pi4_rasbian_10.deb
and
tcampimipisrc_1.3.0.662_armhf.deb

FYI - I tried the same on a RPi3. The camera is recognized but no image is displayed. Will TIS look into this? Should I open a new ticket for this support or is RPi3 just to old?

TIS-Stefan, 2023-03-29:
Hello

I am sorry to disappoint you, but we wont look into RPI 3 anymore. 

Stefan

palacidr, 2023-04-03:
Upon connecting the RPi 7inch Touchscreen. The camera is no longer detected by TCam-Capture I ran python 00-list-devices.py and get the following warnings and errors:
[image]

Any idea what conflict is happening here? Is it an issue with the I2C bus? Is it an overlay issue?

I am using the RPi4 and the latest Raspbian Bullseye OS and the latest tiscamera and PIMIPIsrc as indicated earlier.

TIS-Stefan, 2023-04-04:
I never saw this before. What happens, if the display is disconnected? Is the camera visible again?
Can you also please give the model name of the 7" display?

Stefan

TIS-Stefan, 2023-04-04:
Hello

Can you please connect the display and run
i2cdetect -y 10
and show me the output?

palacidr, 2023-04-04:
Hello Stefan,

Please see the images below. Note I obtain the screenshots through VNC
Viewer. When I connect the touchscreen and reboot the resolution changes to
match the touchscreen resolution of 800 x 600.

Display ribbon cable removed, camera connected
[image: image.png]
Display connected, camera connected, followed by a system reboot:
[image: image.png]

Display connected, camera ribbon cable removed, system reboot:
[image: image.png]
Display connected, camera connected, no reboot:
[image: image.png]
Display connected, camera connected, system reboot:
[image: image.png]
Display ribbon cable removed, camera connected, no reboot:
[image: image.png]
Display ribbon cable removed, camera connected, reboot
[image: image.png]
I do not know what the other addresses that appeared (1a, 36, and 50) are.
I did not connect any other components.
Display connected, camera connected, no reboot (the addresses are seen as
if the touchscreen is connected but the touchscreen does not work and the
camera is still detected:
[image: image.png]
Display connected, camera connected, reboot:
[image: image.png]

The touchscreen display works (display and touch ability) after the display
is connected and a reboot is done on the system. Otherwise, it does not
display or touch.

Thank you for looking into this!

TIS-Kevin, 2023-04-05:
Dear palacidr,

in your last Message are no images attached. I can just see the Text "[image: image.png]"
Would you please try again to send your Message with the Images?

Best Regards

Kevin

TIS-Kevin, 2023-04-05:
Dear palacidr,

i am sorry to say that but i can not see any PDF Document attached. Did you forget to attach it?
If not, is there any other way you could share the images to us?

Best Regards

Kevin

palacidr, 2023-04-05:
Hello Kevin,

I attached the pdf. I attach it again and provide a link to a google drive
folder for you to access.

https://drive.google.com/file/d/1AoAea0-X1Sky9-ZLrvXhkztKYPThiFNP/view?usp=sharing

Thank you,

Diego Palacios

TIS-Kevin, 2023-04-06:
Dear palacidr,

thank you for the PDF. The link worked.

I would like you to do following:

`export GST_DEBUG=tcampimipisrc:5`

This will enable more Debug Output for the Terminal Session. (So in the Terminal where you used the command and only for the time where the Session is active since you used the command)

We would like to know what the Output is if all is connected to the Pi and you try to find the camera with `tcam-ctrl -l`

Best Regards

Kevin

palacidr, 2023-04-06:
Hello Kevin,

I ran the commands you requested. Please see the link below:

https://drive.google.com/file/d/1UEoO7QYtIw3KgpooClp_8-GCy9al1m_n/view?usp=share_link

Thank you

palacidr, 2023-04-10:
Hello Kevin,

I am following up on this issue.

Thank you,

Diego Palacios

TIS-Stefan, 2023-04-11:
Hello Diego

It is not really clear, what happens. 

It looks like the connected touch screen blocks the access to the clock chip of the camera. But the i²c addresses seem to be different, no conflicts visible. 

Could there be a power issue? The PI may shows the power warning icon. 

I am clueless. 

Stefan

LusenieTuray, 2024-03-11:
@TIS-Stefan Hi can you please help me?  I am using the Arducam Day-Night Vision  Camera [](https://a.co/d/7IzUsAD) on the Rasberry Pi 3 model B and i am running in the same problem that camera is not available . Please help me 
[image]
[image]
[image]

TIS-Stefan, 2024-03-12:
I am sorry, but this is not our camera, thus I can not be of help. Please ask the camera manufacturer for help.

closed 2024-03-12 by TIS-Stefan