TheImagingSource/tiscamera #485: No element tisvideobufferfilter, tiscolorize, tis_auto_exposure, bayer2rgb

jdog816, 2022-09-28:
I'm trying to run a custom script based on tiscamera, using a DFM 22BUC03-ML camera in Ubuntu 18.  The script is saving images to a folder, but they are coming out grey and I'm getting these errors:

     Running in verbose mode
     0:00:00.029717975 31576    0xbdb20 ERROR           GST_PIPELINE ./grammar.y:661:priv_gst_parse_yyparse: no element 
     "tisvideobufferfilter"
     0:00:00.032245564 31576    0xbdb20 ERROR           GST_PIPELINE ./grammar.y:661:priv_gst_parse_yyparse: no element 
     "tiscolorize"
     0:00:00.032288690 31576    0xbdb20 ERROR           GST_PIPELINE ./grammar.y:661:priv_gst_parse_yyparse: no element 
     "tis_auto_exposure"
     0:00:00.032331274 31576    0xbdb20 ERROR           GST_PIPELINE ./grammar.y:661:priv_gst_parse_yyparse: no element 
     "bayer2rgb"
     camera base time=45:54:12.641184001
     GLib-GObject-CRITICAL **: 17:25:02.821: g_object_get: assertion 'G_IS_OBJECT (object)' failed

I believe this is the relevant portion of the script:

     //Build the pipeline
     const gchar* Pipeline = "v4l2src name=camera "
                          "! video/x-raw-gray,width=640,height=480,framerate=15/1 "
                          "! tisvideobufferfilter name=videofilter "
                          "! videocrop top=0 left=80 right=80 bottom=0 "
                          "! tiscolorize "
                          "! tis_auto_exposure name=exposure " //region-x0=80 region-y0=80 region-x1=400 region-y1=400 "
                          "! bayer2rgb "
                          "! ffmpegcolorspace "
                          "! jpegenc name=jpegencoder quality=" INITIALJPEGQUALITY " "
                          "! appsink name=jpegframe ";

Would it just be a matter of commenting out some of those lines, or editing them with the 1.0 plugin names?

jdog816, 2022-09-28:
For example, were "tisvideobufferfilter" and "jpegenc" changed to something new when migrating from 0.1 to 1.0 ?

jdog816, 2022-09-28:
Sorry for multiple questions, but I'm getting:

     GStreamer-CRITICAL **: 21:00:37.623: gst_buffer_get_sizes_range: assertion 'GST_IS_BUFFER (buffer)' failed

     GStreamer-CRITICAL **: 21:00:37.790: gst_buffer_map_range: assertion 'GST_IS_BUFFER (buffer)' failed

Any ideas?

TIS-Stefan, 2022-09-29:
In tiscamera 1.0 these elements are no longer available. 
First of all I suggest to use GStreamer 1.0. I can not provide support for Gstreamer 0.10 anymore
Also I suggest to use the current tiscamera 1.0.
Your pipeline is that:
```
const gchar* Pipeline = "tcambin name=camera "
                      "! video/x-raw,format=BGRxwidth=640,height=480,framerate=15/1 "
                      "! videocrop top=0 left=80 right=80 bottom=0 "
                      "! videoconvert "
                      "! jpegenc name=jpegencoder quality="INITIALJPEGQUALITY " "
                      "! appsink name=jpegframe ";
```
We only need to configure the auto functions ROI by tcambin properties. As soon as I have access to camera and Linux computer, I will have a look.

jdog816, 2022-10-24:
Solved some of the problems by following this document, but getting an error that's not explained here:
https://github.com/Xilinx/gstreamer/blob/master/docs/random/porting-to-1.0.txt
I opened a separate issue for it, so I'll close this one.

closed 2022-10-24 by jdog816