TheImagingSource/tiscamera #486: Save/Load camera configuration into camera's memory slot

dinhtuanan, 2022-10-04:
I read the info of the camera https://s1-dl.theimagingsource.com/api/2.5/packages/documentation/manuals-trm/trmdfk33gx265/b1db2c66-e201-5866-8dda-163c2bea225d/trmdfk33gx265.en_US.pdf
At the page 41 I saw memory slot, User Set Selector, Load User Set,... I read the document and github but I couldn't find solution

My problem is when I disconnect the camera and then reconnect it, all the properties I set like Gain, Exposure, BlackLevel,... are reset.

I would like to know if i can save the camera configuration to the memory slot and restore the camera configuration from the memory slot. So, I can keep all my setting when I use the camera for another works.

If you have examples of this or instruction, could you suggest them to me?

Regards,
An

TIS-Kevin, 2022-10-04:
Dear An,
after you changed your Properties you can save your Settings into a UserSet (not in Default). So you have to set the UserSetSelector as example to UserSet1 and do a UserSetSave. After that you should also set your UserSetDefault as UserSet1. Otherwise you have to change the UserSetSelector on UserSet1 and do a UserSetLoad everytime you start the camera.

Best Regards,
Kevin Herbrich

dinhtuanan, 2022-10-04:
Hi Kevin, thank you so much for replying
Do you have any example samples on how to do them in these 2 directions:
- Code with python, c++, c,...
- Use tcam capture application

Best Regards,
Kevin Herbrich

TIS-Kevin, 2022-10-04:
Dear An,
i have to say i am sorry. I did not notice this ticket is in tiscamera. No you do not have any access to userSets in tiscamera. 

But tcam capture (1.0) provides a function to save a device state and load the device state.
You can find this in the "Info" Menu. There is a category named "State".
If you want to use tcam-capture you have to load the config every time manually.

If you want to load the deviceState.json automatically, you write your own program which loads the .json on every start.

Code example how to do load the json can be found here:
- Python: https://github.com/TheImagingSource/tiscamera/blob/master/examples/python/11-json-state.py
- c: https://github.com/TheImagingSource/tiscamera/blob/master/examples/c/11-json-state.c

dinhtuanan, 2022-10-04:
Hi Kevin, Thank you so much for the great explanation
I use tcam-capture application to set properties, in the "Properties" menu I saw an "Update (F5)" button. I noticed that even if i press that button or not I see the properties are still saved. So I have some questions to ask you

- When I set the property (by dragging the slider). I wonder if the "State" of the camera is created or overwritten in any file in the system or in the tiscamera folder?
- Same question above but with "Update (F5)" button. When I press this button will any "State" file be created or overwritten in the system or in the tiscamera folder?
- Is there anyway I can use tcam-capture to snap image? Or is the only way to snap image is that programming?

Thank you so much
Best Regards,
An

TIS-Kevin, 2022-10-05:
Dear An,

> I noticed that even if i press that button or not I see the properties are still saved. So I have some questions to ask you

I have to be honest i do not know exactly if there is a temporary safe in the camera or in software. Our Software Developer for tcam-capture is also in holidays so i can not ask him at the moment.
In some way the properties are safed for a while but if you reconnect the camera (cable unplugged) the properties should be resetted. 

tcam-capture 1.0 does not provide a function to snap images currently. So for tiscamera 1.0 you need programming.
Examples can be found here:
https://github.com/TheImagingSource/Linux-tiscamera-Programming-Samples/tree/master/python

> Is there anyway I can use tcam-capture to snap image? Or is the only way to snap image is that programming?

If you use tiscamera 0.14 you have a snap image function for tcam-capture.

Best Regards,
Kevin

dinhtuanan, 2022-10-05:
Hi Kevin, 
Maybe after your Software Developer tcam-capture holiday I'll ask again about where "State" file is

As you said before tcam capture (1.0) provides a function to save a device state and load the device state in the "Info" Menu. 
I would like to know if tiscamera 0.14 also have this feature?

Thank you so much Kevin

Best Regards,
An

TIS-Kevin, 2022-10-05:
Hi An,

> 
> As you said before tcam capture (1.0) provides a function to save a device state and load the device state in the "Info" Menu.
> I would like to know if tiscamera 0.14 also have this feature?

No tiscamera 0.14 does not have this feature. It is a new feature which came with tiscamera 1.0. The Snap Image button should be back in tiscamera 1.1. I do not have any information about release date for tiscamera 1.1.

I think the only solution right now is to use tiscamera 1.0 and enhance the snap image program example.

Best Regards,
Kevin

TIS-Edgar, 2024-04-12:
Closed due to inactivity.

closed 2024-04-12 by TIS-Edgar