TheImagingSource/tiscamera #488: Build will target Qt6 on Arch

ShadeTechnik, 2022-10-17:
Trying to build this on Manjaro (Arch based) it always fails during make. The errors always point to Q....with a directory that goes into /usr/something/qt6. I noticed a Ubuntu dependency was Qt5, not Qt6 which on a standard Manjaro install both are present. I did manage to get it to build with cmake option -DTCAM_BUILD_TOOLS=OFF, but their is likely some issue in cmake because then it will not install the .so file that tcam-uvc-extension-loader is looking for.

With the help of a friend he he suggested I used -DQT_MAJOR_VERSION=5 as a cmake option but that did not work. What did work was editing tools/tcam-capture/CMakeLists.txt
`find_package(QT NAMES Qt6 Qt5 COMPONENTS Widgets MultiMediaWidgets Multimedia REQUIRED)
find_package(Qt${QT_VERSION_MAJOR} COMPONENTS Widgets REQUIRED)`

to

`find_package(QT NAMES Qt5 COMPONENTS Widgets MultiMediaWidgets Multimedia REQUIRED)
find_package(Qt5 COMPONENTS Widgets REQUIRED)`

I'd be more specific but I didn't keep track of the actual outputs while I was doing this. Looks like it prefers Qt6 but doesn't actually work with Qt6

jasonKercher, 2022-10-18:
Looks like issue for QT6 (for me anyway) is `QRegExp` in `[tools/tcam-capture/mainwindow.cpp](https://github.com/TheImagingSource/tiscamera/blob/dd20eed75267a6bf5e80b93fb5da572e29bc35a2/tools/tcam-capture/mainwindow.cpp)`.  I believe this is not available in QT6 in favor of `QRegularExpression`.

ShadeTechnik, 2022-10-18:
Are you expecting compatibility with Qt6? If not why not just edit the file tools/tcam-capture/CMakeLists.txt as mentioned above until it is possible? I suspect the only reason this works on Debian11/Ubuntu is it possibly does not ship with Qt6 or the distributed .deb file is the typical installation method.

I don't really see the point of allowing a possible Qt6 target if it cannot work.

TIS-Edgar, 2022-10-18:
I like to have gradual upgrades in the code so that (potentially) breaking changes can be avoided.
Qt6 compatibility will take longer than I thought.
To prevent any problems with package generation, etc that might occur from qt6 linkage i have removed it for the time being (fd966b63).

If you have any other issues, please let us know.

Thank you.

closed 2022-10-18 by TIS-Edgar