TheImagingSource/tiscamera #490: Detect existing cmake installation

tehkillerbee, 2022-10-20:
Hello,

We currently install cmake in the following way, to make sure we install the latest version as apt does not always provide the most recent version (eg. when installing on Ubuntu 18.04)
```
# download and install latest cmake
wget https://github.com/Kitware/CMake/releases/download/v3.23.1/cmake-3.23.1-linux-x86_64.sh
chmod +x cmake-3.23.1-linux-x86_64.sh
sudo ./cmake-3.23.1-linux-x86_64.sh --prefix=/usr --skip-license
```

When installing the necessary dependencies for tiscamera, cmake will be installed as well - and an old version may be installed.
```
...
The following additional packages will be installed:
  cmake-data librhash0
Suggested packages:
  cmake-doc git-daemon-run | git-daemon-sysvinit git-doc git-el git-email git-gui gitk gitweb git-cvs git-mediawiki git-svn
The following NEW packages will be installed:
  cmake cmake-data librhash0
The following packages will be upgraded:
  git
1 upgraded, 3 newly installed, 0 to remove and 87 not upgraded.
Need to get 9.923 kB of archives.
After this operation, 28,1 MB of additional disk space will be used.
```

Perhaps, it should be possible for the install script to detect if cmake has already been installed?

TIS-Edgar, 2022-11-02:
Hi,

Sorry for the late reply. I have added an argument to the dependency-manager to allow ignoring specific dependencies.

> scripts/dependency-manager install --ignore cmake libzip4

This would ignore cmake and libzip4. The names are the ones that are given to apt-get.
For now this change can be found on the development branch. The script should be copyable to existing copies.

Please let us know if this fixes this issue.

tehkillerbee, 2022-11-14:
@TIS-Edgar Sorry for the late reply. I can confirm the script behaves as expected. We currently checkout the tags when building tiscamera to ensure we always build the same version. So hopefully this fix is added to the master/latest version in the future.

closed 2023-01-06 by TIS-Edgar