TheImagingSource/tiscamera #493: Get Real Time FPS during streaming

dinhtuanan, 2022-11-07:
I use 
https://github.com/TheImagingSource/Linux-tiscamera-Programming-Samples/tree/master/python/webserver
and https://github.com/TheImagingSource/Linux-tiscamera-Programming-Samples/blob/master/python/python-common/TIS.py

to Stream into web browser

I would like to know if  I can get the current real fps while streaming like in tcam-capture app
[image]

If you have examples of this or instruction, could you suggest them to me?

Regards,
An

TIS-Stefan, 2022-11-07:
Hello

The fps is simply a frame counter, which is increased in a callback and a timer in seconds interval. You may see the appsink samples for using callback, e.g. https://github.com/TheImagingSource/tiscamera/blob/master/examples/python/07-appsink.py (which is not with an image counter. You would pass it as user data to the callback function.)
I hope, this answers your question.
Stefan

dinhtuanan, 2022-11-08:
Hi Stefan, thank you so much for replying

 I use these functions to stream into web https://github.com/TheImagingSource/Linux-tiscamera-Programming-Samples/blob/master/python/webserver/tcam-webserver.py

[image]

I see it specifies the number of frames sent in 1 second (but I want dynamic value)

I tried implement 07-appsink.py that you mention above but I cant do it

How can I count frame per second? Could you give me some more suggestions?

Thank you so much

Regards,
An

TIS-Stefan, 2022-11-08:
Hello

> I see it specifies the number of frames sent in 1 second (but I want dynamic value)

Then you may change the value passed to time.sleep() to your needs. 

> I tried implement 07-appsink.py that you mention above but I cant do it
You have no idea about passing user data?
However, in the webserver sample you may declare a public variable "ImageCount" and set it to 0. In the function "imagegenerator()" you increment "ImageCount" by one. 
Then you simply use a timer with e.g. 1 second interval. In the timer's event handler you get the value of "ImageCount" and know, how many frames you got within one second. Then set "ImageCount" to 0 again and continue your timer. 
That is the most simple way.

Keep in mind: If you want to transfer image through a network, you have limited bandwidth. Therefore, a lower frame rate makes more sense than a maximum frame rate.

Stefan

dinhtuanan, 2022-11-08:
Hi Stefan,
"timer with e.g. 1 second interval" = time.sleep(1) isn't it?

Where should I put "timer with e.g. 1 second interval" to get the value of "ImageCount" because in the function "imagegenerator()" it set **time.sleep(0.1)** mean 1s the while loop run 10 times (always 10 frames). So I cant know how many real frames I got within one second

[image]

Thank you so much Stefan

Regards,
An

TIS-Arne, 2022-11-08:
Sorry to chime in here. I had a quick look at the webserver example. 

Unfortunately, this example code is quite old, unmaintained and certainly not up to standard. It is also buggy, so I am a bit surprised it actually worked for you. If it does, it will break with a future Python version nonetheless. 

I uploaded a slightly improved version to this gist: 
https://gist.github.com/TIS-Arne/1d2cea0d51194f515a96e5eb27f0a9dc

This update does two things: It removes the "time.sleep" from the imagegenerator function. Using sleep here might have undesired side effects like adding latency. Instead I modified the tis_camera.py lightly to add a "set_max_rate" function which allows you to set a target FPS (only integer values, sorry). I also modified tcam-webserver.py to set such a rate (in line 14). Secondly I fixed a "use after free" (or rather use after unmap) issue.

As long as the client/network is able to consume all of the frames, you should reach the frame rate set with "set_max_rate". If you still need to get the actual frame rate achieved, you should add a frame counter to the "imagegenerator" function and calculate the frames served per second using this counter. 

Best regards, 

Arne Caspari

dinhtuanan, 2022-11-10:
Hi Arne, Thank you very much for your guidance

I used TIS file https://github.com/TheImagingSource/Linux-tiscamera-Programming-Samples/blob/master/python/python-common/TIS.py
 to stream into website (I combined the old webserver example with TIS file) because I want to use more feature such as set_property, get_property,...

I tried to use your new Webserver sample, I'm having trouble implementing it

I want to ask some question:

- This is how I use TIS file to stream into web
[image]

- Does the self.framerate (TIS file) attribute conflict with self.videorate (new Web server sample)? Because I see in **setCaps** function it set framerate 
[image]

- When I tried to get videorate, it's None so I cant use your **set_max_rate** function
[image][image]
I used 1.0.0.4005 version of tiscamera https://www.theimagingsource.com/support/downloads-for-linux/install/tiscamera1804/

- What is the right way for me to combine your new **Webserver sample** with **TIS** file

Thank you so much

Regards,
An

TIS-Arne, 2022-11-14:
@dinhtuanan 

I am currently reworking the example so that the TIS class could be reused for the webserver too. This takes a bit more effort than expected. I will commit an updated version tomorrow.

TIS-Arne, 2022-11-15:
@dinhtuanan 

I updated the samples. The tcam_webserver example now uses the TIS class. I cleaned up the code and changed the class member names to be more consistent, so if you copied from the example code, please make sure you change it according to the updated version.

dinhtuanan, 2022-11-16:
Thank you so much @TIS-Arne, I will try your new example
I will close the issue for now, if there is a problem I will create another issue

Thank you very much

Regards,
An

closed 2022-11-16 by dinhtuanan