TheImagingSource/tiscamera #494: Using color board camera but images not colorized

jdog816, 2022-11-08:
My program is saving images the way I want to, except the pictures are grey. I tried changing the pipeline to:

"tcamsrc name=camera ! video/x-raw,format=BGRx,width=640,height=480,framerate=15/1 ! videocrop top=0 left=80 right=80 bottom=0 ! videoconvert ! jpegenc name=jpegencoder quality=" INITIALJPEGQUALITY ! appsink name=jpegframe"

but I'm getting these errors:

0:00:00.072662034 27631   0x5ee100 WARN     GST_ELEMENT_FACTORY gstelementfactory.c:456:gst_element_factory_make: no such element factory "video"!
0:00:00.072740160 27631   0x5ee100 ERROR           GST_PIPELINE grammar.y:816:priv_gst_parse_yyparse: no element "video"

GStreamer-CRITICAL **: 22:31:19.771: gst_element_make_from_uri: assertion 'gst_uri_is_valid (uri)' failed
0:00:00.072991999 27631   0x5ee100 ERROR           GST_PIPELINE grammar.y:971:priv_gst_parse_yyparse: no source element for URI "/x-raw,width=640,height=480,framerate=15/1"
0:00:00.075353255 27631   0x5ee100 ERROR           GST_PIPELINE grammar.y:901:priv_gst_parse_yyparse: link has no source [sink=@0x607ad8]

GStreamer-CRITICAL **: 22:31:22.586: gst_element_get_clock: assertion 'GST_IS_ELEMENT (element)' failed

GStreamer-CRITICAL **: 22:31:22.586: gst_clock_get_time: assertion 'GST_IS_CLOCK (clock)' failed

GStreamer-CRITICAL **: 22:31:22.586: gst_element_get_clock: assertion 'GST_IS_ELEMENT (element)' failed

GStreamer-CRITICAL **: 22:31:22.586: gst_clock_get_time: assertion 'GST_IS_CLOCK (clock)' failed

GStreamer-CRITICAL **: 22:31:22.586: gst_element_get_clock: assertion 'GST_IS_ELEMENT (element)' failed

GStreamer-CRITICAL **: 22:31:22.587: gst_clock_get_time: assertion 'GST_IS_CLOCK (clock)' failed

GStreamer-CRITICAL **: 22:31:22.587: gst_element_get_clock: assertion 'GST_IS_ELEMENT (element)' failed

GStreamer-CRITICAL **: 22:31:22.587: gst_clock_get_time: assertion 'GST_IS_CLOCK (clock)' failed
camera base time=99:99:99.9999999990:00:04.003387979 27631   0x606950 WARN                 basesrc gstbasesrc.c:3055:gst_base_src_loop:<tcammainsrc0> error: Internal data stream error.
0:00:04.003443022 27631   0x606950 WARN                 basesrc gstbasesrc.c:3055:gst_base_src_loop:<tcammainsrc0> error: streaming stopped, reason not-linked (-1)

GStreamer-CRITICAL **: 22:31:23.702: gst_message_parse_error: assertion 'GST_IS_MESSAGE (message)' failed
Segmentation fault

jdog816, 2022-11-08:
Full code:

#include <glib.h>
#include <gst/gst.h>
#include <gst/gstutils.h>
#include <glib/gstdio.h>
#include <linux/videodev2.h>
#include <sys/ioctl.h>
#include <tcam-property-1.0.h> /* gobject introspection interface */

// These parameters establish this program's runtime properies
#define MAXFILES              150  // Number of files to retain on disk (older files are removed)
#define MAXFILENAMELENGTH      30  // Permissible length of a filename
#define MINJPEGBUFFERSIZE   55000  // Minimum desired length of JPEG data
#define MAXJPEGBUFFERSIZE   65000  // Maximum desired length of JPEG data
#define MAXJPEGFILESIZE     65339  // Maximum JPEG output file size (if the buffer is too large,
                                   // truncate the end of the buffer)
#define INITIALJPEGQUALITY   "80"  // Nominal value for JPEG compression quality. Range is 0 to
                                   // 100, with 100 for best quality and largest file size.

GST_DEBUG_CATEGORY_STATIC (tiscameratoodroid);  // define category (statically)
#define GST_CAT_DEFAULT tiscameratoodroid       // set as default

//------------------------------------------------------------------------------------------------
// Structure to contain all our information, so we can pass it to callbacks
typedef struct _CustomData {
  GstClockTime      ProgramStartTime;
  struct timespec   ProgramStartSystemTime;
  GstElement        *pipeline, *camera, *videofilter, *exposure, *jpegencoder, *app_sink;
  GMainLoop         *main_loop;
  guint             FileIndex;
  gchar             Filenames[MAXFILES][MAXFILENAMELENGTH];
  guint             JpegQuality;
  gboolean          FirstPass, DoPrintBufferStatus;
  gint              Gain;
  gint              Exposure;
  guint             NmbBytesToWrite;
} CustomData;

//--------------------------------------------------------------------------------------------------
// callback: An error message is posted on the bus
static void error_cb (GstMessage *msg) {
  GError *err;
  gchar *debug_info;

  /* Print error details on the screen */
  gst_message_parse_error (msg, &err, &debug_info);
  g_printerr ("Error received from element %s: %s\n", GST_OBJECT_NAME (msg->src), err->message);
  g_printerr ("Debugging information: %s\n", debug_info ? debug_info : "none");
  g_clear_error (&err);
  g_free (debug_info);
}

//--------------------------------------------------------------------------------------------------
// callback: The appsink element has received a video buffer
static GstFlowReturn new_buffer (GstElement *sink, CustomData *data) {
  GstSample         *sample;
  struct timespec   CurTime, MonotonicTime;
  gchar             Filename[MAXFILENAMELENGTH];
  gchar             JpegComment[80];
  guint16           CommentLength;
  guint             BufferSize;
  guint             NmbBytesToWrite;
  GstClockTime      duration, timestamp;
  GstClockTime      Elapsed = GST_CLOCK_DIFF(data->ProgramStartTime,gst_util_get_timestamp());

  // Fetch the current time from the system clock
  clock_gettime ( CLOCK_REALTIME, &CurTime );
  GST_CAT_INFO(tiscameratoodroid,"Elapsed = %" GST_TIME_FORMAT ", CurTime = [%010li,%09li]",
               GST_TIME_ARGS(Elapsed), CurTime.tv_sec, CurTime.tv_nsec);
[... 177 lines cut ...]
  // Initialize data
  data.FileIndex           = 0;
  data.JpegQuality         = atoi(INITIALJPEGQUALITY);
  data.FirstPass           = TRUE;
  data.DoPrintBufferStatus = verbose;

  gst_element_set_state (data.pipeline, GST_STATE_PLAYING);
  g_print("camera base time=%" GST_TIME_FORMAT, GST_TIME_ARGS(gst_clock_get_time(gst_element_get_clock(data.camera))) );
  data.main_loop = g_main_loop_new (NULL, FALSE);
  g_main_loop_run (data.main_loop);

  // The program should only be terminated by Control-C or a system reboot.
  // We should never arrive at this point.
  printf ("This program is at point D\n");
  // Wait until error or EOS
  msg = gst_bus_timed_pop_filtered (bus, GST_CLOCK_TIME_NONE, GST_MESSAGE_ERROR | GST_MESSAGE_EOS);

  // Free resources
  if (msg != NULL) gst_message_unref (msg);
  gst_object_unref (bus);
  gst_element_set_state (data.pipeline, GST_STATE_NULL);
  gst_object_unref (data.pipeline);

  return 0;
}

jdog816, 2022-11-08:
Camera is DFM 22BUC03-ML

TIS-Stefan, 2022-11-09:
The images are gray scale, because you use the v4l2src. It does not do any debayerring, but the DFM 22BUC03-ML provides bayer raw data only. Better to go with tcambin instead of v4l2src. I do not remember, why we use v4l2src.
tcambin provides debayered color data.

tcambin name=camera ! video/x-raw,format=BGRx,width=640,height=480,framerate=15/1 ! videocrop top=0 left=80 right=80 bottom=0 ! videoconvert ! jpegenc name=jpegencoder quality=" INITIALJPEGQUALITY ! appsink name=jpegframe"

Should do the job.

jdog816, 2022-11-10:
I changed the pipeline to what you wrote and managed to clean up the errors, but it still produces gray images. Maybe there's another element needed in the pipeline?

TIS-Stefan, 2022-11-10:
Please try with tcam-capture and check, whether it shows a colored image.

jdog816, 2022-11-14:
When I run tcam-capture I get this error:

2022-11-14T17:10:17 - unknown:0: warning QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
2022-11-14T17:10:19 - unknown:0: info New device: 46410017-v4l2
2022-11-14T17:10:22 - unknown:0: info device selected: DFM 22BUC03-MLA-46410017-v4l2

2022-11-14T17:10:22 - unknown:0: debug Setting 'conversion-element' property to 'tcamconvert'
2022-11-14T17:10:42 - unknown:0: info settings caps to : video/x-raw, format=(string)GRAY8, width=(int)640, height=(int)480, framerate=(fraction)15/1, binning=(string)1x1, skipping=(string)1x1
2022-11-14T17:10:43 - unknown:0: info Error while retrieving property names: Error: Device not opened

** (tcam-capture:19215): CRITICAL **: 17:10:43.446: tcam_property_enumeration_get_value: assertion 'self != NULL' failed

jdog816, 2022-11-14:
I went back through previous threads and found your solution to the devices not opened error. Had to edit the file /root/.config/the_imaging_source/tcam-capture.conf and add the line video-sink-element=ximagesink

So now when I run tcam-capture it displays the image, but it's still gray.

TIS-Stefan, 2022-11-15:
Hello
Please open your DFK 22 in tcam-capture as follows:
[image]
Make sure "Open Format Dialog" is marked. (You can change the format later) 
Choose the gbrg format.
[image]
The gbrg format is a Bayer raw format, which which automatically converted into RGB or BGRx by tiscamera's tcambin for a colored image. You should get a result like this:
[image]
If the image stays gray, please click on "Format" and chose "GRAY8":
[image]
You may need to restart tcam-capture and choose GRAY8 from beginning. tcam-capture should show an image like this:
[image]
You see a faint checker board pattern, generated by the color sensor of the camera. 
If you see a nice gray image without this checkerboard pattern, then your camera has a monochrome sensor.

Which firmware file did you write in your camera? Its serial number 46410017 shows me, that is is an older model. The firmware in your camera should be version 3017 from dfk22uc03_3017.euvc file. Please refer to https://github.com/TheImagingSource/tcam-firmware-update
The camera I use is one year older than yours.

jdog816, 2022-11-15:
Hi Stefan

My camera was running a much older version than 3017, so I updated the firmware per the link and it got me a little further. When I opened tcam-capture before, it only gave me the GRAY8 option. Now it lists the gbrg option.

However, when I select gbrg it automatically goes back to GRAY8 after it starts taking pictures. And when running my program, the first few images are somewhat colorized but then the rest all fade back to gray.

Is it possible it needs an additional firmware update? (other than the file you mentioned)

TIS-Stefan, 2022-11-16:
Hi

The camera should behave exactly, as it does on the photos. I made the same steps you should have done. If the gbrg format is set, please check the "Saturation" property under "Properties". It should not be 0. Do you have the tiscamera-dutils installed?

Stefan

TIS-Stefan, 2022-11-16:
Hello 

> I followed your steps, but as I mentioned, when I select grbg it doesn't stay selected, it jumps back to GRAY8 by itself.

Weird. The very same works perfectly on my computer. OK, we checked, you have a color camera. 

Please run in a terminal window
```
export GST_DEBUG=tcam*:5
export GST_DEBUG_NO_COLOR=1
expor tGST_DEBUG_FILE=/tmp/tcam-gst.log
gst-launch-1.0 tcambin ! video/x-raw,format=BGRx,width=640,height=480,framerate=15/1 ! videoconvert ! ximagesink
```
After a video is shown, end gst-launch-1.0 and attach the file "/tmp/tcam-gst.log" here.

See also
https://www.theimagingsource.com/documentation/tiscamera/environment.html?highlight=debug#gst-debug

jdog816, 2022-11-16:
See attached

jdog816, 2022-11-16:
0:00:00.161318933  4060   0x602160 INFO                 tcambin gsttcambin.cpp:364:select_transform_element: Auto selected transform element: tcamconvert
0:00:00.167714857  4060   0x602160 INFO            tcam-libtcam libtcam_base.cpp:69:print_setup_to_logger: 
The following library versions are used:
	Tcam:	1.0.0_master/d9ada738_rev_4011
	Aravis:	0.8_version_d9ada738
	Modules:	gst_aravis_arvu3v_v4l2_libusb_tools_doc
0:00:00.188717918  4060   0x6026c0 INFO            tcam-libtcam aravis_utils.cpp:187:fetch_gige_daemon_device_list: Failed to create shmkey. Not using gige-daemon to enumerate devices.
0:00:02.228425143  4060   0x602160 DEBUG            tcammainsrc mainsrc_device_state.cpp:207:open_camera:<tcammainsrc0> Trying to open device with serial='46410017 and type='v4l2'.
0:00:02.987984705  4060   0x602160 DEBUG           tcam-libtcam V4l2Device.cpp:991:index_formats: Found format: Mono8
0:00:02.988103583  4060   0x602160 DEBUG           tcam-libtcam V4l2Device.cpp:991:index_formats: Found format: GBRG8
0:00:02.988143042  4060   0x602160 INFO            tcam-libtcam V4l2Device.cpp:261:get_framerate: Current framerate is 15 / 1 fps
0:00:02.988336212  4060   0x602160 DEBUG           tcam-libtcam V4l2Device.cpp:178:get_available_video_formats: Returning 2 formats.
0:00:02.988694970  4060   0x602160 INFO            tcam-libtcam SoftwarePropertiesExposureAuto.cpp:38:generate_exposure_auto: Adding software ExposureAuto.
0:00:02.988741763  4060   0x602160 INFO            tcam-libtcam SoftwarePropertiesExposureAuto.cpp:104:generate_gain_auto: Adding software GainAuto.
0:00:02.988780680  4060   0x602160 INFO            tcam-libtcam SoftwarePropertiesExposureAuto.cpp:208:generate_auto_functions_roi: Adding software AutoFunctionsROI
0:00:02.988827473  4060   0x602160 INFO            tcam-libtcam SoftwarePropertiesBalanceWhite.cpp:81:generate_balance_white_auto: Adding BalanceWhiteAuto software based.
0:00:02.988916474  4060   0x602160 DEBUG           tcam-libtcam V4l2Device.cpp:178:get_available_video_formats: Returning 2 formats.
0:00:02.988990809  4060   0x602160 DEBUG           tcam-libtcam V4l2Device.cpp:178:get_available_video_formats: Returning 2 formats.
0:00:02.989044185  4060   0x602160 DEBUG           tcam-libtcam V4l2Device.cpp:178:get_available_video_formats: Returning 2 formats.
0:00:02.989099270  4060   0x602160 DEBUG           tcam-libtcam V4l2Device.cpp:178:get_available_video_formats: Returning 2 formats.
0:00:02.989162563  4060   0x602160 DEBUG           tcam-libtcam V4l2Device.cpp:178:get_available_video_formats: Returning 2 formats.
0:00:02.989212314  4060   0x602160 DEBUG           tcam-libtcam V4l2Device.cpp:178:get_available_video_formats: Returning 2 formats.
0:00:02.989376192  4060   0x602160 DEBUG            tcammainsrc mainsrc_device_state.cpp:230:open_camera:<tcammainsrc0> Device provides the following caps: video/x-raw, format=(string)GRAY8, width=(int)320, height=(int)240, framerate=(fraction){ 15/1, 25/1, 30/1, 60/1, 80/1, 120/1, 5000/31 }; video/x-raw, format=(string)GRAY8, width=(int)640, height=(int)480, framerate=(fraction){ 5/1, 15/2, 15/1, 25/1, 30/1, 60/1, 87/1 }; video/x-raw, format=(string)GRAY8, width=(int)744, height=(int)480, framerate=(fraction){ 5/1, 15/2, 15/1, 25/1, 30/1, 60/1, 20000/263 }; video/x-bayer, format=(string)gbrg, width=(int)320, height=(int)240, framerate=(fraction){ 15/1, 25/1, 30/1, 60/1, 80/1, 120/1, 5000/31 }; video/x-bayer, format=(string)gbrg, width=(int)640, height=(int)480, framerate=(fraction){ 5/1, 15/2, 15/1, 25/1, 30/1, 60/1, 87/1 }; video/x-bayer, format=(string)gbrg, width=(int)744, height=(int)480, framerate=(fraction){ 5/1, 15/2, 15/1, 25/1, 30/1, 60/1, 20000/263 }
0:00:02.989725866  4060   0x602160 INFO                 tcamsrc gsttcamsrc.cpp:342:open_source_element:<tcambin-source> Opened device with serial: '46410017' type: 'v4l2'
0:00:02.989870161  4060   0x602160 INFO                 tcambin gsttcambin.cpp:177:tcambin_create_source:<tcambin0> Opened device has serial: '46410017' type: 'v4l2'
0:00:02.990141708  4060   0x602160 INFO                 tcambin gsttcambin.cpp:411:tcambin_create_elements:<tcambin0> creating all tcambin pipeline elements
0:00:02.992579384  4060   0x602160 DEBUG                tcambin gsttcambin.cpp:379:create_and_add_element:<tcambin0> Adding tcamconvert(0x622090) to pipeline
0:00:02.996100624  4060   0x602160 DEBUG            tcamconvert tcamconvert.cpp:179:transform_caps: Returning INPUT: video/x-bayer, format=(string)bggr, width=(int)[ 1, 2147483647 ], height=(int)[ 1, 2147483647 ]; video/x-bayer, format=(string)bggr10, width=(int)[ 1, 2147483647 ], height=(int)[ 1, 2147483647 ]; video/x-bayer, format=(string)bggr10sp, width=(int)[ 1, 2147483647 ], height=(int)[ 1, 2147483647 ]; video/x-bayer, format=(string)bggr10m, width=(int)[ 1, 2147483647 ], height=(int)[ 1, 2147483647 ]; video/x-bayer, format=(string)bggr12, width=(int)[ 1, 2147483647 ], height=(int)[ 1, 2147483647 ]; video/x-bayer, format=(string)bggr12p, width=(int)[ 1, 2147483647 ], height=(int)[ 1, 2147483647 ]; video/x-bayer, format=(string)bggr12sp, width=(int)[ 1, 2147483647 ], height=(int)[ 1, 2147483647 ]; video/x-bayer, format=(string)bggr12m, width=(int)[ 1, 2147483647 ], height=(int)[ 1, 2147483647 ]; video/x-bayer, format=(string)bggr16, width=(int)[ 1, 2147483647 ], height=(int)[ 1, 2147483647 ]; video/x-bayer, format=(string)bggr10, width=(int)[ 1, 2147483647 ], height=(int)[ 1, 2147483647 ]; video/x-bayer, format=(string)bggr1 [... cut ...]
0:00:12.638869345  4060   0x602160 DEBUG           tcam-libtcam V4l2Device.cpp:418:stop_stream: Stopped stream
0:00:12.732410203  4060   0x602160 DEBUG           tcam-libtcam V4l2Device.cpp:395:stop_stream: Stopping stream
0:00:12.732466287  4060   0x602160 DEBUG           tcam-libtcam V4l2Device.cpp:418:stop_stream: Stopped stream
0:00:12.732504580  4060   0x602160 DEBUG           tcam-libtcam V4l2Device.cpp:395:stop_stream: Stopping stream
0:00:12.732528664  4060   0x602160 DEBUG           tcam-libtcam V4l2Device.cpp:418:stop_stream: Stopped stream

jdog816, 2022-11-16:
Sorry about that, ha...

TIS-Stefan, 2022-11-16:
I have to ask my expert. The pipeline is correct, the caps are set correctly, but you still see a gray scale image. 

In the meantime, you may install the tcam-dutils from https://www.theimagingsource.com/support/downloads-for-linux/ for your platform. It has a different debayering. (And saturation). Try the pipeline with gst-launch-1.0 after installation and see, what happens. 

Stefan

jdog816, 2022-11-16:
I tried installing tcam-dutils for 32 bit OS from here: https://www.theimagingsource.com/support/downloads-for-linux/install/tiscameradutilsarmhfpi/
but it won't install:  Error: Dependency is not satisfiable: tiscamera-tcamproperty (>= 1.0)

I can't find tiscamera-tcamproperty anywhere, but this link says the tiscamera package (which I have installed) is a drop-in replacement for it, so I'm not sure why it's not working.

TIS-Stefan, 2022-11-16:
Have to check.

TIS-Stefan, 2022-11-17:
Hello

> One thing I noticed is in the Info tab in tcam-capture, it says all the listed packages are not installed. But I cloned tiscamera from the github page and followed the instructions to install: https://github.com/TheImagingSource/tiscamera

That is the reason, why tcam-dutils do not find the tcamprop package. Well, lets ignore that. 

Please change into the your firmware updater directory and let us see the output of

cam-firmware-update -id 46410017

Also please show the output of:

tcam-ctrl --all-info

At least, please make
ls /dev/v4l/by-id/

and show he output of

v4l2-ctl --list-formats-ext -d /dev/v4l/by-id/usb-The_Imaging_Source_Europe_GmbH_DFK_22BUC03_......-index0

Replace "......" with the data shown from the ls command.

We checked on a second computer and also got a nicely colored image. Thus, we are puzzled.

Stefan

jdog816, 2022-11-17:
output of tcam-ctrl --all-info

=============
=  General  =
=============
======= lsb_release -a

No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.3 LTS
Release:	18.04
Codename:	bionic

======= uname -a

Linux odroid 4.14.141-169 #1 SMP PREEMPT Sat Aug 31 23:19:59 -03 2019 armv7l armv7l armv7l GNU/Linux

======= lscpu

Architecture:        armv7l
Byte Order:          Little Endian
CPU(s):              8
On-line CPU(s) list: 0-7
Thread(s) per core:  1
Core(s) per socket:  4
Socket(s):           2
Vendor ID:           ARM
Model:               3
Model name:          Cortex-A7
Stepping:            r0p3
CPU max MHz:         2000.0000
CPU min MHz:         200.0000
BogoMIPS:            90.00
Flags:               half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae

=============
=    USB    =
=============
======= lsusb

Bus 006 Device 002: ID 0bda:8153 Realtek Semiconductor Corp. 
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 002: ID 05e3:0616 Genesys Logic, Inc. hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 004: ID 046d:c018 Logitech, Inc. Optical Wheel Mouse
Bus 003 Device 003: ID 0461:3f41 Primax Electronics, Ltd 
Bus 003 Device 002: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 005: ID 199e:8302 The Imaging Source Europe GmbH 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

======= lsusb -t

/:  Bus 06.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 5000M
    |__ Port 1: Dev 2, If 0, Class=Vendor Specific Class, Driver=r8152, 5000M
/:  Bus 05.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 480M
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 5000M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/2p, 5000M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/2p, 480M
        |__ Port 1: Dev 3, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 1: Dev 3, If 1, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 2: Dev 4, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=exynos-ohci/3p, 12M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=exynos-ehci/3p, 480M
    |__ Port 1: Dev 5, If 0, Class=Video, Driver=uvcvideo, 480M
    |__ Port 1: Dev 5, If 1, Class=Video, Driver=uvcvideo, 480M

=============
=  Network  =
=============
======= ip a

1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 1000
    link/ether 00:1e:06:39:fb:ee brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.108/24 brd 192.168.1.255 scope global dynamic noprefixroute eth0
       valid_lft 3069537860sec preferred_lft 3069537860sec
    inet6 fe80::c4df:acf5:15e2:93b4/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever

=============
= Packaging =
=============
tiscamera is not installed.
tiscamera-tcamproperty is not installed.
tcamdutils is not installed.
tcampimipisrc is not installed.
tcamtegrasrc is not installed.
tcamdutils-cuda is not installed.
theimagingsource-drivers is not installed.
ic_barcode is not installed.

jdog816, 2022-11-17:
output of ls /dev/v4l/by-id/

usb-The_Imaging_Source_Europe_GmbH_DFM_22BUC03-MLA_46410017-video-index0

TIS-Stefan, 2022-11-17:
I will answer to your last question right now:
Building and sudo make install do not make any entries in Linux packet lists. Therefore, the installation of tcam-dutils can not find the tiscamera package and wont install. 
Tiscamera packets can be found at https://www.theimagingsource.com/support/downloads-for-linux/

However, that is no reason, why the GBRG is not colored on your computer. I must discuss with my Linux expert tomorrow. I am totally lost, why this does not work correctly on your odroid .

Stefan

jdog816, 2022-11-17:
I had unplugged the camera and plugged it back in, and now it throws this error when running tcam-capture:

unknown:0: info New device: 46410017-v4l2
tcam-capture: relocation error: tcam-capture: symbol _ZN11QMetaObject16invokeMethodImplEP7QObjectPN9QtPrivate15QSlotObjectBaseEN2Qt14ConnectionTypeEPv version Qt_5 not defined in file libQt5Core.so.5 with link time reference

I tried rebooting but no luck. I was able to get a screenshot of the gst-launch-1.0 tcambin pipeline so I'll send you that.

jdog816, 2022-11-17:
Please disregard my last comment - I re-cloned and re-installed tiscamera from github and now I can use tcam-capture again. I tested it out on another camera (same model) but the images are still gray. It sounds like it's because I'm using the github clone and not the downloadable package, and thus is not on the linux package list, like you said. So I think I'll need to switch to a different platform. But I'm curious what your linux expert has to say.

jdog816, 2022-11-17:
I switched from the XU4 to the C4 odroid running ubuntu 22, and was able to install the tiscamera and tcamdutils arm64 packages from the website. But when I try to run tcam-capture, it won't launch and I get this error:

    2022-11-17T20:56:05 - unknown:0: warning could not connect to display 
    2022-11-17T20:56:05 - unknown:0: info Could not load the Qt platform plugin "xcb" in "" even though it was found.
    2022-11-17T20:56:05 - unknown:0: fatal This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

    Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, xcb.

    Aborted (core dumped)

My code runs on this machine and takes pictures, but they are gray here too. The gst-launch-1.0 pipeline isn't working and there isn't a /root/.config

I plan to try installing tiscamera on a pi 2b when I get a chance, and see if that works.

TIS-Stefan, 2022-11-18:
I am a little bit lost. So lets start with:

```
gst-launch-1.0 v4l2src ! bayer2rgb ! videoconvert ! ximagesink
```

That should provide a green, but colored image. You can detect other colors in the image. If that works use
```
gst-launch-1.0 tcamsrc ! bayer2rgb ! videoconvert ! ximagesink
```
That gives the same result. If that works, then do
```
gst-launch-1.0 tcambin ! bayer2rgb ! videoconvert ! ximagesink
```
That should give a white balanced image with correct colors.

jdog816, 2022-11-18:
Got it working! I got a second camera (same model) and it didn't work at first, but then I pushed the same firmware update to it, and now it works with the pipelines you just posted, as well as with the pipeline I originally posted at the top of this thread. I guess the other camera went bad somehow. So it was the firmware update that did the trick, along with porting to the latest software.

Thank you for all your help with this, Stefan!